rstrouse/ESPSomfy-RTS #348: Sometimes crashes when unplugging Ethernet cable on wt32-eth01

irrwisch1, 2024-04-26:
### Hardware

ESP32 WT32-ETH01

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

Hi,
I recently switched to a wt32-eth01 board due to wifi issues and am currently struggling with a few issues.
I had the board hanging up or dropping ethernet connection a few times (but I'm not sure what exactly happened since I didn't have anything connected to the debug port at that time) and while playing around I found another issue.

When unplugging the ethernet cable it sometimes happens that the firmware crashes. It does not always happen though.
I've attached the log messages that appear during the crash.

Thanks and Best Regards,
Julien

### How to reproduce it (step by step)

```bash
1. Power up the board
2. When booted up unplug the ethernet cable
```

### Logs

```bash
[09:21:53]Free Heap: 141692
[09:21:53]Min Heap: 128368
[09:22:45]Socket [1] Disconnected!
Ethernet Disconnected
[09:22:45]CORRUPT HEAP: Bad head at 0x3ffe3080. Expected 0xabba1234 got 0x3ffe301c
[09:22:45]
[09:22:45]assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
[09:22:45]
[09:22:45]
[09:22:45]Backtrace: 0x40083cf9:0x3ffbbd50 0x4008f8a5:0x3ffbbd70 0x40094d25:0x3ffbbd90 0x4009491d:0x3ffbbec0 0x40084299:0x3ffbbee0 0x40094d55:0x3ffbbf00 0x40196879:0x3ffbbf20 0x400f4b49:0x3ffbbf40 0x401ad4f1:0x3ffbbf60 0x400fcb22:0x3ffbbf80 0x401ad48b:0x3ffbbfa0 0x400fc61f:0x3ffbbfc0 0x400fc65f:0x3ffbc000 0x400e6439:0x3ffbc020 0x400dd295:0x3ffbc040 0x400da012:0x3ffbc060 0x400f5c2c:0x3ffbc090 0x400f5ecd:0x3ffbc220
[09:22:45]
[09:22:45]
[09:22:45]
[09:22:45]
[09:22:45]ELF file SHA256: d6ecab55ea8f4e64
[09:22:45]
```

rstrouse, 2024-04-27:
I’ll run some tests.

rstrouse, 2024-05-08:
I found the issue and will be publishing a release to fix it.

rstrouse, 2024-05-12:
v2.4.3 has been released and this issue has been fixed.

closed 2024-05-12 by rstrouse