rstrouse/ESPSomfy-RTS #350: Groups not working after restore

alka79, 2024-04-26:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

I changed my hardware for an ESP32 board. I reinstalled from scratch and restored from backup file. 
All the shades work again normally individually after the restore.
However, I have 4 groups. Two of them do not respond to any command anymore. The two other groups work fine.
I see in the Serial port that the commands are sent to the correct groupID with the correct remote address. The members of the group change status in the WebUI but the shades themselves do not react at all. As if they did not receive or not recognize the command sent from the group remote. 

Could it be a rolling code issue with the group remote ? 

Any idea other than unpair the group remotes from all the shades and recreate the groups ? 

### How to reproduce it (step by step)

```bash
click any button for these groups
```

rstrouse, 2024-04-26:
It does appear the rolling code is out of sync.  Sometimes you can send enough command to get it to catch up.

alka79, 2024-04-26:
I'll try some more times to try to catch up.  
Shouldn't the last rollling code for group remotes be saved in the backup file like the other remotes ?

alka79, 2024-05-01:
Unfortunately, catchup does not work.
The two non-working-groups members are old shutters. The two working-groups members are recent blinds.
Maybe the old shutters have less tolerance to rollingcodes out of sync.

After digging deeper in the source code, I see that the rollingcodes for all remotes (shades and groups) are stored in the preferences, ie in non volatile flash supposedly not wiped during OTA updates.
Shade remote lastrollingcode are also stored in a config file in flash with the Shade settings  and can be restored from there. Group remote lastrollingcode are not stored on file and thus can not be restored. Backup file is a copy of the config file.

Someone updating on the same board may never notice the problem as the rcodes are still in NVS flash. But updating to a new board will certainly create an issue with the groups. Their rollingcodes will restart at 0 and thus be seriously out of sync with the shades motor memory. It should be fixed or mentioned as a limitation. 

Group remote can be paired with the shades in the group, but can not be unpaired. 

The only option I see is to delete these two groups from the controller and recreate them with a new remote address.

rstrouse, 2024-05-01:
>  It should be fixed or mentioned as a limitation.

I agree.  I cannot believe I missed that and will be adding the lastRollingCode to the group definition.  You are correct in that users on the same board would not see and issue but restoring to a new one would be a problem.

rstrouse, 2024-05-04:
The rolling code has been added to the group backup/restore in v2.4.3

rstrouse, 2024-05-11:
I am going to close this for now as we detected why the issue was occurring and I fixed it in v2.4.3 that I expect to release to the gen-pop this weekend.

closed 2024-05-11 by rstrouse