rstrouse/ESPSomfy-RTS #351: Blind Position

Darozas, 2024-04-26:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

Blind position if set say 40% in Home assistant only opens line 2% and shows as fully open.

### How to reproduce it (step by step)

```bash
1. Set position to 40% in home assistant
2. only opens up by 2% 
3. is there a way to control this?
...
```

rstrouse, 2024-04-27:
Did you set the timing for the blind movement in milliseconds?

Darozas, 2024-04-29:
Thanks for your time. No i have all as default did i missed something here on how to set timings?

Regards

rstrouse, 2024-04-29:
You need to set the timing on the motors. 

Here are some instructions 
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#motor-timing

closed 2024-04-29 by Darozas