rstrouse/ESPSomfy-RTS #352: Home page only shows logo when using ethernet connection

Screech1, 2024-04-27:
### Hardware

LilyGO T-Internet POE

### Firmware version

2.4.2

### Application version

2.4.2

### What happened? What did you expect to happen?

If I go from WiFi to ethernet connection the home page wil only show a logo instead of the complete menu.

I use ESPSomfy RTS on a LilyGo ETH POE and build and upload it with PlatformIO.

2.4.1 still works tho.

### How to reproduce it (step by step)

```bash
Go from WiFi to Ethernet connection
```

rstrouse, 2024-04-27:
Install the firmware again and clear your browser cache.

Screech1, 2024-04-28:
Thanks! haven't thougt about that, so simple🤓

rstrouse, 2024-04-29:
Yeah sometimes the browser caches the files when it shouldn’t.  I am going to close this for now.

closed 2024-04-29 by rstrouse