rstrouse/ESPSomfy-RTS #353: Pinout

Boosmoo, 2024-04-28:
I have a esp32 c3 super mini and I have a dumb question about the pinout. Should the RX pin on the esp32 be connected to the TX pin of the cc1101 and vise versa? because of tx should be connected to tx because then I cant select the right pins.

rstrouse, 2024-04-28:
The communication between the cc1101 and the ESP32 is not using a serial protocol like rsXXX. The chip communicates over SPI which is the top row of pins in the config pages and the transceiver over gpio.  Simply assign the tx pin to GDO0 on the transceiver and the rx to GDO2.  Ignore the rx and tx assignments those are for another purpose.

Boosmoo, 2024-04-28:
Okay. But its not available in the menu then the tx pin is 21 on my esp but the menu only goes to 20

rstrouse, 2024-04-28:
Use another pin.  The pin marked tx on the c3 is likely a jtag pin.

Boosmoo, 2024-04-28:
Used 9 and 10 but it still doesnt work. It keeps saying radio not initialized.

rstrouse, 2024-04-28:
Then that is not because of the rx and tx pins.  The upper row of pins are the spi chip configuration.  If the pin selection does not match the wiring for sck,mosi,miso,and csn it will not be able to initialize the transceiver.  Just to be sure you did you check the enable radio checkbo?

Boosmoo, 2024-04-28:
Yes I triple checked the connections and the radio is enabled.

Boosmoo, 2024-04-28:
[image]

I used this pinout.

rstrouse, 2024-04-29:
I would not connect anything to the rx/tx pins on that board.  Most often that is attached to uart0 and is for serial console.

Boosmoo, 2024-04-30:
I connected them to 9 and 10 but it still doesnt work.

rstrouse, 2024-05-01:
According to your pic the wiring is setup like below and does not match the configuration screen.   

| Tr Pin | Color | Func | ESP32 |
| --- | --- | --- | --- |
| 1 | White | GND | GND |
| 2 | Yellow | 3v3 | 3v3 |
| 3 | Purple | GDO0/TX | GPIO7 |
| 4 | Gray | CSN | GPIO9 |
| 5 | Yellow | SCLK | GPIO6 |
| 6 | Orange | MOSI | GPIO4 |
| 7 | Green | MISO | GPIO10 |
| 8 | Brown | GDO2/RX | GPIO5 |

Your configuration looks like it should be like below

[image]

rstrouse, 2024-05-11:
@Boosmoo Did you get your issue resolved or is there anything more that I can do to help?

Boosmoo, 2024-05-11:
It still doesnt work

rstrouse, 2024-05-11:
Just in case I missed something, I inventoried your wiring again.  I assume the wire going from pin 3 on the E07-M1101D is gray and terminates at GPIO9.  Please verify that the connections are as I believe they are since I assumed this from your picture.

| Tr Pin | Color | Func | ESP32 |
| --- | --- | --- | --- |
| 1 | White | GND | GND |
| 2 | Yellow | 3v3 | 3v3 |
| 3 | Gray | GDO0/TX | GPIO9 |
| 4 | Purple | CSN | GPIO7 |
| 5 | Orange | SCLK | GPIO4 |
| 6 | Yellow | MOSI | GPIO6 |
| 7 | Brown | MISO | GPIO5 |
| 8 | Green | GDO2/RX | GPIO10 |

If this is how you have it wired and your configuration is as pictured on from your screenshot above we should do some additional troubleshooting.  First update your firmware to v2.4.3 pre.   This uses the Arduino core 2.0.16 which has fixes to the C3 chips.  I did rewire a C3 as above and it worked fine for me on v2.4.2.  However, mine is a Seeed Studio board and not ICBuy board and according to the PRs in the Arduino core there have been fixes for the SPI interface.

Next please look very closely and make sure that no stray strands of wire are connecting especially between GPIO5, GPIO6, or GPIO7.  Look very closely on the transceiver as well.  The Radio not Initialized message happens when the configuration is sent to the E07M1101 over the SPI interface and it cannot be read back.   

If after performing these steps it still does not initialize then you must have either a defective ESP32 or E07M1101D board.  There is no real way to tell which one is goofy.

rstrouse, 2024-05-21:
@Boosmoo have you figured out what the issue is.  I have confirmation that once the wiring was fixed in the issue linked above for the same hardware that it worked fine.

Nickduino, 2024-05-22:
> I would not connect anything to the rx/tx pins on that board. Most often that is attached to uart0 and is for serial console.

I used them myself
[image]

Are they used only to receive data when the radio is listening?

rstrouse, 2024-05-23:
The rx pin is attached to an interrupt when listening and the tx pin is pulsed to send the frame.   However, the tx pin also receives when listening.  Typically a uart pulls the pins but I think it depends on how it is implemented.

Nickduino, 2024-05-23:
But they're used to listen only, so the fact they're assigned to UART pins should not negatively impact the emitting range of the device, right?

Jezza34000, 2024-05-23:
Hi on my side i just start to build a ESPSomfy RTS with a ESP32-C3 & AS07-M1101S and with this wiring i only receive RF but i can't send command, i've try to switch RX/TX but still same.

| Pin  | RF    | ESP32  |
|--------|---------|--------|
| 1      | GND     | GND    |
| 2      | 3v3     | 3v3    |
| 3      | GDO0/TX | GPIO20 |
| 4      | CSN     | GPIO7  |
| 5      | SCLK    | GPIO4  |
| 6      | MOSI    | GPIO6  |
| 7      | MISO    | GPIO5  |
| 8      | GDO2/RX | GPIO21 |

The AS07-M1101S RF module was working well with a "classic" ESP32

rstrouse, 2024-05-23:
Actually a uart is a transceiver chip where TX pin is pulled low.   I don't know if this is done even if the UART is not enabled or if this is simply the wiring.  I guess it depends on its implementation on the board.

Nickduino, 2024-05-24:
I mean by ESPSomfy. The use of the TX/RX pins is for **receiving** frames from other remotes, right? When the CC1101 is receiving.

Said differently, can you omit the TX/RX connection and still control the blinds?

Nickduino, 2024-05-25:
Alright, thank you!

I had made this assumption because at some point I made a mistake and inverted the Tx and Rx pins but the frames were still sent. Not received, though.

rstrouse, 2024-05-25:
Yeah most of us have been through through some level of full duplex serial communication and its hard to get the twisted pair out of our heads.

Unfortunately, with some of the board manufacturers they leave out important details regarding their implementation.  A simple footnote on the pinout diagram would suffice.  Especially, when they solder their own peripherals onto the board.

rstrouse, 2024-06-15:
@Boosmoo I am going to close this as I assume you have gotten the issue resolved by now.

closed 2024-06-15 by rstrouse