rstrouse/ESPSomfy-RTS #355: io-homecontroll support

MarekVigas, 2024-05-02:
### Describe you new feature you'd like

@rstrouse  Thanks for your amazing work on this project. 

I wanted to get my new blinds with RTS and us this project. When I reached out to my local installers they told me that nowadays they only install IO motors. They installed last RTS motors years ago and they are not able to order RTS motors. Looks like io-homecontroll is inevitable. 

I found [this article](https://deralchemist.wordpress.com/2021/05/10/reverse-engineering-remote-controlled-somfy-blinds-part-1/) that started reverse engineering the protocol. Looks like [iown-homecontrol](https://github.com/Velocet/iown-homecontrol) started building on top of that article and they are able to control the blinds. 

Is there a way to add support for IO to this project ?

rstrouse, 2024-05-02:
Unfortunately, not from me.  I do not have any IO home control motors and will never buy them.  I have no desire to install any product that has functionality reliant on external servers with no clear reason for it to be required.

While your installers are telling you this I see folks with new installs all the time on RTS.  However inevitable I don't have the desire or bandwidth to spend time on IO.  That will be for another Somfy user to figure out how to get the 3rd party booger hooks out of their blinds.

MarekVigas, 2024-05-09:
> I have no desire to install any product that has functionality reliant on external servers with no clear reason for it to be required.

The same is true for me. I found out that IO should only be newer version of RTS with 2 way comunication and it is still just radio protocol. There is no need to use any of their hubs and connect IO motors to cloud. 

Looks like the [iown-homecontrol](https://github.com/Velocet/iown-homecontrol) team was able to successfully reverse engineer the IO protocol and are able to control the blinds with ESP32.

rstrouse, 2024-05-09:
@Velocet should be able to help you out there.  He gives some advice here on RTS protocol differences and has a grasp on IO protocols.

Just wait!  Cloud infrastructure and bandwidth isn't getting cheaper, these costs are growing exponentially.   When those costs start eating into their operational budgets it has to be offset with Annual Recurring Revenue (ARR) under subscription.  This has become so prevalent that there is a market for analyzing personal expenses to identify subscriptions that you may have forgotten about so you can churn out.  

It is one thing if there are clear subscription benefits but it gets really insidious when service ends and features that you originally purchased go away or the product just stops working.  For me those products will remain on the shelf.  My box of retired hubs and hobbled equipment is full at this point.

rstrouse, 2024-05-18:
I am going to close this but if anyone wants to strap this on feel free to use anything here.

closed 2024-05-18 by rstrouse