rstrouse/ESPSomfy-RTS #356: Sliding for opening the covers in HomeAssistant

fft1010, 2024-05-03:
Hi,
I do not exactly know if this is the right place, but I think here is the target audience :-) By accident I found out, that I could open the sliders in any opening percentage I want to use. In  another small project I am using "sliders" from here: https://github.com/thomasloven/lovelace-slider-entity-row

If you combine this, you can get something like

[image]

which is a card in HA that looks like: (Rolladen is one cover, Rolläden are more than one cover.

```
type: entities
state_color: true
title: Rolläden
show_header_toggle: true
entities:
  - entity: cover.rolladen
    name: links
    type: custom:slider-entity-row
    step: 1
    full_row: true
    toggle: false
    hide_state: false
    hide_when_off: false
    show_icon: false
    colorize: true
  - entity: cover.l1
    name: L1
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.l2
    name: L2
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.l3
    name: L3
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.l4
    name: L4
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.r1
    name: R1
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.r2
    name: R2
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.r3
    name: R3
    type: custom:slider-entity-row
    step: 1
    toggle: true
  - entity: cover.r4
    name: R4
    type: custom:slider-entity-row
    step: 1
    toggle: true
```

in ESPSSomfy I had to add a "repeat command" for every of my 8 covers. 2 was enough. Before not adding this, not all covers seem to get the commands. 
[image]

I hope this helpy others as well. If you think there is a better place for this, than just tell me here, and I can put it somewhere else.

Have fun.
Juergen

alka79, 2024-05-04:
Nice finding :) thanks for sharing. Maybe readers here https://community.home-assistant.io/t/esp-somfy-rts-integration/543401 would like it too.

PS: I wonder how you got such  precise timings for up and down.

fft1010, 2024-05-04:
Hi @alka79,
I just did the timing for each cover for up and down as described in the doc here in espsomfy. 
setup -> somfy -> shades -> individual shade .   Than just use a stopwatch for timing, and put in the time in ms I think. Be sure to stop up and down individually, as at least my shades are way different in up and down.

works very accurat, but as the shades do not feedbak their position, it is not working in 100%, only 99,5 :-)

thanks
Juergen

alka79, 2024-05-04:
I was surprised by your setting at 19180 ms. That is quite precise ! I was thinking you had some kind of electronic device to measure.
I use a stopwatch too on my smartphone. But I only trust my accuracy at half a second (at best) ;)

fft1010, 2024-05-05:
Hi @alka79,
that‘s true for sure and I was aware of it, but as I did not know if I was too fast or too slow, I simply took what was shown (for betterr results I should measure each cover 10 times each up and down :-) but still I have a life to live ). It works quit well as it is.