rstrouse/ESPSomfy-RTS #357: Cannot get it installed on several boards

magtimmermans, 2024-05-03:
### Hardware

ESP32, ESP32-S3

### Firmware version

V2.4.1

### Application version

V2.4.1

### What happened? What did you expect to happen?

I upgraded to V2.4.2 but it became very unstable. So I try to flash the previous version but I couldn't get it done.  Lots of `[14:34:49]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:34:49]invalid header: 0x2f746567`
errors.

I tried different ESP boards (MINI d1 ESP32 VROOM 4MB etc) but all the same error, also tried different versions. Seems not working either. Tried all different flash tools, also same result.
Finally I ordered a brand new ESP32-S3 board and this installed but gave a checksum error :(

### How to reproduce it (step by step)

```bash
1. Download software (V2.4.1)
2. Install it using esptool
3. Check the logging
```

### Logs

```bash
Az-Delivery board (ESP32 VROOM 4MB)

C:\Users\info\Downloads\esptool-v4.5.1-win64\esptool-v4.5.1-win64>esptool.exe --port COM3 write_flash 0x0 c:\users\info\Downloads\SomfyController.ino.esp32.bin
esptool.py v4.5.1
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:a3:b3:27:de:14
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0013dfff...
Compressed 1301888 bytes to 808916...
Wrote 1301888 bytes (808916 compressed) at 0x00000000 in 71.6 seconds (effective 145.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Startup log:

[14:34:49]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:34:49]invalid header: 0x2f746567
[14:34:49]invalid header: 0x2f746567
[14:34:49]invalid header: 0x2f746567
[14:34:49]invalid header: 0x2f746567
[14:34:50]invalid header: 0x2f746567
[14:34:50]invalid header: 0x2f746567
[14:34:50]invalid header: 0x2f746567
[14:34:50]invalid header: 0x2f746567
[14:34:50]ets Jul 29 2019 12:21:46

ESP32-S3 board:
C:\Users\info\Downloads\esptool-v4.5.1-win64\esptool-v4.5.1-win64>esptool.exe --port COM8 write_flash 0x0 c:\users\info\Downloads\SomfyController.ino.esp32s3.bin
esptool.py v4.5.1
Serial port COM8
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:da:0c:29:ca:00
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0011dfff...
Compressed 1170576 bytes to 745396...
Wrote 1170576 bytes (745396 compressed) at 0x00000000 in 65.4 seconds (effective 143.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Logfile:
[13:29:28]Attempting to boot anyway...
[13:29:28]entry 0x40377468
[13:29:29]ESP-ROM:esp32s3-20210327
[13:29:29]Build:Mar 27 2021
[13:29:29]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[13:29:29]Saved PC:0x40379b26
[13:29:29]SPIWP:0xee
[13:29:29]mode:DIO, clock div:1
[13:29:29]load:0x3c0e0020,len:0x338a0
[13:29:29]ESP-ROM:esp32s3-20210327
[13:29:29]Build:Mar 27 2021
[13:29:29]rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[13:29:29]SPIWP:0xee
[13:29:29]mode:DIO, clock div:1
[13:29:29]load:0x3c0e0020,len:0x338a0
[13:29:29]load:0x3fc953d0,len:0x56dc
[13:29:29]load:0x40374000,len:0x706c
[13:29:29]load:0x42000020,len:0xd2100
[13:29:29]load:0x4037b06c,len:0xa360
[13:29:29]SHA-256 comparison failed:
[13:29:29]Calculated: b7c5cfe4a529ac4dc21bc60c1485b859b56a1baf2dd2c6a01a3eeb6ea85570b3
[13:29:29]Expected: 84692e1c9a88e3a850779a371b52a063ba3ad8dd4fb50f8868c044a6ea322e92
[13:29:29]Attempting to boot anyway...
```

rstrouse, 2024-05-03:
You are not flashing the onboard file you are flashing the firmware sector.  To downgrade it is as simple as selecting the version in the github update to a previous version.

magtimmermans, 2024-05-03:
> You are not flashing the onboard file you are flashing the firmware sector. To downgrade it is as simple as selecting the version in the github update to a previous version.

Thanks for the quick reply. I try again.

magtimmermans, 2024-05-03:
It works, thanks

closed 2024-05-03 by magtimmermans