rstrouse/ESPSomfy-RTS #358: Backup Restore not possible v2.4.0 -> v2.4.2

DirkHBo, 2024-05-04:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

As I had trouble with my v.2.4.0 install upgrading to a newer version, I today decided to do a clean fresh install. I flashed 2.4.2 to my ESP32-Board. So far the AP was created and Webgui is available.
Before I erased the old install I did a backup of the configuration. A short look into the .backup-file seems to be ok.
When I tried to restore the .backup-File, it seems to accept the file and reboots.
But the config was not imported, so everything seems to be in the same configuration as it was while uploading the .backup-File.
I tried around with a second ESP32 and witth different versions including v2.4.0. But the picture is always the same. After restoring the config, there is no config imported.

I can handover the backup-file to someone who has a deeper understanding of what happens, but I do not want to place the active rolling codes right here public.

As I am stuck in options I opened the issue here. Thank you in advance.

### How to reproduce it (step by step)

```bash
I think the problem may be reproduced with my .backup file.
```

DirkHBo, 2024-05-04:
My problem is solved. I'm not entirely sure if there's a problem or if it's intentional.
Since I wanted to try uploading the configuration from another operating system, I connected the EPS Somfy to my network via WiFi. When it was connected to the WiFi, the configuration was also available and everything worked fine (except that no security setting was set - as expected).
So the only question remains: Is it intended that no restored configuration will be used while it is in AP mode?

Thank you for this beautiful project!

PS: The upgrade from 2.4.0 to 2.4.2 worked perfectly again after a reinstallation.

rstrouse, 2024-05-04:
Backup and restore should work just fine in AP Mode.  However, it will not reboot until the network connection settings are applied.  This it intentional to allow onboard configuration changes to be made without having to reconnect to the AP.  The restore process reboots the ESP32.

There are also some rules regarding backup files.  First it does not include any connection information in the file.  This also includes MQTT broker configuration.  It also will not restore ethernet pin settings to from one board to the other.  The risk is just too great.  Some of these boards have no protection on the PHY ethernet adapter to the point of being destroyed with incorrect settings.

rstrouse, 2024-05-08:
I am going to close this for now.  If you have any further issues create a new issue.

closed 2024-05-08 by rstrouse