rstrouse/ESPSomfy-RTS #359: Can't find information of icons on the dashboard

luckyzor, 2024-05-04:
### Hardware

ESP32

### Firmware version

2.4.2

### Application version

2.4.2

### What happened? What did you expect to happen?

I can't find what are the icons on the shades I have:
[image]

### How to reproduce it (step by step)

```bash
go to the esp32 ip and check the dashboard
```

rstrouse, 2024-05-04:
Those icons are windy and sunny condition icons.  At some point you likely sent sunny and windy conditions using the virtual remote.

Open up the virtual remote tab, select an afflicted shade from the dropdown and send a sensor command without any of the checkboxes checked.
[image]

rstrouse, 2024-05-09:
@luckyzor did performing the above operation clear the windy and sunny flags from your shutters?

luckyzor, 2024-05-09:
Yes, sorry for the late reply, I went on vacations and totally forgot to test it.
It's cleared now, no more weird icons 😅
Tks