rstrouse/ESPSomfy-RTS #360: 2.4.3-Pre feedback

alka79, 2024-05-04:
Hello,
I played all day with 2.4.3pre.
I recompiled locally with the latest Arduino core 2.0.16.
Update via OTA went smoothly. As they might depend on each other, I never know which to upload first, littlefs or firmware ?  I uploaded App first.

2.4.3pre has been stable all day. I can see in the serial output that some network stuff is async now. The group's rollingcode is now in the backupfile, right after the bits length. I did not try to fully restore though.

The new feature is the MY Simulation for shades that don't have a MY memory. I have plenty of those old shutters and was eager to try that feature.
Set a My position and check the simMy option in the settings of the shades, hit the Shade's My button in the WebUI and the shade moves gently to the set My Position :)  

It does not work though with the groups. Hit the My button in the WebUI sends the My command to all paired members. Of course my old shutters ignore this. A nice improvement would be to check the group members and give a special treatment to those with MySim setting to move them one by one to their My position. 

My expectation was to be able to also use the physical remote My button with my shutters. Unfortunately it is not implemented. 
I hacked the code in `SomfyShade::processWaitingFrame()` to add a call to `moveToTarget` when a My command is received for a shade with simMy. That works ! it also works with all the linked remotes :)
I hope it can be officially implemented. I am afraid that my hack might have uncontrolled (by me) side effects.

rstrouse, 2024-05-04:
You should always install the firmware first.  That way if there are any dependencies on the script the supported elements will be available.

I have posted updates to the code to allow for the Simulated my functions to be operated via remote and using groups.  There is also a hardware watchdog now that will reboot the ESP32 if any process takes longer than 5 seconds and does not feed the dog.

alka79, 2024-05-04:
that was fast. I'll look into it and report back (and also look how you implemented. I am getting curious now :) )

> You should always install the firmware first. That way if there are any dependencies on the script the supported elements will be available.

It could well be reversed. The old app might not expect a change in the new firmware and hang and then I can not update via OTA anymore.
Not knowing and having the possibility to upload firmware with Arduion, I felt more secure to update App first. 
Now that I know, I'll just do firmware first.

> EDIT: And one more thing I had to move the rolling code for groups to the end of the file. This way those folks that downgrade can still use a newer backup file.

clever :)

alka79, 2024-05-06:
I keep reporting here. I installed 2.4.3 Pre second release. 
My on the Physical remote control works fine with all linked remotes :)

However some strange behaviour:
1. watchdog has reset twice in the night (3:44:33  and 3:45:38 PM) for no apparent reason.
Here is the  serial output.
Aslo there is a huge number of unexplained onvif/subscription requests starting from 3:46 PM after the second reset. Maybe it is the HA onvif integration ?  I reset manually the board and got a new IP Address. These onvif requests are now gone. Never got anything like that before. Strange.
<details>

```
00:06:53.037 -> Setting Last Rolling code 24823
00:06:53.083 -> Timing Somfy: 122ms
00:07:12.717 -> Timing Somfy: 460ms
03:44:33.022 -> E (54147413) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
03:44:33.022 -> E (54147413) task_wdt:  - loopTask (CPU 1)
03:44:33.071 -> E (54147413) task_wdt: Tasks currently running:
03:44:33.071 -> E (54147413) task_wdt: CPU 0: wifi
03:44:33.071 -> E (54147413) task_wdt: CPU 1: IDLE1
03:44:33.071 -> E (54147413) task_wdt: Aborting.
03:44:33.071 -> 
03:44:33.071 -> abort() was called at PC 0x4010dc14 on core 0
03:44:33.071 -> 
03:44:33.071 -> 
03:44:33.071 -> Backtrace: 0x40083b55:0x3ffbed6c |<-CORRUPTED
03:44:33.071 -> 
03:44:33.071 -> 
03:44:33.071 -> 
03:44:33.071 -> 
03:44:33.071 -> ELF file SHA256: 794fe227891e21cb
03:44:33.071 -> 
03:44:33.116 -> Rebooting...
03:44:33.116 -> ets Jul 29 2019 12:21:46
03:44:33.116 -> 
03:44:33.116 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
03:44:33.164 -> configsip: 0, SPIWP:0xee
03:44:33.164 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
03:44:33.164 -> mode:DIO, clock div:1
03:44:33.164 -> load:0x3fff0030,len:1184
03:44:33.164 -> load:0x40078000,len:13260
03:44:33.164 -> load:0x40080400,len:3028
03:44:33.164 -> entry 0x400805e4
03:44:33.681 -> 
03:44:33.681 -> Startup/Boot....
03:44:33.681 -> Mounting File System...
03:44:33.681 -> File system mounted successfully
03:44:33.681 -> Chip Model ESP32-
03:44:33.728 -> Preference IP Free Entries: 453
03:44:33.728 -> SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
03:44:33.728 -> Connection Type: 1
03:44:33.728 -> NTP Settings 
03:44:33.728 -> pool.ntp.org TZ:CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
03:44:33.728 -> WIFI Settings
03:44:33.728 ->  SSID: ***
03:44:33.728 -> 
03:44:33.728 -> Launching web server...
03:44:33.728 -> Creating Web MicroServices...
03:44:34.732 -> WiFi Mode: 0
03:44:34.777 -> WiFi interface ready
03:44:34.823 -> WiFi station mode started
03:44:38.016 -> Completed scan for access points
03:44:38.016 -> Scanned 26 Networks...
03:44:38.016 -> ***
03:44:38.153 -> App Version:2.4.3
03:44:38.153 -> shades.cfg exists so we are using that
03:44:38.153 -> Reading header at 0
03:44:38.153 -> version:24 len:76 roomSize:29 roomRecs:4 shadeSize:276 shadeRecs:15 groupSize:200 groupRecs: 4 pos:76
03:44:38.346 -> Applying radio settings Setting Data Pins RX:22 TX:21
03:44:38.346 -> Setting SPI Pins SCK:18 MISO:19 MOSI:23 CSN:5
03:44:38.346 -> Radio Pins Configured!
03:44:38.346 -> Successfully set up the radio
[... 140 lines cut ...]
15:30:29.511 -> Enabled receive on Pin #22 Timing: 1
15:30:29.616 -> Timing Somfy: 358ms

individual shade #3 My button : works fine
15:38:31.719 -> Sending Shade Command
15:38:31.719 -> Received:{"shadeId":3,"command":"my"}
15:38:31.719 -> Moving to My Position
15:38:31.719 -> Moving to simulated favorite
15:38:31.719 -> Moving to 89.00% from 100.00% using Up
15:38:31.817 -> CMD:Up ADDR:6003 RCODE:65 REPEAT:1
15:38:32.003 -> Enabled receive on Pin #22 Timing: 1
15:38:32.118 -> Timing WebServer: 322ms
15:38:34.687 -> CMD:My ADDR:6003 RCODE:66 REPEAT:1
15:38:34.968 -> Enabled receive on Pin #22 Timing: 1
15:38:35.077 -> Timing Somfy: 365ms

Up command to the group works fine
15:42:37.011 -> Sending Group Command
15:42:37.011 -> Received:{"groupId":4,"command":"up"}
15:42:37.107 -> CMD:Up ADDR:615821 RCODE:40 REPEAT:1
15:42:37.292 -> Enabled receive on Pin #22 Timing: 1
15:42:37.413 -> Timing WebServer: 322ms
15:42:59.205 -> Timing Somfy: 453ms
```
</details>

rstrouse, 2024-05-06:
> watchdog has reset twice in the night (3:44:33 and 3:45:38 PM) for no apparent reason.

Perhaps 5 seconds is not enough.  More likely though there is something that can potentially be long running that is not feeding the dog.  I'll look through it to see what I see.

> Individually, Shade3 reacts well. As member of a group the command to move to target to My postion are sent but the shutter does not move at all.

Increase the number of repeats to shade 3.  I believe some motors miss incoming frames when they are processing the last frame even if it is not for them.  During this time it probably misses the next preamble and does not begin decoding the frame.  Hence it never sees the command.  If you were using a remote you probably would not notice because repeats come very fast when you press the button.  It is not uncommon to get 2-3 repeats from pressing the button.

That being said I may try to add additional cooling off period after the last frame to see if this additional silence allows slower motors to keep up.

> Aslo there is a huge number of unexplained onvif/subscription requests starting from 3:46 PM after the second reset. 

I would bet that this is your onvif integration.   These probably started with SSDP responses but instead of calling the returned device profile url somewhere in that code it thinks that ESPSomfy RTS has a sensor or camera data that it wants.  The call is most likely asking for the UPnP description of a service.  

This is really why devices such as these should have an IP reservation so the router does not hand out new IPs when a referencing device has cached it.   The udp datagrams for ESPSomfy RTS do not contain the specified url and a 404.3 response is the correct response to the request.  

The calling device should have given up or rediscovered the devices.  Given the timing there would also have been quite a few ::alive udp packets that should have been picked up between the calls for both ESPSomfy RTS and perhaps the actual device it was processing discovery.

alka79, 2024-05-06:
I forgot to give my new board a fixed DHCP address. It's done. Maybe the IP Address received after the nightly reset by DHCP was previously assigned to my camera and HA somehow remembers. I should give fixed addresses to those fixed devices!

Increase repeats from 1 to 2 for shade3 helped. 'Group My' now works. I see that there is no delay at all when sending to shades in groups. My old shades probably not like the burst.

No watchdog issue during the day. The serial output is still running. I'll report if there is something.

alka79, 2024-05-08:
Hi again tonight, watchdog restart for no apparent reason.
Serial Output

<details>

```
01:44:12.781 -> E (123838730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
01:44:12.880 -> E (123838730) task_wdt:  - loopTask (CPU 1)
01:44:12.880 -> E (123838730) task_wdt: Tasks currently running:
01:44:12.880 -> E (123838730) task_wdt: CPU 0: IDLE0
01:44:12.880 -> E (123838730) task_wdt: CPU 1: IDLE1
01:44:12.880 -> E (123838730) task_wdt: Aborting.
01:44:12.880 -> 
01:44:12.880 -> abort() was called at PC 0x4010dc14 on core 0
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> Backtrace: 0x40083b55:0x3ffbed6c |<-CORRUPTED
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> ELF file SHA256: 794fe227891e21cb
01:44:12.880 -> 
01:44:12.937 -> Rebooting...
01:44:12.937 -> ets Jul 29 2019 12:21:46
01:44:12.987 -> 
01:44:12.987 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:44:12.987 -> configsip: 0, SPIWP:0xee
01:44:12.987 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:44:12.987 -> mode:DIO, clock div:1
01:44:12.987 -> load:0x3fff0030,len:1184
01:44:12.987 -> load:0x40078000,len:13260
01:44:12.987 -> load:0x40080400,len:3028
01:44:12.987 -> entry 0x400805e4
01:44:13.443 -> 
01:44:13.443 -> Startup/Boot....
01:44:13.443 -> Mounting File System...
01:44:13.443 -> File system mounted successfully
01:44:13.443 -> Chip Model ESP32-
01:44:13.543 -> Preference IP Free Entries: 453
01:44:13.543 -> SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
01:44:13.543 -> Connection Type: 1
01:44:13.543 -> NTP Settings 
01:44:13.592 -> pool.ntp.org TZ:CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
01:44:13.592 -> WIFI Settings
01:44:13.592 ->  ***
01:44:13.592 -> 
01:44:13.592 -> Launching web server...
01:44:13.592 -> Creating Web MicroServices...
01:44:14.529 -> WiFi Mode: 0
01:44:14.529 -> WiFi interface ready
01:44:14.622 -> WiFi station mode started
01:44:17.621 -> Completed scan for access points
01:44:17.621 -> Scanned 23 Networks...
01:44:17.621 ->***
01:44:17.786 -> App Version:2.4.3
01:44:17.786 -> shades.cfg exists so we are using that
01:44:17.786 -> Reading header at 0
01:44:17.786 -> version:24 len:76 roomSize:29 roomRecs:4 shadeSize:276 shadeRecs:15 groupSize:200 groupRecs: 4 pos:76
01:44:17.936 -> Applying radio settings Setting Data Pins RX:22 TX:21
01:44:17.936 -> Setting SPI Pins SCK:18 MISO:19 MOSI:23 CSN:5
01:44:17.936 -> Radio Pins Configured!
01:44:17.987 -> Successfully set up the radio
01:44:17.987 -> Enabled receive on Pin #22 Timing: 1
01:44:17.987 -> Initializing RX Queue
01:44:17.987 -> Connecting to AP
01:44:17.987 -> WiFi clients stopped
01:44:17.987 -> WiFi interface ready
01:44:17.987 -> WiFi station mode started
01:44:18.080 -> Set hostname to:ESPSomfyRTS
01:44:24.449 -> Completed scan for access points
01:44:24.449 -> Found strongest AP 7 A0:40:A0:A5:DC:D9
01:44:24.449 -> Timing Net: 6540ms
01:44:32.022 -> Disconnected from WiFi access point
01:44:39.443 -> Disconnected from WiFi access point
01:44:44.446 -> Fell into timeout
01:44:44.558 -> 
01:44:44.558 -> Turning the HotSpot On
01:44:44.558 -> WiFi clients stopped
01:44:44.558 -> WiFi interface ready
01:44:44.558 -> WiFi station mode started
01:44:44.558 -> WiFi AP Started
01:44:44.610 -> Initializing AP for credentials modification
01:44:44.610 -> 
01:44:44.610 -> SoftAP IP: WiFi AP Stopped
01:44:44.660 -> 192.168.4.1
01:44:44.660 -> WiFi AP Started
01:44:44.660 -> Socket Server Started...
01:44:44.660 -> Heap Max: 110580, Free: 178124, Min: 175892
01:44:44.660 -> ....................................................................................................
01:47:14.994 -> ....................
01:47:44.609 -> Stopping AP Mode
01:47:44.609 -> WiFi AP Stopped
01:47:44.609 -> Timing Net: 180121ms
01:47:44.609 -> Fell into timeout
01:47:44.609 -> WiFi AP Started
01:47:44.609 -> 
01:47:44.609 -> Turning the HotSpot On
01:47:44.609 -> WiFi AP Stopped
01:47:44.609 -> WiFi clients stopped
01:47:44.609 -> WiFi interface ready
01:47:44.609 -> WiFi station mode started
01:47:44.609 -> WiFi AP Started
01:47:44.710 -> Initializing AP for credentials modification
01:47:44.710 -> 
01:47:44.710 -> SoftAP IP: WiFi AP Stopped
01:47:44.710 -> 192.168.4.1
01:47:44.710 -> WiFi AP Started
01:47:44.710 -> Socket Server Started...
01:47:44.710 -> Heap Max: 110580, Free: 178308, Min: 175892
01:47:45.080 -> ....................................................................................................
01:50:15.488 -> ....................
01:50:44.717 -> Stopping AP Mode
01:50:44.717 -> WiFi AP Stopped
01:50:44.717 -> Timing Net: 180119ms
01:50:44.717 -> Fell into timeout
01:50:44.717 -> WiFi AP Started
01:50:44.717 -> 
01:50:44.717 -> Turning the HotSpot On
01:50:44.717 -> WiFi AP Stopped
01:50:44.717 -> WiFi clients stopped
01:50:44.717 -> WiFi interface ready
01:50:44.717 -> WiFi station mode started
01:50:44.717 -> WiFi AP Started
01:50:44.814 -> Initializing AP for credentials modification
01:50:44.814 -> 
01:50:44.814 -> SoftAP IP: WiFi AP Stopped
01:50:44.814 -> 192.168.4.1
01:50:44.814 -> WiFi AP Started
01:50:44.814 -> Socket Server Started...
01:50:44.814 -> Heap Max: 110580, Free: 178308, Min: 175892
01:50:45.564 -> ....................................................................................................
01:53:15.934 -> ....................
01:53:44.847 -> Stopping AP Mode
01:53:44.847 -> WiFi AP Stopped
01:53:44.847 -> Timing Net: 180120ms
01:53:44.847 -> Fell into timeoutWiFi AP Started
01:53:44.847 -> 
01:53:44.847 -> 
01:53:44.847 -> Turning the HotSpot On
01:53:44.847 -> WiFi AP Stopped
01:53:44.847 -> WiFi clients stopped
01:53:44.847 -> WiFi interface ready
01:53:44.847 -> WiFi station mode started
01:53:44.847 -> WiFi AP Started
01:53:44.963 -> Initializing AP for credentials modification

.... going on again and again

```

</details>

rstrouse, 2024-05-08:
After the reboot it appears to perform the correct operation for when the network is not allowing connections.  If you look at the two lines indicating disconnected from access point at `01:44:32.022` you will see that it was never issued an IP address from the AP.  ESPSomfy RTS will then open the SoftAP.

However, after the SoftAP times out after 3 minutes it thinks that it is still connecting from the previous attempt that aborted unceremoniously in between getting a valid AP and still not getting an IP address.   It looks like I did not clear the connecting flag on the first failure condition since I do not see it resetting the station mode.   It is simply opening and closing the SoftAP.

rstrouse, 2024-05-08:
Yep what I explained above is exactly the condition I did not expect.  It is weird that the SSID would partially connect but not issue an IP address and simply reject the connection on the first try but that could simply be a case where the router was rebooting and its DHCP services were not yet available.  I suspect that simply clearing the connecting flag when the SoftAP is opened will capture this condition completely.

The important bits about what triggered the watchdog timer is missing from your log.  This would have been the lines just above the watchdog trigger log entry (the first line).  Unfortunately the stack trace from will always point to the ESP32 library location in the core and will tell us nothing about what triggered it.  I have a suspicion that the trigger is a delay in response when checking whether there is internet available.

rstrouse, 2024-05-09:
I updated the 2.4.3 release to clear the connecting flag when the soft ap is started.  Please update the v2.4.3 firmware on your device.

rstrouse, 2024-05-11:
@alka79 did the v2.4.3 updates perform better overnight for you?

alka79, 2024-05-11:
> The important bits about what triggered the watchdog timer is missing from your log. This would have been the lines just above the watchdog trigger log entry (the first line). Unfortunately the [...]

There is actually nothing on the serial port just before the reset by watchdog
<details>

```
23:52:47.491 -> Timing Somfy: 267ms
23:52:47.589 -> Timing Somfy: 111ms
23:52:56.564 -> Setting Last Rolling code for 6911313 : 4724
23:52:56.564 -> Setting Last Rolling code for 6911313 : 4724
23:52:58.952 -> Timing Somfy: 511ms
01:44:12.781 -> E (123838730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
01:44:12.880 -> E (123838730) task_wdt:  - loopTask (CPU 1)
01:44:12.880 -> E (123838730) task_wdt: Tasks currently running:
01:44:12.880 -> E (123838730) task_wdt: CPU 0: IDLE0
01:44:12.880 -> E (123838730) task_wdt: CPU 1: IDLE1
01:44:12.880 -> E (123838730) task_wdt: Aborting.
01:44:12.781 -> E (123838730) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
01:44:12.880 -> E (123838730) task_wdt:  - loopTask (CPU 1)
01:44:12.880 -> E (123838730) task_wdt: Tasks currently running:
01:44:12.880 -> E (123838730) task_wdt: CPU 0: IDLE0
01:44:12.880 -> E (123838730) task_wdt: CPU 1: IDLE1
01:44:12.880 -> E (123838730) task_wdt: Aborting.
01:44:12.880 -> 
01:44:12.880 -> abort() was called at PC 0x4010dc14 on core 0
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> Backtrace: 0x40083b55:0x3ffbed6c |<-CORRUPTED
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> 
01:44:12.880 -> ELF file SHA256: 794fe227891e21cb
01:44:12.880 -> 
01:44:12.937 -> Rebooting...
```
</details>

I was out a few days. Just reinstalled latest 2.4.3 Pre. For a change, I installed via github.
Download/upload worked well. At the end of the process the board resets wich is normal behaviour.
Then just after there was a second reset due to the watchdog which seems abnormal to me.
Now, each time I try to connect from the web UI, the board resets.
I'll take sometime tomorrow to build locally and uplaod locally as I used to.

Here is the serial output of the three resets
<details>

```
20:08:59.911 -> Update.end Called...
20:08:59.911 -> Update SomfyController.littlefs.bin complete
20:08:59.911 -> End update SomfyController.littlefs.bin
20:09:00.006 -> Committing Configuration...
20:09:00.125 -> Timing WebServer: 64530ms
20:09:00.604 -> Disconnected from SSDP...
20:09:00.604 -> Socket [0] Disconnected!
20:09:00.747 -> Disconnected from WiFi access point
20:09:00.747 -> WiFi clients stopped
20:09:00.813 -> ets Jul 29 2019 12:21:46
20:09:00.813 -> 
20:09:00.813 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:09:00.813 -> configsip: 0, SPIWP:0xee
20:09:00.813 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:09:00.813 -> mode:DIO, clock div:1
20:09:00.813 -> load:0x3fff0030,len:1184
20:09:00.813 -> load:0x40078000,len:13260
20:09:00.813 -> load:0x40080400,len:3028
20:09:00.813 -> entry 0x400805e4
20:09:01.259 -> E (346) psram: PSRAM ID read error: 0xffffffff
20:09:01.382 -> 
20:09:01.382 -> Startup/Boot....
[... 232 lines cut ...]
20:15:14.167 -> Socket Server Started...
20:15:14.167 -> Max Heap: 110580
20:15:14.167 -> Free Heap: 174832
20:15:14.167 -> Min Heap: 170256
20:15:14.167 -> 
20:15:14.167 -> Successfully Connected to WiFi!!!!192.168.1.55 (-57dbm)
20:15:14.167 -> MDNS Responder Started: serverId=596BF4
20:15:14.218 -> SSDP: urn:schemas-rstrouse-org:device:ESPSomfyRTS:1 - true
20:15:14.218 -> Connected to SSDP...
20:15:14.218 -> Max Heap: 110580
20:15:14.218 -> Free Heap: 163252
20:15:14.218 -> Min Heap: 161604
20:15:55.503 -> Check github for updates...
20:15:55.503 -> Max Heap: 110580
20:15:55.599 -> Free Heap: 161840
20:15:55.599 -> Min Heap: 154200
20:15:57.289 -> Internet is Available: 1761ms
20:15:58.935 -> [HTTPS] GET... code: 200
20:15:58.935 -> [HTTPS] GET... code: 200 - -1
20:15:59.312 -> Max Heap: 110580
20:15:59.312 -> Free Heap: 158740
20:15:59.312 -> Min Heap: 106908
20:15:59.312 -> Timing WebServer: 3776ms
```
</details>

rstrouse, 2024-05-11:
I see that there are places that are taking 4 seconds to download a file from the file system.  I’ll look at that streaming function to see if there is a place to feed the dog.  Once it is cached it will be ok but each reboot got one more file and the index.js file was just too slow.

rstrouse, 2024-05-12:
Looking deeper at that log it does look like it had a corrupt fs.  See shade config file invalid.

alka79, 2024-05-12:
> Looking deeper at that log it does look like it had a corrupt fs. See shade config file invalid.

That is what I think as well. The /data upload via OTA went wrong for some unknown reason.
Since I uploaded manually via littlefs upload tool, it is working fine again. 

As per the watchdog : can you set it's food timing expectation to be variable ? during OTA or github update, 5sec is not suspicious.

rstrouse, 2024-05-12:
I moved the dog bowl so that it resets regardless of whether the data stream is suspended or not.  The process was already looking for stream timeouts.

rstrouse, 2024-05-12:
I increased the wdt reset to 7 seconds.  If it is bouncing now it will get fixed in the next release.  Please report any issues in a new issue.  Thanks for the help!

closed 2024-05-12 by rstrouse