rstrouse/ESPSomfy-RTS #362: Homekit plugin picture

AragornSRB, 2024-05-05:
Can someone share picture how this plugin looks in Homekit?Im using Hombridge-rpi-rts plugin and works ok but in homekit looks like switch.Is this one different?

[image]

[image]

rstrouse, 2024-05-07:
For me they show up as a cover entity.

rstrouse, 2024-05-08:
@AragornSRB I suspect that this is driven by the entity type so if it is a blind, shade, or shutter, it will show up as a a cover.  If it is a garage door then it will show up as a garage door.  And if it is a gate it will show up as a gate.   All of those include the current position of the entity.

It will never show up as a switch unless you define it as a dry contact.  ESPSomfy RTS fills in the proper entity classes in the plugin and I suspect the other plugin does not.

closed 2024-05-08 by AragornSRB