rstrouse/ESPSomfy-RTS #364: PIN Layout ESP32-S3

magtimmermans, 2024-05-07:
### Describe you new feature you'd like

Hi,

I am a bit confused with PINs can be used for ESP32-S3:
http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference

rstrouse, 2024-05-07:
You can use any of the pins in the Usable I/O pins but I would avoid any of them that have a note next to them.  For instance if it indicates that it is a strapping pin, or it has a pull up/down, or it is used for some other peripheral.  Each board maker tweaks this list to support the features they have on the board.

Here are the pins that I used on the mini https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build.  While GPIO4 has a weak pull up on reset it should not affect the operation of the RX pin in this case.   I used it because of the limited number of exposed pins on the mini.

magtimmermans, 2024-05-07:
Okay, Thanks. But maybe a stupid question, how does the software now the correct pins?

rstrouse, 2024-05-07:
On the configuration pages select the radio section then the transceiver tab.  From there you define the pins that you used for each of the functions of the transceiver.

[image]

closed 2024-05-07 by magtimmermans