rstrouse/ESPSomfy-RTS #365: issue firmware

YvesDC, 2024-05-08:
Hi

I use the following ESP :https://www.amazon.com.be/dp/B09PLBPBCC?ref=ppx_yo2ov_dt_b_product_details&th=1

and installed the firmware. I tried first with the S3 an later the other firmware 

Via web.esphome.io. 

installation is succesful,

but after restarting I never get a wifi AP?

Coudl you confirm the ESP is OK and which firmware I should use

rstrouse, 2024-05-08:
You should download the `SomfyController.onboard.esp32.bin.zip` file for that board and extract the `SomfyController.onboard.onboard.esp32.bin` file from it.  Then use this .bin file to onboard the firmware.   This contains the all of the files required to run the software.  You must extract the onboard file from the archived zip file.

That board is an early esp32 version which they designated as ESP32S but they dropped the designation later when they released the S2 and S3.

YvesDC, 2024-05-08:
I did not look very well and took the wrong SomfyController.ino.esp32.bin instead of the onboarding.
Thank you !
yves

rstrouse, 2024-05-08:
Now that you have the proper firmware I will close this for now.

closed 2024-05-08 by rstrouse