rstrouse/ESPSomfy-RTS #366: No boot on Seed Studio XIAO ESP32C3

black9404, 2024-05-11:
### Hardware

ESP32-C3

### Firmware version

V2.4.2

### Application version

V2.4.2

### What happened? What did you expect to happen?

I tried to install the latest firmware on a Seed Studio XIAO ESP32C3 but I won't show an AP after rebooting by removing the USB cable or pressing the onboard Reboot button. Even tried previous versions with the same problem... I double checked that I'm using the proper .bin.zip file

### How to reproduce it (step by step)

```bash
Download 2.4.2 firmware
Install on device
Fails to show an AP
```

rstrouse, 2024-05-11:
Please ensure you downloaded onboard .zip file and extract the contents.  Use that onboard .bin file to flash the first time.

black9404, 2024-05-11:
> Please ensure you downloaded onboard .zip file and extract the contents. Use that onboard .bin file to flash the first time.

I managed to eventually flash the device although I was using the correct onboard .bin.
The only way I was able to finally get it to boot and show the AP was by flashing it with the ESP32 Flash Download tool and with firmware V2.4.1 and clicking the reboot button on the device after it being flashed

black9404, 2024-06-20:
After a lot of experimenting I was able to fix an inconsistent boot problem with the Seed Studio ESP32C3 that happened after I flashed the firmware. The problem was related to the pinout I initially chose which was 

Pin | Color | ESP32 | Description
-- | -- | -- | --
1 | Black | GND | Ground
2 | Red | 3v3 | VCC
3 | Green | GPIO4 | GDO0 - TX Pin
4 | Blue | GPIO7 | CSN
5 | Orange | GPIO8 | SCK
6 | Purple | GPIO10 | MOSI
7 | White | GPIO9 | MISO
8 | Yellow | GPIO5 | GDO2 - RX Pin

This was causing the device to not show an AP after being successfully flashed most of the times (sometimes the AP would appear and I would be able to connect and do everything including getting the RF signals but after a reboot the device would stop working). So to fix the issue I reused the pinout from the wiki for the Seed Studio ESP32S3

Pin | Color | ESP32 | Description
-- | -- | -- | --
1 | Black | GND | Ground
2 | Red | 3v3 | VCC
3 | Green | GPIO3 | GDO0 - TX Pin
4 | Blue | GPIO6 | CSN
5 | Orange | GPIO7 | SCK
6 | Purple | GPIO9 | MOSI
7 | White | GPIO8 | MISO
8 | Yellow | GPIO4 | GDO2 - RX Pin

rstrouse, 2024-06-27:
I think there may be something goofy with GPIO5 on those boards.  It uses a separate ADC which may need to be pulled in one direction or the other during boot for reliable operation.

rstrouse, 2024-06-30:
I am going to close this but Seeed indicates in their documentation that their entire line of XIAO ESP32 boards are pin compatible.  The don't actually call out anything special wrt pin 5 but we will go with what works.

closed 2024-06-30 by rstrouse