rstrouse/ESPSomfy-RTS #368: strange behaviour

yvesdecloedt, 2024-05-13:
Hi,

I have set up the system and have paired 3 blinds
I use the smooth Somphy for my shutters
The blinds are metallic slates on wires which decend out of a box and they can tilt (as described in your document as integrated) 

setup configurtation
Type : Blind
Up time 30000ms 
Down time 30000ms
Tilt type Integrated 
Tilt Time 1500ms

when using the virtual remote :

If I continuously pres up, shutter briefly moves up and stops
If I continuously press down, it goes down while I am pressing;

but sometimes it's the otherway around and sometime I have this behavior in both directions

If my tilt is completly closed, then the My button does nothing. I first need to press the UP and then the My works

On my remote if you press My it goes to the predefined position, if you press it again it stops. (using the software 'if' the My works then pressing it again does nothing)

I tried to play with the shutters in the Home tab (using target position an target tilt) but this has even more erratic behavior in my case

Probably misconfigured something, but not sure what else i can try ?

rstrouse, 2024-05-13:
It sounds a lot like you have the motor configured for Euro-Tilt mode.  However, there is really no command sequence that I have found in the protocol for continuous press of the buttons.   It breaks this out into short press, medium press, and long press.  The up and down buttons are simply a repeat until enough of them are received to call it a long press.  ESPSomfy RTS has a Euro-Tilt setting.

Also, if you are using the same address as your remote for the ESPSomfy RTS configuration you will have strange issues where commands are first ignored (due to out of sequence rolling codes) and then can be activated.   If this is the case then eventually neither the remote or ESPSomfy RTS will continue to function.  At that point you will need to factory reset the motor and start over.

Some motors also have a pre-provisioned mode to allow you to set the endstops.  However, these are typically done with a series of short presses. 

As an aside the virtual remote tab was not designed to be the operational control for your blinds.  It is there to emulate a remote so that if you do not actually have one you can get the provisioning accomplished.  You need to verify whether you are using the Euro-Tilt setting on the motor, ensure you are not using the same address for ESPSomfy RTS as the remote, and time the movements for tilt and up/down.  These will likely be different as gravity comes into play.

yvesdecloedt, 2024-05-13:
The address on the remote is 9911904 (in the log)
[image]
The adress on the ESP Somfu RTS is 649036
The Tilt is set to integrated:
[image]
timings have been set approximately correct 30 seonds down to up (or vice versa) and the Tilt takes 1,5 seconds to complete

If I press the button UP on my remote just once it does the same thing (small movement). If I continously push the button, say for 1-2 seconds, ths shutter moves for 1-2 seconds (in the log I see a series of up commands passing :
[image]
If I push for approx 3 seconds and release it goes up without stopping (or unless I Press My or down)
[Output log.pdf](https://github.com/rstrouse/ESPSomfy-RTS/files/15296978/Output.log.pdf)

PS the target and tilt position does not do what it needs to do either. The animation shows it correctly but regardsless of the setting the shutter goes completely down (although it shows 0% )
[image]

yvesdecloedt, 2024-05-13:
I just had one finding and comment to avoid confusion;

when I mentioned virtual remote I meant the ESPSomphy RTS ((Home) as opposed to Smoove Somphy (physical remote)

What is strange is that the behavior on the ESPSomphy home is different then on the virtual remote tab.
The behavior is described is when pushing the button on the home tab.
When I push he up button on the virtual remote tab continously, then the shutter behaves as expected (moving for the duration of the push and after 3 seconds if continues)

rstrouse, 2024-05-13:
That sounds like the motor is configured for Euro-Tilt.  Either select that as your tilt type or even better look at the manual to turn that off.  On the main home screen repeats are not processed while holding down the button.  This is intentional to avoid long presses when not required.

yvesdecloedt, 2024-05-15:
The Euro-tilt does work somewhat better. (not sure what Euro-tilt stands for). On the ESPSomphy pressing once up, shiutter move up all the way. Pressing once down, moves down all the way. Settinh target position also seems to work. Target Tilt does not (only a very short movement)
The smoove remote and ESPSomph are out out of sync. After pressing remote down once, the shutter moves down a bit but the ESPSompy thinks the shutter is going down completely.
I added a video of the shutter:

https://github.com/rstrouse/ESPSomfy-RTS/assets/26707140/24de1b9d-2eca-44db-8b5a-0d2e30fe7bf5

rstrouse, 2024-05-15:
Honestly, I would change the mode on that motor.  To get a target position for the tilt will be near impossible because of the long press processing and the short timing on the tilt duration it will never be accurate.  The entire timing for long presses to start and stop the movement will be too close to the timing of the tilt.

yvesdecloedt, 2024-05-15:
ok I understand. Unfortunately these shutters are outside and I don't have access, nor any manual on how this could be done

rstrouse, 2024-05-15:
Look online.  Even if they are not Somfy branded they will likely follow the same power off then remote button combinations as Somfy to turn it on and off.

rstrouse, 2024-06-15:
@yvesdecloedt did you get this sorted?   If these are Somfy branded motors the way to change the mode on the motor is to move them half way with the remote.  Then hold down the up and down buttons until the motor jogs.

yvesdecloedt, 2024-06-17:
Hi,
For now I am just using it to open and close. I'll plan on experimenting when i have some time.
thank you for the tip

rstrouse, 2024-06-30:
I am going to close this for now.  If you have further questions re-open this issue.

closed 2024-06-30 by rstrouse