rstrouse/ESPSomfy-RTS #370: Backup Restore not working

alka79, 2024-05-13:
### Hardware

ESP32

### Firmware version

2.4.3

### Application version

2.4.3

### What happened? What did you expect to happen?

I updated to release 2.4.3 stable via github. Update went well.

```
19:32:52.700 -> App Version:2.4.3
19:32:52.700 -> shades.cfg exists so we are using that
19:32:52.700 -> Reading header at 0
19:32:52.700 -> version:24 len:76 roomSize:29 roomRecs:4 shadeSize:276 shadeRecs:15 groupSize:200 groupRecs: 4 pos:76
19:32:52.887 -> Applying radio settings Setting Data Pins RX:22 TX:21
```
Stupid me!, I then  restored from the backup file created just before update from 2.4.3-pre3. Restore failed. 

I tried with other backup files, back to 2.4.2 stable : always the same error in the serial port:

```
19:55:04.627 -> Restore: ESPSomfyRTS 2024-05-13T19_31_39.backup
19:55:04.723 -> {"shades":true,"repeaters":false,"settings":false,"network":false,"mqtt":false,"transceiver":false}
19:55:04.723 -> Reading header at 0
19:55:04.769 -> version:1 len:0 roomSize:0 roomRecs:0 shadeSize:0 shadeRecs:0 groupSize:0 groupRecs: 0 pos:0
19:55:04.769 -> Invalid Shade Record Size:0
19:55:04.769 -> Shade restore file invalid!
19:55:04.769 -> Rebooting ESP for restored settings...

```

and later after reset
```
...
19:55:10.966 -> App Version:2.4.3
19:55:10.966 -> shades.cfg exists so we are using that
19:55:10.966 -> Reading header at 0
19:55:10.966 -> version:1 len:0 roomSize:0 roomRecs:0 shadeSize:0 shadeRecs:0 groupSize:0 groupRecs: 0 pos:0
19:55:10.966 -> Invalid Shade Record Size:0
19:55:10.966 -> Shade config file invalid!
19:55:11.013 -> Applying radio settings Setting Data Pins RX:22 TX:21
...
```
All my Somfy settings are gone :-(

Attached is the latest backup file. I don't see anything wrong. Is there something I can fix by editing the file ?

[ESPSomfyRTS 2024-05-13T19_31_39.backup.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/15298573/ESPSomfyRTS.2024-05-13T19_31_39.backup.zip)

### How to reproduce it (step by step)

```bash
see above
```

rstrouse, 2024-05-13:
Interesting I just unzipped your file and restored it to two different ESP32 devices and it worked just fine.  Be sure you are not editing that file before uploading it.  Some text editors change the line endings and do other goofy things.

[image]

alka79, 2024-05-13:
????
I did not edit any of the backup files.
I'll install 2.4.3 and data locally instead of github and restore again.

rstrouse, 2024-05-13:
Are you recompiling the firmware?  If so which board are you selecting in the IDE?

alka79, 2024-05-13:
I did not recompile, updated from 2.4.3pre to 2.4.3 final via github update.

I just downloaded source and uploled locally firmware an litlefls (no recompile). Then restored my backup and it is all back again :-)

I have no clue what happen and why update via github gave a different result.

rstrouse, 2024-05-13:
I'll try to replicate this by downgrading/upgrading and see where it ends up.  The ESP32 filesystem (littlefs) does not give any error if the file write fails.  It could be that all you needed to do is re-install v2.4.3 from the github update and restore your file.  I turn everything off but the update process when downloading the fs but there really is no way to verify a successful write when it is done.

Did you have to use this method in the past?

rstrouse, 2024-05-14:
I tried this several times and was not able to duplicate it.  The only thing I can think of is that the write to littlefs failed.   Unfortunately, there is no way to read back bytes that were just written to verify them when writing filesystem blocks.  Prior to the update to Arduino core 2.016 there was a PR that fixed the writing of the fs buffers in some circumstances where writing OTA.

Perhaps this is what you experienced.  I am going to close this for now but keep it in the back of my mind.

closed 2024-05-14 by rstrouse

alka79, 2024-05-14:
> Did you have to use this method in the past?
only two attempts to update via github, first was 2.4.3pre. Both failed. Not a the same stage.
First time the webUi could not load  .js file.

In the future I will always download first and update local OTA. More reliable.