rstrouse/ESPSomfy-RTS #371: Informative question related to Somfy remote

TibsD, 2024-05-14:
### Hardware

ESP32

### Firmware version

v 2.4.2

### Application version

v 2.4.2

### What happened? What did you expect to happen?

Not really a bug in de ESPSomfy-RTS programming but a strange thing I noticed with my Somfy remote.
After I added all shutters to the Telis 16 remote as individual shutters and also making some groups on the remote I added them to ESPSomfy aswell. From ESP side everything works perfectly but I have something strange on the Somfy remote.

When I want to use a group all shutters respond, but when I want to use a single shutter it does not respond.

Any ideas?

The people from Somfy don't really know what to do.

### How to reproduce it (step by step)

```bash
Not relevant.
```

rstrouse, 2024-05-14:
Did you assign the same address to the ESPSomfy RTS shade and group definitions.  If you did, then that is the problem.  The rolling codes for the remote are out of sync.

TibsD, 2024-05-14:
That's right. When I set up the ESPSomfy I used the address that was shown in the log when I pressed a button on the remote. How can I modify this? Can I do this while everything is configured or do I start over completely?

rstrouse, 2024-05-14:
You can assign the address to any address that is not in use then pair them using the pairing process described in the documentation.  There is a ton of information in the wiki.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#pairing-a-shademotor

TibsD, 2024-05-15:
And how can I re-pair the Somfy remote? Will it work when I press the prof button on the remote in ESPSomfy and the the prog button on the original Somfy remote? Then delete the shutter in ESPSomfy and re-pair with another address?

rstrouse, 2024-05-15:
If you press the remote several times the rolling code may catch up if you did not cycle it too many times.   If it does not catch up you will need to factory reset the motor.

TibsD, 2024-05-15:
But I can still operate the shutter with the remote if I use a group. Do these groups have a different rolling code then? I only rarely use the ESPSomfy remote so the rolling code can only be off by a few clicks. Do they need to be exact or do they have a margin?

rstrouse, 2024-05-15:
> But I can still operate the shutter with the remote if I use a group.
Yes.  Please review the documentation in the wiki on grouping and linking remotes.

> Do they need to be exact or do they have a margin?
That margin is up to you.  If you are ok with the position being off by a couple of percent then that is your margin if you need it to be down to the mm then spend the time to get an accurate timing.

TibsD, 2024-05-18:
So I managed to set up the remote again and deleted all the shades in ESPSomfy.

I will no add them again, with a unique address this time.
Is it OK if I take the address I see in the logs and add 1000 to it? For example the shutter address on the remote is "3049705" and I would take the address in ESPSomfy as "3049805".
Is this OK or must there be a minimum difference between the 2?

Maybe a stupid question but I want to be sure I don't end up in the same situation as before...

rstrouse, 2024-05-18:
Any address so long as it is unique in your environment will work fine.  ESPSomfy RTS generates unique addresses based on the Chip ID of the ESP32.  Under normal circumstances it would be extremely rare that the generated address would conflict.  If the generated address does not match your remote then just let ESPSomfy RTS fill in the address for you.

TibsD, 2024-05-18:
Perfect, it worked with the generated addresses.

You can close this topic if you want.

rstrouse, 2024-05-18:
Awesome glad you got it working!

closed 2024-05-18 by rstrouse