rstrouse/ESPSomfy-RTS #374: Docker port

cook1emr, 2024-05-16:
### Describe you new feature you'd like

Has anyone looked at porting this firmware over to docker? These USB C1011 dongle maybe good hardware to allow this
https://a.aliexpress.com/_Ex9a4yj

rstrouse, 2024-05-16:
I am pretty sure that would take a bit more than just porting the firmware to docker.   ESPSomfy RTS communicates with the transceiver over SPI and not over RS232.  There would need to be some driver work done to bitbang on RS232.  In the end you would be back to an MCU.

I guess what I am saying is that it is possible but I don't see a need for it.

cook1emr, 2024-05-16:
Thank you for the response, I truly appreciate it.

In terms of use case I think projects like zigbee2mqtt or zwavejs are good examples. Being able to attach radios to a single machine runing a docker stack acting as a hub can offer simplicity and flexibility.

Would this SPI project be useful to create a USB to spi bridge https://pypi.org/project/spi-adapter/

rstrouse, 2024-05-26:
I am going to close this as I don't plan to create a docker port.

closed 2024-05-26 by rstrouse