rstrouse/ESPSomfy-RTS #376: Platform espsomfy_rts does not generate unique IDs. since Update of HA to 2024.5.2

fft1010, 2024-05-18:
### Hardware

ESP32

### Firmware version

v2.4.3

### Application version

v2.4.3

### What happened? What did you expect to happen?

Since the Update to 2024.5.4 I see the below errors in the logfiles. 
I did not chance anything :-) 
Everything works also as expected. No idea, how to get rid of the errors.

I see all my covers and group definitions plus the ESPSomfy definitions.

### How to reproduce it (step by step)

```bash
..
```

### Logs

```bash
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_1 already exists - ignoring cover.l1
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_2 already exists - ignoring cover.l2
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_3 already exists - ignoring cover.l3
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_4 already exists - ignoring cover.l4
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_5 already exists - ignoring cover.r1
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_6 already exists - ignoring cover.r2
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_7 already exists - ignoring cover.r3
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_8 already exists - ignoring cover.r4
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_group1 already exists - ignoring cover.rolladen_auf
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.cover] Platform espsomfy_rts does not generate unique IDs. ID espsomfy_D0AB04_group2 already exists - ignoring cover.7_hoch
2024-05-17 17:30:28.701 ERROR (MainThread) [homeassistant.components.update] Platform espsomfy_rts does not generate unique IDs. ID update_espsomfy_D0AB04 already exists - ignoring update.firmware_update
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID host_name_espsomfy_D0AB04 already exists - ignoring sensor.hostname
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID chip_model_espsomfy_D0AB04 already exists - ignoring sensor.chip_type
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID conn_type_espsomfy_D0AB04 already exists - ignoring sensor.connection
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID wifi_sensor_espsomfy_D0AB04 already exists - ignoring sensor.wifi_strength
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID wifi_ssid_espsomfy_D0AB04 already exists - ignoring sensor.wifi_ssid
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID wifi_channel_espsomfy_D0AB04 already exists - ignoring sensor.wifi_channel
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID free_memory_espsomfy_D0AB04 already exists - ignoring sensor.free_memory
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID max_memory_espsomfy_D0AB04 already exists - ignoring sensor.max_memory
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID min_memory_espsomfy_D0AB04 already exists - ignoring sensor.min_memory
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.sensor] Platform espsomfy_rts does not generate unique IDs. ID ip_addresss_espsomfy_D0AB04 already exists - ignoring sensor.ip_address
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.button] Platform espsomfy_rts does not generate unique IDs. ID reboot_espsomfy_D0AB04 already exists - ignoring button.reboot_esp_device
2024-05-17 17:30:28.702 ERROR (MainThread) [homeassistant.components.button] Platform espsomfy_rts does not generate unique IDs. ID backup_espsomfy_D0AB04 already exists - ignoring button.backup_espsomfy_rts
```

rstrouse, 2024-05-18:
The HA update is unrelated.  This has to do with zero conf discovery.   Make sure you do not have two discovered devices.

rstrouse, 2024-06-15:
I am going to close this for now.  I believe you may have had more than one device configured in HA.

closed 2024-06-15 by rstrouse