rstrouse/ESPSomfy-RTS #378: Local UI / Touchscreen support

Freddoyea, 2024-05-20:
### Describe you new feature you'd like

I LOVE this project, and was so sad to see that it only has the capability to be accessed through a client on the network. I don't have any programming knowledgde, nor have i never flashed an esp32, but i wish that it could run locally, through a touch screen connected to the esp32, as my interest is to have a touch-screen control mounted in my house, rather than having to use my phone or some other client to access the UI.

rstrouse, 2024-05-20:
I hear you.  However, the touch screens out there for ESP32 are pretty poor beyond displaying some low res information and a few touch areas.   

You are much better off and more likely to be successful using a low cost Fire, Andriod, Kindle, or retired phone to do what you want.  They cost about the same, are way more functional, and require very little tinkering.  These displays have way more engineering in them and you won't find yourself wishing you could lose weight on your fingertips.

Freddoyea, 2024-05-21:
I understand your purpose, and admire all your work. It would also be a lot of work to make a new version for this purpose, as i guess you'd have to rebuild the html site using libraries/packages for the arduino touchscreens, which seems terribly complicated from my research. 

However i think i settled on a solution which will satisfy my need, and also let me keep the option of accessing the UI from a phone, pc, etc. 

I don't really care too much about resolution, as long as it's not utterly crap (lol), so my idea is to built and 3D Print a wall mounted enclosure for a 4 inch touch screen, that's connected to a Raspberry pi, which will act as a "permanent" client. 

It'll run raspbian lite, and then upon boot, it'll auto open a chromium/browser window, which auto-connects to the static IP of the ESP32. 

Do you think this will work? Obviously you can't speak for the raspbi, but is there anything in the ESP32 preventing it from always having a client connected?

Thanks!

rstrouse, 2024-05-21:
> Do you think this will work? Obviously you can't speak for the raspbi, but is there anything in the ESP32 preventing it from always having a client connected?

Actually this will work just fine.   And yes the UI supports chromium on the Pi which will maintain a socket connection to the ESP32.  If you want to DIY an interface the index.js contains the socket management, connection, and all the bits needed to do what you want.  You will find the entire UI in the data directory in the repository.

rstrouse, 2024-06-15:
I am going to close this one for now.  If you have any issues let me know.  You should be able to have any modern browser connected to ESPSomfy RTS at all times.  It also works when the browser sleeps and needs to reestablish connection on waking.

closed 2024-06-15 by rstrouse