rstrouse/ESPSomfy-RTS #379: Unable to connect to the interface

lamen, 2024-05-20:
### Hardware

ESP32

### Firmware version

v2.4.3

### Application version

?

### What happened? What did you expect to happen?

For around 10 days I have no longer been able to access espsomfy, when I try to connect to the address I have infinite loading.
I tried on 3 different esp32s (Aliexpress wroom32, Azdelivery wroom32), with version 2.4.3 and 2.4.2
[image]

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-05-20:
Clear your browser cache.  Since you are using Chrome, hold down the ctrl key and press F5.

lamen, 2024-05-21:
Thanks for your response,
i have already tried Ctrl+F5 and Ctrl+R, same result

lamen, 2024-05-21:
I have tried in a different place (other wifi network), it's works well.
At home it seems that the get networksettings return null

any idea ?

On the other wifi network :
`{
    "get": "/networksettings",
    "obj": {
        "ssdpBroadcast": true,
        "hostname": "ESPSomfyRTS",
        "connType": 1,
        "chipModel": "",
        "checkForUpdate": true,
        "fwVersion": "v2.4.3",
        "ethernet": {
            "boardType": 0,
            "phyAddress": 0,
            "CLKMode": 0,
            "phyType": 0,
            "PWRPin": -1,
            "MDCPin": 23,
            "MDIOPin": 18
        },
        "wifi": {
            "ssid": "Livebox-2A90",
            "passphrase": "XXXX",
            "roaming": true
        },
        "ip": {
            "dhcp": true,
            "ip": "192.168.1.16",
            "gateway": "192.168.1.1",
            "subnet": "255.255.255.0",
            "dns1": "192.168.1.1",
            "dns2": "0.0.0.0"
        }
    }
}`

At home :
`{"get": "/networksettings","obj": null }`

rstrouse, 2024-05-21:
Out of curiosity are you using a hidden SSID?

Open up a browser and type `http://<ip address>/networksettings` and lets see if if returns something other than json.

lamen, 2024-05-21:
ok i got it, i've a " on my wifi password :
[image]

rstrouse, 2024-05-26:
I am going to close this one out.  I will look at some ways to scrub the json but escaping all of the json output is probably too expensive in terms of memory since the buffer will need to forward look through all the strings.  For now don't include quotes in your passphrase.

closed 2024-05-26 by rstrouse