rstrouse/ESPSomfy-RTS #381: Which bin file for ESP32-C6 WROOM-N8

menaibluey, 2024-05-22:
### Describe you new feature you'd like

What is the recommended firmware bin file to flash a ESP32-C6 version? Thanks.

rstrouse, 2024-05-22:
I have not built firmware for the C6 yet.  Afik there is no Arduino core in release for it yet.

menaibluey, 2024-05-22:
Ok thanks. Do you think that any of the other bin files will work. I don’t intend to use any of the zigbee or thread capabilities. Sent from my iPhone

rstrouse, 2024-05-23:
You can try the C3 firmware but I think the Wi-Fi libraries changed.

bpat227, 2024-06-08:
I tried the C3 firmware but my C6 doesn't boot:
```
Invalid chip id. Expected 13 read 5. Bootloader for wrong chip?
```
If I try to compile it with Arduino 1.8.19 on my M1 Mac it fails with many errors, at least with 
```
In file included from SomfyController/Utils.cpp:3:
Utils.h:16:30: error: 'String' in namespace 'std' does not name a type
   16 |   template <typename T> std::String to_string(const T& t) {
      |                              ^~~~~~
```
Do you have any ideas?
I'm completly new to this environments.

rstrouse, 2024-06-10:
It appears the Arduino Core v3.x was released last week.  One of the things that 3.x includes is support for the C6 generation of chips.  It will take a bit to get this transition over from v2.x since there are some breaking changes.

rstrouse, 2024-06-15:
Unfortunately, the Arduino Core v3.x is 200k bigger than previous versions.  This will require a major code rework to which I have already started.  It will not fit on the ESP32 without removing features or repartitioning.  In the meantime this means that there is no support for the C6 chips yet.

The compile error you are getting is also because the std namespaces have been removed.  I suspect that this has something to do with trying to put the Core on a diet.

bpat227, 2024-06-16:
Hi,
in the meantime I bought another ESP32-S3 board and it works without any issues.
If you support the C6 chips, I will take this one as a backup device (cold stand by).
Many thanks for your support!!

rstrouse, 2024-06-30:
I am going to close this for now as support for the C6 is going to take some longer work.  The Core 3.x is so very big right now and doesn't leave much for programs.

closed 2024-06-30 by rstrouse