rstrouse/ESPSomfy-RTS #382: Hi, thaks for your project and effort.

gralfj, 2024-05-24:
Hi thaks for your project.

I,ve been dealimg with It for a month or so.

First i made a mini device wich worked well, i was able to solder anything on place and connect four roller shades, by the way, IS this what you call a roller shade?, 

[image]

And a "toldo", for wich i selected roller shade too, what do you call this?

[image]

Well, It worked for some days until I had to park the project for some time due to reasons.

Yesterday i was coming back to the project and the device seemee to not work.

No way to connect to It, so, after strugling with no result i ended Up resetimg and starting from Scratch.

Now I encountrr myself that trying to pai the shade leds me to , i have four, #4 working fine, but #3 makes #3 and #1 to move, same on #2 , and #1 works fine.

A bit confused here, does It give you any idea what IS going on?

Thanks in avance.

rstrouse, 2024-05-24:
> First i made a mini device wich worked well, i was able to solder anything in place and connect four roller shades, by the way, IS this what you call a roller shade?,

If the slats are connected together so that they do not tilt open to allow light between them then, yes I would consider that to be a roller shade.   If you can position them so that light can enter between them then they are considered a blind.

> A bit confused here, does It give you any idea what IS going on?

Yes.  What is happening is that you have the shade 1 paired to the address on shade 3 as well as shade 1.   This is very easy to fix but you should know how this happens.  

When you open the memory on the motor by long pressing the remote until the motor jogs, it puts the motor into a mode where it will be listening for the prog command from any remotes.  For some motors, it closes the memory and stops listing for and other remotes.  For others you must press the prog button on the original remote to have it stop listening for new remotes or wait until it times out.

For those motors that do not stop listening until you press the prog button on the original channel to stop listening for prog commands, you will often see the shade jog again when you press the prog button again on the original remote.  This indicates that it is no longer listening for new remotes.  Otherwise, when you go to pair the next shade the one that you just paired will also hear the prog command for the next shade.  So the the second remote will now control shade 1 and shade 2.

Fixing this is a matter of opening the memory with a long press on shade 1 then sending another prog command from shade to to unpair it from the motor.  You can do this using the original remote and the ESPSomfy virtual remote.  Then don't forget to close the memory on the original remote as indicated by a 3rd up/down movement on shade 1.

gralfj, 2024-05-24:
Hi, thanks for your quick reply.

Just to clarify

1 press back button on somfy remote
2 wait for jog
3 press back button on somfy remote
4 unpair

I guess that i have to repeat this and swap step 4 to pair it correctly.

And that's it?

By the way, what do you call the "toldo" of second picture?

rstrouse, 2024-05-24:
> Just to clarify

Not quite

1. Press the back (prog) button on the Somfy remote
2. wait for jog
3. Press the prog button on the virtual remote in ESPSomfy RTS for the shade that should not be moving the motor
4. Short press the back (prog) button on the Somfy remote

> I guess that i have to repeat this and swap step 4 to pair it correctly.

Yes.  Remember every motor hears every command.  Each motor decides what to do with that command based on whether it has the address of the remote in its memory.

> By the way, what do you call the "toldo" of second picture?

I believe that toldo is an awning.  I English an awning is an exterior roof or cover structure attached to the outside of a building.  In this case your toldo is a retractable awning to be specific.

gralfj, 2024-05-24:
WoW what a Messi, now even my original remote acts weird. I think ill have to reset everything. Ill come back later to explain what i did to solve It, if i am able 😅

rstrouse, 2024-05-24:
You will get it.  The problem is that pairing and unpairing a remote is the same command and the same sequence.  If it is paired it will unpair.  If it is not paired then it will pair.  Sometimes the right thing to do is simply factory reset the shade and do each one, one at a time, where only the current motor you are working on has power.  This is actually the way they describe it in the Somfy documentation.  It will eliminate the possibility of selecting the wrong channel or getting the more than one motor to respond to the pairing.

Also, do not use any address from the original remote for an address assigned in ESPSomfy RTS.  If you looked in the log and captured the address then entered it into any shade definition then you are doing it wrong.  There are projects that recommend this but it will create issues with ESPSomfy RTS and for the remote.  

In the end, your remotes and ESPSomfy RTS should work individually to operate your shades.  While ESPSomfy RTS reliably keeps track their position.  Regardless of how they were commanded.  When properly configured it will just work.

gralfj, 2024-05-24:
Hi again.

Done!!!

Now I am an expert hehehehehehehe

It took me some time to realize how actually remote and motor "comunicate". now both are working correcrly.

Even though it works fine there is something I would like to fine tune.

When my shutters go down to fully closed, there is a bit of time the motor is still working thought the shutter is fully closed, we managed to program them like this for the shutter slats to put weight on each other and close totally the, sometimes, annoying light entering through then.

This turns out in a mismatch between the actual opening and that show by the system.

I have taken the time it takes the shutter to fully open and viceversa from 

1 button press
2 first gap opening

And neither of them solved the problem.

Any advice?

gralfj, 2024-05-24:
Sorry, dont understand what you mean, but take a look, this IS similar tom what i have

[https://youtu.be/TAfZ7Vkrq3I?si=NCy-vldFvZfR56wy](URL)

rstrouse, 2024-05-24:
> this IS similar tom what i have

Ok your mechanism is how I thought you described it and not what would be a venetian blind with variable slats.

Did you set the endstops on the motor?  If so 100% should equal the shade completely closed + the time required to let the weight off the slats.  If you press the down button and it closes all the way including the motion to let the slats settle, then the timing you set on the down time should include this time.  ESPSomfy RTS does not issue a stop command when the down button is used.  That is up to the motor to determine that.

gralfj, 2024-05-24:
So from 100% opened i just hit down at the same time as i start taking time and wait to the motor to stop, isn't It?

rstrouse, 2024-05-24:
Yes.  When the motor stops that is what ESPSomfy RTS views as 100%.

gralfj, 2024-05-24:
By the way, I have a question , another hehe, I just integrated in HA, nice translation 😉.

After adding It to HA, i added the awning to rst, but It does not appear in HA, is there anyway to add It? Or do i have to deletr and integrate again?

rstrouse, 2024-05-24:
Click on the integration then choose reload from the menu.

gralfj, 2024-05-24:
Nice, everything fine, just another question, seems every shade has to be on the same área, and my awning IS on a different one.

rstrouse, 2024-05-24:
You should be able to set any device to any area in HA.  Click on the gears icon on the top right of the entity and assign it wherever you like.

You can also define up to 16 rooms in ESPSomfy RTS then set it in the configuration for the particular shade.

gralfj, 2024-05-25:
well, I am back, can´t asign toldo to other room than the assigned to the integration.

If i enter the entity toldo I see this screen

[image]

clicking the gear takes me to

[image]

disabling the last slider won´t do anything because after updating it comes back to enable, and if I choose the change device area, takes me to device area, not entity area. Wich changes all the entities area.

[image]

rstrouse, 2024-05-26:
For me I simply put the devices in my Windows area then toggled the the bottom box and and assigned each entity to the area I wanted.  It works just fine for me and the settings stick.  Did you press the update button after making the changes?

[image]

rstrouse, 2024-06-15:
I am going to close this for now as I believe you should be able to get the entities in any area you would like.

closed 2024-06-15 by rstrouse