rstrouse/ESPSomfy-RTS #383: Thanks for your excellent project.

paul-nzl, 2024-05-25:
Thank you for the excellent project and the detailed information provided.
I have used the Olimex ESP32 POE ISO and the EO7-M1101D-SMA transceiver.
I have used the UEXT connector to connect to the Transceiver.

My Olimex ESP32 connects on either WiFi or Ethernet.
However I can’t get my Radio to work. When i press Save Radio button
I get:
500:Service Error
Service:PUT /setNetwork
Internal Server Error

This same error occurs when I remove the cable between the Olimex and the Transceiver.

Units Wired as follows:

E07-M1101D-SMA		Olimex		Wire Color	
Pin			             Pin		
1	GND	             2   GND	        White
2	VCC		             1	  3v3		Black
3	GDO0 Tx	             3	  GPIO 04		Gray
4	CSN		             6	  GPIO 13		Green
5	SCK		             9	  GPIO 14		Red
6	MOSI	             7	  GPIO 15		Yellow
7	MISO	             5	  GPIO 16		Blue
8	GDO2 Rx	             4	  GPIO 36		Purple

[image]
[image]
[image]
[image]
[image]
[image]

rstrouse, 2024-05-25:
The error message in your first screenshot is from saving the Ethernet adapter not the radio.  So lets first get your network adapter sorted.  I can tell from your screenshot that you are connected via wifi.  However, in the first screenshot it appears that you are attempting to connect via network cable with a wifi fallback.  My guess with the error message above is that the wired connection closed the connection and fell back to wifi momentarily which caused the first error message.

This means that the ethernet adapter is not properly initializing.  Remove your ribbon cable from UEXT for now to ensure it is capable of connecting via the ethernet adapter.  When ESPSomfy RTS is connected via ethernet the screen will appear like below.

[image]

Don't start troubleshooting the transceiver you get the ethernet working properly.  The reason for this is because the Olimex board is extremely complex and the documentation is overly simplified.  For instance I can no longer find the documentation related to the pin settings for the LAN87x0 configuration.  It is infuriating that they removed the original link describing the ethernet settings.  I looked everywhere and cannot seem to find it.  And their example now points to a generic comment.

However, I found this post by others trying to get around the revision changes on the board.
https://github.com/emsesp/EMS-ESP32/issues/438

Try the following settings for the adapter to see if the ethernet comes alive.

[image]

### After we get the Ethernet connected properly

Once the ethernet is connected properly we can now troubleshoot the wiring for the the transceiver.  Let me start out by saying that custom ribbon cables can be difficult to get right.  You really need to inspect the alignment of the ribbon to ensure the proper conductor is attached to the proper pin.  Then you need to test the continuity for each side of the connector.

The radio not initialized message is because one or more of the top row pins on the configuration has either the wrong selection or more likely the connector is not making connection to the pin you intended.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/User-Interface-Reference#transceiver-tab

paul-nzl, 2024-05-26:
> emsesp/EMS-ESP32#438 https://github.com/emsesp/EMS-ESP32/issues/438

Hi
 
Thanks for your help, i have isolated the problem to the wiring of the connector to the Transceiver.
After replacing the connector everything is now working.
I'm extremely happy, many thanks for your excellent project.
 
I have included a screen of how my Olimex ESP32-POE shows up in my router.
 
Manufacturer: Konnected
Model: Konnected Alarm System Retrofit
 
Many Thanks
 
Paul

rstrouse, 2024-05-26:
Awesome!  Closing this.

closed 2024-05-26 by rstrouse