rstrouse/ESPSomfy-RTS #385: Wifi reconnect

mattfro, 2024-05-26:
### Hardware

ESP32

### Firmware version

v2.4.3

### Application version

v2.4.3

### What happened? What did you expect to happen?

I had some wifi issues, so I needed to reboot my router.
EspSomfy has been running for a few days at least. After I restarted my router, EspSomfy did not reconnect. it was in AP mode, so I connected to its AP and then rebooted it, after that it was connected on the router(at least I could see it in the router, and AP was not active) but could not access it(UI, did not try to ping).
After that I cut the power and then it was connected and I could access it... strange. At this point radio was not initialized for some reason. So I did another reboot from the GUI/UI and then radio was working, I could even see some sensors etc from the Logs so it was really working.

My question is, does it support some kind of reconnect to wifi after a longer period or does it stop after a while?
Second question, any way to debug this, logs for what is happening? Like see why radio was not first working or other logs why it could not connect etc. 
Hard to troubleshoot when not getting any information what could be wrong. 

Otherwise love this project and its great, good job!! 
Works very well...until ;)

Ps. could be hw problems also. Who knows, need to keep an eye if it happens again.
HW used:
ESP32 D1 Mini NodeMCU
CC1101 433Mhz Module (cc1101 v2.0 e07-m1101d)

### How to reproduce it (step by step)

```bash
Not really sure...
Tried to reproduce it myself and did not manage to do it in 30min
1. Blocked EspSomfy for accessing wifi(for e period of time, I think it was for hours when it happen)
2. Access esp wifi AP, restart it.
3. Was not able to access webpage/UI but it was according the the router accessing it(could not see any wifi AP)
4. Cut the power for a short moment
5. Access the webpage/UI and the radio is uninitialized.
6. Restart EspSomfy again
7. Now it's working
```

rstrouse, 2024-05-26:
It will keep the AP open for 3 minutes then close it and reconnect if it does not get connected to.  Currently, this happens if it loses the SSID for 30 seconds or more.  I am in the process of rewriting this to reduce the amount of time in the AP after the first connection.

rstrouse, 2024-05-27:
If you used Dupont connections double check to ensure they are not loose.  These connectors if pushed onto a pin several times will eventually become loose and can create an intermittent connection.

mattfro, 2024-05-27:
> If you used Dupont connections double check to ensure they are not loose. These connectors if pushed onto a pin several times will eventually become loose and can create an intermittent connection.

yes, that would be true. But I have solder them. Maybe I can use some serial tool to check if I can see something there?

rstrouse, 2024-05-27:
You can monitor the log with web.esphome.io.  Just to be sure did your refresh the browser window to verify whether the break in the link was simply a dropped socket vs a frame that was not heard?

rstrouse, 2024-05-27:
Also, the pre-release of v2.4.4 has improvements to the way the AP is managed.

mattfro, 2024-05-27:
Great thanks, will try the new version and also use esphome to check for logs, if get some issues again.

rstrouse, 2024-06-15:
I assume you haven't had further issues.  I am going to close this for now.

closed 2024-06-15 by rstrouse