rstrouse/ESPSomfy-RTS #386: ESPSomfy AP back online

gralfj, 2024-05-27:
Hi, my device has been runnig well for a couple of days.

Yesterday it became unavailable, looking on the wifi netwoks available from my laptop, I sudenly realised that the AP I used to first setup the device was there.

The network config, shades, groups, rooms, everything was still there, but the AP keeps on and I cant access the device from my wifi networks.

I have made no changes to the device, nor to thw router or networks.

rstrouse, 2024-05-27:
I am guessing that you have a computer that has the setting to automatically reconnect to the access point set.  This means that if your network goes down and the AP network comes back up again that computer will connect to it.  ESPSomfy RTS will consider that a connected device and not shut down the AP.

Check the computer that you used to configure ESPSomfy RTS and forget the AP.  It likely has the automatically connect setting turned on.

gralfj, 2024-05-28:
hi, it is having strange behavior, later yesterday worked again, and now the AP is back again.

Have to say that I think I´m having issues with the operator router and its leasing times, in fact some smart relays I have installed are going offline sometimes, maybe this is what´s going on with the device, we have to name it same way hehe, but what I can´t understand is why the AP goes online again....

rstrouse, 2024-05-28:
The AP for ESPSomfy RTS comes online whenever it cannot connect to the currently defined network connection after some period of time.  It will not stop that AP if something connects to it but it will go away again when a connection can be made.  Likely you have whatever device you used to connect initially set up to automatically connect to the AP when it is visible.  In this case it will not shut down the AP which is the desired behavior.

The pre-release of v2.4.4 will perform this operation much quicker and will attempt to connect even when the AP is available.  Still it will not shut down the AP if there is a computer connected to it.  Please change your connection so that it does not automatically connect to ESPSomfy RTS AP when it is available.

gralfj, 2024-06-01:
Hi, well i have tried every device at home and the ap still online and i am unable to connect to the device vía my home wifi, vía its ap i can access It and have to say It can't see any wifi

rstrouse, 2024-06-01:
Gotta ask.  Are you one of those folks that use a hidden SSID?

gralfj, 2024-06-01:
🤣🤣🤣

Nope! What for? Hehehehe

rstrouse, 2024-06-01:
If you had said that you use hidden SSIDs then I would have aske you the same question... what for?  Anyhow ESP32 only supports 2.4ghz wifi so is your router set up for 2.4ghz?

gralfj, 2024-06-01:
Well, I have a dual band wifi and a 2.4 for smart devices

rstrouse, 2024-06-01:
I'll try to get it set up where I have a failed connection for the AP and the soft AP is open.  In the meantime you can enter the AP manually then restart the device.

gralfj, 2024-06-02:
Ok, I'll fome back later

gralfj, 2024-06-02:
Done, i have changed host name, and set ip to dhcp, now I connect via host.local address

gralfj, 2024-06-02:
Hi, something happens, i went out and everything worked fine. Just came back hours later and nothing works.

gralfj, 2024-06-02:
Do you think something happens with the device? any malfunction, or defective, or something?

[image]

rstrouse, 2024-06-02:
What do you mean nothing works?  Did the device drop off the network and the Soft AP reappear?

gralfj, 2024-06-02:
Yes and the device does not see any wifi

rstrouse, 2024-06-02:
The first thing that comes to mind is the power supply.  If this is connected to a data cable you can see the logs in web.esphome.io to see what it is doing.  If ESPSomfy RTS locks up it will reboot as there is a watchdog timer.  However, if the power supply drops below power output it will not boot.

gralfj, 2024-06-02:
Hi, usually I use to plug It to a mobile charger, but i Will plug tomlaptop and send log

gralfj, 2024-06-02:
`[22:26:53] 1
NTP Settings 
[22:26:53]pool.ntp.org TZ:CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
[22:26:53]WIFI Settings
[22:26:53] SSID: [gus] PassPhrase: [Josesalo123]
[22:26:53]
[22:26:53]Launching web server...
[22:26:53]Creating Web MicroServices...
[22:26:53]WiFi Mode: 0
[22:26:53]WiFi interface ready
[22:26:53]WiFi station mode started
[22:26:53]Completed scan for access points
[22:26:53]Scanned 2 Networks...
[22:26:53]*0: gus (-75dBm) CH:1 MAC:78:94:B4:2F:38:BB
[22:26:53] 1: vodafonegus24 (-76dBm) CH:1 MAC:8A:94:B4:2F:38:BE
[22:26:53]App Version:2.4.3
shades.cfg exists so we are using that
[22:26:53]Reading header at 0
version:24 len:76 roomSize:29 roomRecs:2 shadeSize:276 shadeRecs:4 groupSize:200 groupRecs: 1 pos:76
Setting S3 Transceiver Defaults...
[22:26:53]Applying radio settings Setting Data Pins RX:4 TX:3
Setting SPI Pins SCK:7 MISO:8 MOSI:9 CSN:6
Radio Pins Configured!
[22:26:53]Successfully set up the radio
[22:26:53]Enabled receive on Pin #4 Timing: 1
Initializing RX Queue
[22:26:53]Connecting to AP
[22:26:53]WiFi clients stopped
[22:26:53]WiFi interface ready
[22:26:53]WiFi station mode started
[22:26:53]Set hostname to:espsomfygus
[22:26:58]Completed scan for access points
[22:27:04]Found strongest AP 1 78:94:B4:2F:38:BB
Timing Net: 5930ms
Connected to access point
[22:27:07]Got WiFi IP: 192.168.0.234
[22:27:07]Socket Server Started...
[22:27:07]Max Heap: 204788
[22:27:07]Free Heap: 218428
[22:27:07]Min Heap: 215664
[22:27:07]
[22:27:07]Successfully Connected to WiFi!!!!192.168.0.234 (-75dbm)
[22:27:07]MDNS Responder Started: serverId=97633C
Max Heap: 196596
[22:27:07]Free Heap: 211988
[22:27:07]Min Heap: 211944
[22:27:48]Check github for updates...
[22:27:48]Max Heap: 196596
[22:27:48]Free Heap: 212192
[22:27:48]Min Heap: 209124
[22:27:51]Internet is Available: 1244ms
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - 32481
Max Heap: 155636
[22:27:51]Free Heap: 209140
[22:27:51]Min Heap: 153076`

gralfj, 2024-06-02:
after another devicec reboot,back to normal function,and this is the log

[22:30:48]s (-75dBm) CH:1 MAC:78:94:B4:2F:38:BB
[22:30:48] 1: vodafonegus24 (-76dBm) CH:1 MAC:8A:94:B4:2F:38:BE
[22:30:48]App Version:2.4.3
shades.cfg exists so we are using that
[22:30:53]Reading header at 0
Connected to access point
[22:30:55]Got WiFi IP: 192.168.0.234
[22:30:55]Socket Server Started...
[22:30:55]Max Heap: 204788
[22:30:55]Free Heap: 218680
[22:30:55]Min Heap: 215720
[22:30:55]
[22:30:55]Successfully Connected to WiFi!!!!192.168.0.234 (-87dbm)
[22:30:55]MDNS Responder Started: serverId=97633C
Max Heap: 196596
[22:30:55]Free Heap: 212036
[22:30:55]Min Heap: 211988
[22:30:58]Loading file /index.html
[22:31:16]Timing WebServer: 2631ms
Socket [0] Connected from 192.168.0.127 url: /
Initializing Socket Client 0
Timing Socket: 277ms
Max Heap: 172020
[22:31:16]Free Heap: 193992
[22:31:16]Min Heap: 187928
[22:31:32]Check github for updates...
[22:31:32]Max Heap: 188404
[22:31:32]Free Heap: 207840
[22:31:32]Min Heap: 180228
[22:31:35]Internet is Available: 1275ms
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - -1
Max Heap: 147444
[22:31:35]Free Heap: 204948
[22:31:35]Min Heap: 149004

rstrouse, 2024-06-03:
Both logs are perfect boot sequences.  Do you have an ip reservation for the device?

rstrouse, 2024-06-15:
@gralfj has this problem been resolved for you?

gralfj, 2024-06-15:
Sorry, but no. Just very busy right now, ill come back.

rstrouse, 2024-06-19:
I am going to close this for now but feel free to re-open it when you get time.

closed 2024-06-19 by rstrouse