rstrouse/ESPSomfy-RTS #388: Somfy keytis

goanes, 2024-05-28:
### Describe you new feature you'd like

Hi
I have a gateway door witch is a somfy keytis 2 ns rts.
I wondering if I can use this device to open/stop/close the gateway door.
If it is possible, is there a specific procedure to follow?
Thanks

rstrouse, 2024-05-28:
I believe this should work just fine although I do not have one.  The keytis is only the remote the motor you are controlling will have specific pairing instructions.

goanes, 2024-06-14:
Hi rstrouse

I finnaly had spare time to try pairing ESPSomfyRTS with the gate Somfy RTS.
So here is how I pair a new remote control (there is no pairing buton)
https://www.youtube.com/watch?v=h9WnXJ3oaSA

So I have try to do it at the same time with ESPSomfyRTS and the remote control.
Unfortunaly I did not suceed doing it.

If you have any idea to move on it will be nice.

May I ask another question.
I use another remote not from somfy (but a remote bought on aliexpress.)
(https://fr.aliexpress.com/item/1005006724253172.html?spm=a2g0o.tm1000004601.d0.1.3de75610lr93BU&sourceType=561&pvid=1e1aaaf2-9073-41aa-9e6b-0e1032b55c6e&pdp_ext_f=%7B%22ship_from%22:%22CN%22,%22sku_id%22:%2212000038097472125%22%7D&scm=1007.28480.379202.0&scm-url=1007.28480.379202.0&scm_id=1007.28480.379202.0&pdp_npi=4%40dis%21EUR%21%E2%82%AC%2022%2C06%21%E2%82%AC%208%2C09%21%21%21168.13%2161.68%21%40211b80e117183677286768954ef99e%2112000038097472125%21gsd%21FR%21184364220%21&channel=sd&aecmd=true)

Is there a way to "copy" it and reproduce the RF Code ?

rstrouse, 2024-06-14:
Which gate motor/controller do you have?  

You will likely need to use the same button commands as you are using when pairing the remote.  All of those button combinations are available on the virtual remote tab but you should use the same button commands on the remote and watch the log in ESPSomfy RTS to capture the frames.  When finished press the copy button and paste the sequence here.

rstrouse, 2024-06-15:
Just to clarify, it probably uses some combination of the buttons to pair.  Also, some of these motors allow you to turn on and off the ability to connect 3-button remotes.  If I can get a model number on the motor I may be able to guide you on how to get this paired up and operating.

goanes, 2024-06-16:
Hi
Thank you for your support.
Here attach photo of my device.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-16:
Ok so it appears that the keytis is sending a Toggle command.  I suspect based on what I see in the instructions (I could only find it in French) it uses a weaker antenna when in pairing mode so you need to get the device close into the area for pairing.  You will need to get your ESPSomfy RTS device antenna close to the place where you put the keytis when pairing.

First press the button on the motor to put it into pairing mode.  Pairing mode will be indicated by a red light on the motor.
Then place your ESPSomfy RTS near the location where you place the remote when pairing.

Finally, in the ESPSomfy RTS virtual remote select the shade you configured for the gate from the dropdown and press the Toggle button.  It should be paired at that point.

goanes, 2024-06-17:
Hi rstrouse

I did what you recommanded and it finnaly work.
I put here a little procedure of what I did. I think it may help other people.

I first create a new room then a new shade.
I have to change on the shade tab the bit from 56 bits to 80 bits (and save).
I put the ESPSomfy RTS device (antenna) on the somfy gateway motor device (I don't have any button on it.)
Then I took the virtual remote tab and I did as I was on the real keytis remote.
That's mean the TOGGLE button. I push long the1st time then the gateway device jump to the pairing mode (LED RADIO turn on), I then push TOGGLE again the LED RADIO is still ON and the LED "Sequential auto closing" is ON and finaly, I push TOGGLE again "RAZ" LED is blinking then all LED is turning OFF.
That's it.

Then I push toggle again to test if it is working. and YES it did ... so happy. 

Thank you again rstrouse for this wonderfull tool.

I would recommand / request to change the UP/MY/DOWN button by the TOGGLE button as is it a gateway (or have the possibility to choose).
and the same thing happen on home assistant integration (no toggle button and only up/my/down button).

Thank again for your help (y)

goanes, 2024-06-17:
I just find the the Garage 1 button which is my case. so that is fine on the ESPSomfy RTS Web interface.
But on Home assistance. the entity did not change to a toggle.

rstrouse, 2024-06-17:
> But on Home assistance. the entity did not change to a toggle.

Reload the integration.  I think that even though there is a toggle going on it is smart enough to handle up and down.

goanes, 2024-06-17:
I reloaded the integration bit still the same. I see up stop down button but none of the 3 close or open the gate.
I deleted et recreate the integration and still the same.
Is there something I missed?

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-17:
Did you try the entity to see if it effectively opens/closes the door?  Most of those motors digest the up/down/my commands just fine.  If not I will see if there is a cover entity that matches the command buttons.

goanes, 2024-06-17:
Yes I have tried the up my down button inside the entity and it does not open close the gate.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-17:
Just to be sure when you use the virtual remote, if the gate is closing and you press the toggle button does it reverse direction?

goanes, 2024-06-18:
Yes I only use 1 button for close and for open.
for exemple, if the gate is close...
1 push on toggle wille open the door
another push on toggle (while opening) will stop the door
another push on toggle will close the door

any other key is not working

rstrouse, 2024-06-18:
I will be posting an update to the Home Assistant component next week but in the meantime you can update to the pre-release of v2.4.5 firmware (it will be the last item under the GitHub update dropdown).  This will now allow you to select any of the 1 button gate options for the shade type and it will operate as described above.

If I find some time I may get to the HA component earlier.

rstrouse, 2024-06-20:
v2.4.5 firmware has been released as well as the v2.4.5 integration for HA.  This should allow you to operate this gate as a 1 button gate type (left, right, or center).  I had to do some tango dancing with HA to make it work correctly so when in movement only the stop button will be available.  Once it reaches either endpoint only the appropriate button will be available.  If it does not know the current state it will show both open and close but only the proper one will work.

goanes, 2024-06-21:
Hi
I upgraded the esp SW successfully.
I will do the HA part in 2h. I let you know...

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

goanes, 2024-06-21:
But just to let you know...
HA did not see the new version 2.4.5 but only 2.4.4

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

goanes, 2024-06-25:
Hi
I have tested all the week end fine but some time there was something wrong but I don't know what it is.
This morning I see something weird.

When I use the 3rd cover name 'benjamin' it open the gateway when I use up. When use 'down' it close the gateway.

I really don't know what is happening. As the 3rd cover use 56bit and the gateway use 80bit.

After some more test. It looks that any cover is opening the cover and the gateway.  Same for closing.

Do you have any idea ?
Do you need my back up file ?

Thanks for your help.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-25:
A backup file would be great.

goanes, 2024-06-25:
Here attached

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-26:
Unfortunately, your backup file did not make it into the post.   However, did you go back in and change the type to one of the 1 button gate types?

rstrouse, 2024-06-26:
Oh I think I understand.  Unfortunately the command is just a toggle so ESPSomfy has no idea of what the starting position is.  To fix this make a note of where ESPSomfy thinks the position should be then unplug it and move the gate to the position.  Then plug ESPSomfy back in.

goanes, 2024-06-26:
I didn't understand your last point.
Regarding the backup file, should I copy paste here ?
I downgrade to 2.4.3.
I have the same issue.
Any up button on any cover is opening the corresponding cover and the gateway.
Any down button on any cover is closing the corresponding cover and the gateway. (This is happening only if the gateway is in open position)

If the gateway is close, the down button of a cover will not toggle (to open) the gateway.

If I only use the toggle button, the gateway works perfectly and no cover is moving.

Is it possible there is something wrong with the espsomfy code ? (As the cover is using 56bits and the gateway is 80bits)

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-06-27:
Unfortunately, if you are using a mobile phone the attachments will be removed from your post.  Copy/Paste here will work just fine.

goanes, 2024-06-28:
Hi
here is the file:
[ESPSomfyRTS 2024-06-26T08_28_08.zip](https://github.com/user-attachments/files/16027173/ESPSomfyRTS.2024-06-26T08_28_08.zip)

rstrouse, 2024-06-28:
The only manual for the Evolvia I could find was in French.  From the looks of it, it appears that if you pair using the up button and not the toggle button it will then use the 3 button mode but I cannot really read the instructions.
[image]

However, if you select one of the 1-button gate options that I added for you then it will operate just like your remote but still keep track of the positions.  Keep in mind that the same command is sent to open, close, and stop when the motor only listens for a toggle command (which is what keytis outputs).

rstrouse, 2024-07-02:
Is it physically opening the gate or is it just reflecting that in the ESPSomfy RTS UI?  If it is physically opening the gate then the pairing mode was open on the gate when you were pairing the roller shades.  I cannot read the documentation as my French is not nearly as good as your English.

goanes, 2024-07-02:
Yes it is physical opening the gate.
I was thinking the same thing.
But I am pretty sure this is not the case.
Because obviously I can make a mistake on 1 but not on all 3 cover pairing with the gate.
Can you give me the link to the document you found ? I will try to factory reset the gate.

Do you think it is possible there is something wrong with the code ?
I mean maybe inside the code it send RF to the cover and also to the gate ?
Can we run a debug or something to understand more ?

I am just curious... where are you from?

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-07-03:
I am in the US.   There is no chance that it is sending more than one signal.  Not even the groups do that.  If the motor has the address in its memory it will process the frame.

https://service.somfy.com/downloads/fr_v5/5063223dnotice_evolvia.pdf

rstrouse, 2024-07-11:
@goanes did you get your pairing sorted out?

goanes, 2024-07-11:
Hi

I just did it this afternoon.
I have done a factory reset of my gateway.
And it works.

I think last time when I did the pairing, right after I did a test  with all my cover.
As the esp somfy device was too close of the gateway device, it also pair the signal cover with the gateway device.

I have 2 more questions.
1) It seems I can not synch more the one

2) is it possible to have a log of monitoring a rts device with a real remote or with a virtual remote?

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

rstrouse, 2024-07-11:
> 1) It seams I can not synch more than one

Do you mean more than one remote to the gateway?  That doesn't make any sense as that would limit the entry to only one vehicle remote or only one pedestrian with the specific remote.  Typically each remote including ESPSomfy RTS has a unique address that is paired wit the motor.

> 2) is it possible to have a log of monitoring a rts device with a real remote or with a virtual remote.

You can create another ESPSomfy RTS device that simply monitors traffic or use your existing one.  When the config pages are open it will display all traffic from any RTS signal.  Simply click on the Radio section and select the log tab.

goanes, 2024-07-16:
Hi
sorry for the delay
for the question 1:
please see the screen shoot
[image]
Currently I can not add more the one remote.
Is it a limitation ?

rstrouse, 2024-07-17:
No you can link up to 7 remotes per motor.  Click the link remote button again and it will link any time it hears a unique address.

rstrouse, 2024-07-17:
Also if this is a gate you can now select any of the gate 1-button types.  You can switch this at any time by selecting it from the drop-down.

rstrouse, 2024-07-26:
I am going to close this as I believe our Keytis functionality is working now.

closed 2024-07-26 by rstrouse

goanes, 2024-08-21:
Hi
I was on holliday.
Thank you again for your support.