rstrouse/ESPSomfy-RTS #389: Question: support hardware "Connair" (foto inside) possible?

c1328, 2024-06-02:
### Describe you new feature you'd like

Hi all,

at first I want to say thanks to that project - I have it running smoothly for some weeks and it works quite nice.

I remembered I used a project called "Connair" a few years ago:
Device: https://www.amazon.de/ConnAir-433-Home-Automation-Gateway/product-reviews/B008W1G620
Project page: I'll try to find it....

My question is, if this device can anyhow used with this somfy project - I know it's not a ESP32 but maybe someone has an idea. Here is a foto:
[image]

Thanks and regards
Carl

rstrouse, 2024-06-02:
This will not work with ESPSomfy RTS.  That processor is not supported.

rstrouse, 2024-06-15:
I am going to close this as I don't plan on supporting this hardware.

closed 2024-06-15 by rstrouse