rstrouse/ESPSomfy-RTS #390: Will 433.92 motorline work?

gralfj, 2024-06-02:
Hi, just installing new 3 screens, they will have motorline tub35e as motors, do you think our device will make it work?

rstrouse, 2024-06-02:
Many of the 433.92 motors work just fine.  However, there are so many clones out there that I cannot say for sure.  It isn't an overly expensive endeavor to try it out if you already have them.  I do know that another user has Tube branded motors on 433.92.  Keep in mind you cannot mix 433.42 and 433.92 motors on the same ESPSomfy RTS device.

If you are just now purchasing these I would look at motors that we know are compatible.

gralfj, 2024-06-02:
So i cant mix my somfy ones with this new ones on the same device?

I have already bought them, just few days to hve them in place.

rstrouse, 2024-06-02:
You cannot tune the transceiver at 433.42 and 433.92 at the same time because the radio center frequency is different.  You can have two ESPSomfy RTS devices though.

gralfj, 2024-06-02:
Ok, nice. I will try this week, when I expect to have everything set up

rstrouse, 2024-06-02:
Give the second one a different host name than the first one.

rstrouse, 2024-06-15:
@gralfj have you tried the Motorline motors?

rstrouse, 2024-06-20:
@gralfj did the Motorline motors work for you?

magarto, 2024-07-22:
For muy motorlines I bought a zigbee relay. It is very eqsy

gralfj, 2024-07-23:
> For my motorlines I bought a zigbee relay. It is very easy

Please explain a bit more.

gralfj, 2024-07-27:
> @gralfj have you tried the Motorline motors?

Hi, back on track, i was a bit "lost".

Have to say that, about other post of mine, now the device works nice, i have changed main router, so now device does not go offline anymore, so far.

And about this topic, a zigbee gateway on its way, ill come back soon.

Thank you all.

gralfj, 2024-08-11:
Hi, sorry if reactivating this thread is not allowed.

I have been trying to scan the new remote with no result.
Tried every option RTS, rtv, rtw on 56 and 80 bits.

Looking the log didn't show any row of data.

I have read the wiki again in search of something i could have missed, and I think i left nothing behind.

Any tip?

By the way, this IS a picture of the back of my remote.
[image]

rstrouse, 2024-08-11:
Unfortunately, if it is not hearing the remote at all then it is likely sending frames that ESPSomfy RTS cannot decode.  To be sure you have it wired correctly, you can set the frequency back to 433.42 and try your Somfy remote.   If it is hearing that it should be able to hear the 433.92 frequency if the remote is sending a frame preamble that ESPSomfy RTS understands.

gralfj, 2024-08-12:
Hi, no its not hearing anything.
Let me explain the process.

I have my device fully working with my somfy shaded

[image]

What I did was to go to radio tad and hit scan button, this was the procedure, then I try different options among protocolo and bit lenght, i didn't touch anything else, and got no result.

This IS the reading of my somfy remote👇
[image]

rstrouse, 2024-08-12:
That scan is scanning for any of the recognized protocols so it will not matter which one you choose.  It is simply not finding the preamble from the motorline remote.   I have no idea what that protocol looks like unfortunately.

gralfj, 2024-08-13:
Yeah, that scan IS just a try with my somfy remote.

Maybei should reset the device in order to find the motorline remote, or should It be found with current settings?