rstrouse/ESPSomfy-RTS #393: Wrong Tilt position when pressing My function through TaHoma

hajar97, 2024-06-07:
### Hardware

ESP32

### Firmware version

v2.4.3

### Application version

v2.4.3

### What happened? What did you expect to happen?

I have an integrated tilting blind.
I have configured My favourite position to be tilt of 50%
When I press my in the virtual remote or in my physical remote ESPSomfy correctly sets tilting to 50%
I also have TaHoma integration, which is also linked as a remote to ESPSomfy 
When I press My function in TaHoma ESPSomfy always incorrectly assumes that tilting is now 97% even though my blinds actually tilt to 50%. 
I looked at the logs of the transmitter and noticed an interesting difference of what happens when I press My on my physical remote vs TaHoma. 
When I press My on my physical remote it registers 2 times My command with 80-S bits. 
When I press My on my TaHoma remote for some reason it registers 2 times My command with 56-S bits and 1 Favourite command with 80-S bits. (see attached image, the bottom 2 commands are from physical remote, top 3 are from TaHoma)
I assume this different behaviour of My on a physical remote and My on TaHoma is what causing ESPSomfy to set tilting value to incorrect 97%. 
Also another peculiarity I noticed is that if I keep pressing My on TaHoma then my blinds won't move as they are already in my favourite position after pressing it the first time, but ESPSomfy continues reducing tilt by 3% each time I press My on TaHoma. 

[image]

### How to reproduce it (step by step)

```bash
Press My on Somfy TaHoma and it will reduce tilt by 3% each time it is pressed instead of just setting it to the right value of tilt and then not change regardless how many times I keep pressing My in TaHoma.
```

rstrouse, 2024-06-07:
> Also another peculiarity I noticed is that if I keep pressing My on TaHoma then my blinds won't move as they are already in my favourite position after pressing it the first time, but ESPSomfy continues reducing tilt by 3% each time I press My on TaHoma.

The first 3 frames look like they are from the Tahoma  are the last two from a remote?  It looks like the Tahoma sends a my command to get the ball rolling in 56-bits then follows up with a favorite command for any 80 bit motors.  Can you capture a few of the my commands that move the ESPSomfy RTS device ~3% and paste that output here is a code block?

hajar97, 2024-06-07:
I have attached a few images. 

Step 1 is how my blind looks like when I close it.
Step 2 is how it looks like after I press My on TaHoma. This time it got to 94%, but it should have been 50%. Blind itself moves correctly to 50% tilt. 
Step 3 is how it looks like after I press My on TaHoma again. This time it got to 92%. Blind itself correctly didn’t move at all.

And so on if I keep pressing my the blind won’t move after it took My position, while ESPSomfy will keep reducing tilt each time I press My button in TaHoma. Again, this behaviour is only when using My in TaHoma. It works correctly when using My button on a physical remote.

Attached is alos a screenshot of pressing My in Tahoma 3 times. Is it what you were after?

￼￼￼￼

hajar97, 2024-06-07:
Yes, you’re right. First 3 frames are from tahoma, bottom 2 are from remote. As you can see both send different frames when pressing My button.

rstrouse, 2024-06-07:
Unfortunately, your screenshots never made it.  Part of the discrepancy is because the Tahoma is trying to cover its bases.  It is first sending a Favorite on 80 bits then repeats it as a my command.  Notice that the rolling code did not change.  This is the method that they are using to allow for 80 and 56 bit motors.  I'll have to think about whether I want to adopt that hack or simply ignore my commands that follow a favorite.  

I can't imagine that doesn't cause problems with some motors.  Perhaps stuff like this is why Tahoma is not well liked.

I do want a copy of the log when using only the Tahoma.  Not a screenshot but click the copy button on the log and paste it into a reply.

hajar97, 2024-06-07:
Got it. Here you go. I pressed My 3 times on Tahoma:

[{"encKey": 196,
  "address": 6443519,
  "rcode": 2597,
  "command": "Favorite",
  "rssi": -86,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2015,2497,2480,2520,2482,2509,2489,2506,2484,2526,2482,2505,2508,2464,2557,2462,2500,2504,2503,2517,2474,2505,2498,2514,4805,1278,629,653,1274,632,657,631,625,1294,1282,641,630,1276,657,629,1279,1273,1285,649,620,659,633,636,636,1298,633,635,1284,1279,653,626,1276,1282,1286,1272,676,603,656,643,620,658,634,652,623,648,625,658,633,635,647,651,1255,650,648,1282,632,652,624,649,1277,1279,644,631,633,658,1271,653,629,1287,630,649,1276,653,624,648,624,658,632,1272,647,634,1285,646,647,1261,632,646,646,632,1283,646,645,655,607,653,651,1274,1281,639,646,624,653,631,642,1299,1255,1291,645,633,1279,638,648,1275,1286,647,623,1285,633,635,1299,1262,27,660,624],
  "time": "2024-06-08T01:48:32.831+0200"
},
{"encKey": 196,
  "address": 6443519,
  "rcode": 2597,
  "command": "My",
  "rssi": -83,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [645,2936,1397,2454,2497,2471,2526,2502,2501,2488,2502,2488,2523,2493,4802,1280,632,650,1274,655,631,644,634,1273,1281,647,645,1283,633,647,1271,1289,1261,649,651,631,631,644,646,1287,664,595,1294,1275,647,628,1303,1277,1276,1283,623,647,626,659,636,647,621,654,630,643,651,648,623,637,656,635,1276,632,656,1269,644,631,632,653,1270,1305,634,639,644,629,1275,649,619,1307,632,645,1273,656,609,656,634,648,674,1240,658,630,1272,634,659,1248,673,635],
  "time": "2024-06-08T01:48:32.944+0200"
},
{"encKey": 196,
  "address": 6443519,
  "rcode": 2597,
  "command": "My",
  "rssi": -83,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [663,1271,632,649,1269,1299,633,632,1287,1283,1270,630,649,670,604,652,2967,1335,2482,2486,2517,2477,2503,2503,2501,2502,2496,2497,2526,4777,1299,631,653,1276,632,630,656,628,1297,1254,649,646,1276,634,649,1268,1288,1282,646,625,662,633,648,621,1282,634,649,1296,1260,654,647,1276,1283,1275,1280,634,615,647,655,632,657,630,647,622,658,633,653,623,651,622,642,1295,628,654,1287,625,648,626,662,1258,1297,628,642,649,622,1300,636,628,1303,627,632,1287,647,624,634,657,637,650,1273,631,643,1270,659,618,1294,645,633],
  "time": "2024-06-08T01:48:33.074+0200"
},
{"encKey": 193,
  "address": 6443519,
  "rcode": 2598,
  "command": "Favorite",
  "rssi": -85,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2048,2487,2457,2523,2475,2502,2501,2514,2499,2494,2499,2519,2475,2499,2501,2508,2505,2489,2497,2506,2499,2497,2505,2519,4786,1298,624,655,1260,645,646,634,633,639,653,646,624,1294,631,629,647,657,1257,1294,1270,646,623,1301,658,625,642,627,626,662,1259,1290,631,662,1280,626,654,1263,634,646,651,634,632,650,648,628,653,633,633,647,645,631,630,644,648,646,1275,650,644,1278,631,650,618,649,1276,1290,646,634,635,649,1270,634,656,1267,645,653,1280,622,675,626,632,631,631,1301,625,636,1284,647,621,1307,635,646,620,637,1281,648,646,631,656,621,646,1275,1283,638,647,650,629,658,631,1290,1249,1308,645,634,1253,650,647,1282,1281,648,620,1287,631,646,1272,1284,644,647],
  "time": "2024-06-08T01:48:37.778+0200"
},
{"encKey": 193,
  "address": 6443519,
  "rcode": 2598,
  "command": "My",
  "rssi": -84,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [624,3023,1277,2522,2460,2502,2502,2495,2501,2507,2494,2508,2499,2519,4803,1254,656,625,1296,632,630,662,630,646,622,657,631,1280,622,656,631,652,1295,1237,1309,645,646,1260,632,649,647,632,658,627,1281,1279,631,646,1278,659,632,1264,646,654,654,607,655,647,621,656,634,651,624,646,624,661,634,636,646,650,1252,662,622,1303,632,644,622,625,1297,1301,619,659,618,642,1297,605,677,1273,621,657,1285,647,620,633,644,650,622,1304,636,634,1281,654,629,1288,633,648],
  "time": "2024-06-08T01:48:37.924+0200"
},
{"encKey": 193,
  "address": 6443519,
  "rcode": 2598,
  "command": "My",
  "rssi": -93,
  "bits": 56,
  "proto": 0,
  "valid": true,
[... 15 lines cut ...]
},
{"encKey": 136,
  "address": 6443519,
  "rcode": 2599,
  "command": "My",
  "rssi": -84,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [624,2966,1328,2501,2483,2528,2500,2486,2506,2486,2504,2498,2496,2522,4806,1281,1251,644,651,624,650,1292,1281,628,653,633,632,1288,1273,659,628,1272,1303,1273,618,658,1258,1299,1277,646,633,1269,644,644,643,652,1256,646,646,1279,1285,1276,627,658,636,639,1272,1278,652,645,646,610,656,635,1279,1280,658,623,1270,660,634,1294,621,654,1256,648,644,651,629,1294,1271,633,631,644,646,654,630,1289,623,650,626,662,1281,1275,630,633,655,646],
  "time": "2024-06-08T01:48:41.081+0200"
},
{"encKey": 136,
  "address": 6443519,
  "rcode": 2599,
  "command": "My",
  "rssi": -83,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [629,1287,619,657,1284,1258,687,589,1295,1285,1276,638,681,602,642,627,3455,833,2498,2488,2522,2485,2500,2505,2496,2508,2504,2504,2498,4779,1305,1260,646,647,633,632,1293,1272,659,628,645,621,1284,1282,648,625,1289,1274,1282,633,661,1262,1306,1259,644,647,1283,626,646,646,635,1285,648,622,1289,1284,1272,629,644,647,646,1277,1280,646,629,631,644,651,646,1274,1274,644,634,1281,635,647,1281,633,651,1273,659,631,645,625,1273,1283,649,648,629,630,653,628,1301,630,644,629,645,1299,1281,620,652,653,627],
  "time": "2024-06-08T01:48:41.230+0200"
}]

hajar97, 2024-06-07:
Hopefully these images make it now:

[image]
[image]
[image]
[image]
[image]

rstrouse, 2024-06-08:
Ok so I will process the secondary my command that is defined as repeated to the original favorite as a synonym for the original 
command and that should do it. 

> Also another peculiarity I noticed is that if I keep pressing My on TaHoma then my blinds won't move as they are already in my favourite position after pressing it the first time, but ESPSomfy continues reducing tilt by 3% each time I press My on TaHoma.

I think this is because it is getting an indication that it should move to the my position then about 240ms later it is getting a my but it looks at this as a stop command because movement has already started from the original favorite command.  I guess because the TaHoma doesn't know whether the motor is in transit it has to send all the commands.  The bad part of that is that this has the effect of making the stop command not work.

Can I get a log of you stopping the movement with the motor already moving?

rstrouse, 2024-06-08:
Install v2.2.4 pre-release using the GitHub update button.  I think I have it all working.   I don’t have a TaHoma box so I need to rely on you.

hajar97, 2024-06-08:
In Tahoma app, stop is a separate button to my. And it makes Tahoma send a STOP command to stop the motor and not My command. Here is a copy of the commands when I press My and then follow by a Stop press. And Stop does seem to stop motor from travelling to My favourite position when it is in transit:

[{"encKey": 178,
  "address": 6443519,
  "rcode": 2616,
  "command": "Favorite",
  "rssi": -82,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2019,2487,2501,2492,2507,2476,2505,2511,2510,2475,2513,2501,2488,2501,2512,2501,2513,2475,2506,2496,2506,2505,2508,2496,4783,1281,1286,1271,652,632,1284,648,624,1286,1302,1258,1286,1276,1277,1287,1296,1261,652,621,647,636,1288,1273,1278,647,645,626,652,636,633,1278,648,638,1288,623,667,1261,657,620,1296,643,629,1272,652,637,631,647,645,633,631,628,676,621,1297,1271,640,633,658,638,1280,1276,1286,1272,1277,1290,1272,637,656,1277,1272,1265,1302,1281,1263,1296,629,637,1288,649,622,637,660,624,648,1280,1286,620,649,629,657,644,623,1277,1282,1296,624,659,1282,620,647,1284,1270,655,629,1281,649,627,1278,1297,645,636],
  "time": "2024-06-08T07:37:17.593+0200"
},
{"encKey": 178,
  "address": 6443519,
  "rcode": 2616,
  "command": "My",
  "rssi": -82,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [633,2935,1353,2496,2485,2511,2497,2489,2508,2505,2510,2486,2505,2487,4801,1282,1287,1274,653,637,1284,620,648,1288,1281,1279,1285,1272,1276,1280,1297,1270,627,644,647,634,1287,1272,1279,652,647,626,651,632,632,1271,649,637,1286,645,645,1285,629,646,1271,656,630,1273,652,635,630,647,644,632,632,665,626,645,1271,1288,645,623,654,636,1284,1267,1282,1271,1299,1269,1297,633,642,1297,1255,1254,1297,1289,1302,1257],
  "time": "2024-06-08T07:37:17.593+0200"
},
{"encKey": 178,
  "address": 6443519,
  "rcode": 2616,
  "command": "My",
  "rssi": -82,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [655,1304,602,653,1268,1277,655,630,1287,1273,1285,657,622,645,632,657,2914,1346,2493,2533,2467,2511,2499,2512,2499,2483,2516,2509,2494,4783,1289,1284,1273,658,611,1281,648,652,1267,1275,1280,1311,1249,1278,1291,1293,1263,652,623,645,634,1286,1273,1279,649,644,652,628,633,660,1273,625,637,1288,645,646,1284,629,646,1270,643,643,1273,653,617,667,624,651,607,657,645,644,633,1286,1281,647,635,633,651,1273,1284,1270,1300,1261,1296,1275,626,668,1250,1284,1270,1298,1266,1296,1269],
  "time": "2024-06-08T07:37:17.727+0200"
},
{"encKey": 167,
  "address": 6443519,
  "rcode": 2617,
  "command": "Stop",
  "rssi": -82,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2044,2461,2513,2503,2488,2500,2510,2499,2506,2492,2506,2491,2509,2474,2530,2478,2507,2501,2503,2497,2507,2480,2524,2477,4808,1280,1284,1268,1280,651,647,1277,633,644,646,626,657,629,1282,1274,658,639,628,646,1272,1286,646,625,652,636,1288,1272,631,645,1291,628,655,611,655,1272,652,632,1282,648,625,660,633,1275,1274,637,654,621,670,1259,664,634,621,654,1280,1272,1299,1258,1298,1258,1314,1257,652,629,653,623,1298,1257,643,653,1283,1261,1296,636,634,638,645,1278,1284,619,670,1260,1293,629,650,632,657,620,644,1287,1283,621,647,634,658,618,668,1255,1312,1250,652,635,634,647,644,633,657,637,629,646,645,634,634,653,624,648,649,637,634,622,1296],
  "time": "2024-06-08T07:37:20.458+0200"
},
{"encKey": 167,
  "address": 6443519,
  "rcode": 2617,
  "command": "My",
  "rssi": -83,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [3629,1305,2476,2508,2500,2496,2482,2527,2499,2498,2501,2497,2494,4802,1281,1286,1270,1280,650,648,1282,632,645,643,625,652,637,1284,1270,656,616,643,645,1274,1294,646,611,657,626,24,1292,1286,635,644,1269,632,662,636,646,1278,634,651,1269,632,657,629,651,1270,1282,636,646,626,656,1285,622,648,652,617,1312,1253,1284,1272,1300,1257,1297,1289,629,644,644,635,1286,1271,648,636,1285,1275,1281,646,645,630,629,1289,1274,634,657],
  "time": "2024-06-08T07:37:20.458+0200"
},
{"encKey": 167,
  "address": 6443519,
  "rcode": 2617,
  "command": "My",
  "rssi": -82,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 11,
  "pulses": [648,631,658,630,636,645,647,639,632,643,644,626,1304,627,636,1275,1310,3586,1316,2485,2520,2493,2506,2485,2512,2500,2488,2499,2512,2500,4794,1286,1283,1277,1284,621,646,1289,634,635,646,628,653,633,1281,1274,657,628,643,649,1276,1281,644,630,631,643,1288,1307,632,623,1274,635,657,637,628,1297,628,644,1280,627,655,639,627,1294,1280,646,620,655,629,1282,647,627,654,634,1284,1279,1281,1274,1299,1264,1299,1259,638,647,644,631,1283,1290,646,634,1283,1272,1276,648,645,620,653,1284,1276,632,634],
  "time": "2024-06-08T07:37:20.460+0200"
}]

rstrouse, 2024-06-08:
I'm sorry I meant to tell you to install 2.4.4 pre-release not 2.2.4.   This should handle both cases.

hajar97, 2024-06-08:
Confirm, it is working correctly now.

Couple of additional questions:
1. I see that my physical remote always repeats each command twice. Should I setup ESPSomfy do that too, or if everything works well with 1 command no need?
2. I have integrated blinds. Often I need to move blinds to a particular location and keep them closed. Then later tilt. Right now there is only a command to move blinds to a location and then restore tilt that it had before starting the movement. Is there a way to also be able to move blinds to a particular location and leave them closed (i.e. do not restore previous tilt)?
3. Tilt function is not always consistent. Depending on whether blind was previously already tilted, or it was open, closed or moving it gets same % tilt to a slightly different position. Such impression means that I sometimes won’t turn the blinds correctly against the sun. Is it due to impression of the motor I have or is there a way to tune parameters in the ESPSomfy app to make it more precise?

rstrouse, 2024-06-08:
> 1. I see that my physical remote always repeats each command twice. Should I setup ESPSomfy do that too, or if everything works well with 1 command no need?

There is no need.  In fact your remote is actually trying really hard to only send one frame and if you work at it you can get it to send one without any repeats but our thumbs and the mechanical button are not fast enough.  Believe it or not the most common reason to need repeats is because the motors have a cooling off period while they are processing a frame that makes them unable to hear any other frames.   This is likely the reason TaHoma sends 1 80-bit and two 56-bit in a pattern.  Most motors won't hear the first My.

> 2. I have integrated blinds. Often I need to move blinds to a particular location and keep them closed. Then later tilt. Right now there is only a command to move blinds to a location and then [...]

You can send the movement and tilt commands consecutively on an integrated tilt.

> 3. Tilt function is not always consistent. Depending on whether blind was previously already tilted, or it was open, closed or moving it gets same % tilt to a slightly different position. Such impression means that I sometimes won’t turn the blinds correctly against the sun. Is it due to imperfection of the motor I have or is there a way to tune parameters in the ESPSomfy app to make it more precise?

The integrated tilt mechanisms are all mechanical and I imagine that there could be some slop in the mechanism.  The motor only runs at one speed and for the integrated tilt these are typically balanced by pulling on either side of the slat mechanism so the load should be the same.  Tilt time is calculated down to the millisecond so do some tweaking but I believe the wider the slat the more accurate you can be.

> 4. In home assistant automation is there a command I can use to set position and tilt of the blind together? Right now I have to set position, then have some sort of time delay to wait until it gets to that position and then run a separate tilt command. This is not a very elegant way of doing it. You probably have a better way to set those 2 parameters together?

You do not need to wait between these commands.

hajar97, 2024-06-08:
Unfortunately I tried sending consecutive position and tilt commands and it doesn’t work. Second command seems to cancel the first if first hasn’t finished by the time second is run. This is the reason for questions 2 and 4. Is it another kind of bug? Or am I missing something?

hajar97, 2024-06-08:
Update. I did another test of sequential calling of setting position and then tilt and confirm all working now. Strange it didn’t work before, but all good now. Seems false alarm on my side. FYI I played a bit more with tilting and noticed that when I set tilt to 50% then it looks different if it is setting tilt after closing blinds or opening blinds (which direction it was running before setting the tilt). This led me to idea of trying to play with tilt time a bit more. My measured tilt time is consistently 4000ms. However, it is only after I set it to 5000ms that it seems that it sets tilt to roughly the same angle, regardless of which direction it was previously running to.

rstrouse, 2024-06-09:
Awesome.  Glad we got it all worked out.

hajar97, 2024-06-09:
Thank you very much for quick help. Its working flawlessly. So grateful for your project. It is mind boggling why it is not a standard feature of Somfy.

rstrouse, 2024-06-15:
I am going to close this as I believe we have you all sorted out.

closed 2024-06-15 by rstrouse