rstrouse/ESPSomfy-RTS #395: Blind tilts when sending up or down commands

menaibluey, 2024-06-10:
### Hardware

ESP32-S3

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

I have configured 3 Blinds and one Awning. All work fine with the exception of one blind. It paired fine but when sending either an up or down command it just does one tilt step similar when I use the trackball on my remote. The My button also work fine.

Here is my configuration for the blind.
[image]

Looking at the Browser Console log the command is send correctly as far as I can determine. Hower I do get this message "index.js?v=2.4.4c:590 {msg: 'close-died', reason: '', evt: CloseEvent, sock: WebSocket}
". I do not see these with any of the other devices.

The log included below is from one interaction from when I clicked the down button from the Home screen.

The other 80-bit blind using a trackball works fine. I'm not sure what's going on here and would appreciate your help.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash
index.js?v=2.4.4c:4337 {max: 180212, free: 201188, min: 182652, total: 305588}
index.js?v=2.4.4c:2344 Somfy {initialized: true, frames: Array(0), shadeTypes: Array(14), btnDown: 1717984684548, btnTimer: null, …}
index.js?v=2.4.4c:2345 MouseEvent {isTrusted: true, screenX: 766, screenY: 501, clientX: 765, clientY: 380, …}
index.js?v=2.4.4c:2346 mousedown
index.js?v=2.4.4c:2316 Somfy {initialized: true, frames: Array(0), shadeTypes: Array(14), btnDown: 1717984710243, btnTimer: 251, …}
index.js?v=2.4.4c:2317 MouseEvent {isTrusted: true, screenX: 766, screenY: 501, clientX: 765, clientY: 380, …}
index.js?v=2.4.4c:2318 mouseup
index.js?v=2.4.4c:2322 {timer: true, isOn: null, cmd: 'down'}
index.js?v=2.4.4c:383 {put: '/shadeCommand', data: {…}}
index.js?v=2.4.4c:513 {shadeId: 4, remoteAddress: 809000, cmd: 'Down', source: 'internal', rcode: 151, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:4337 {max: 180212, free: 200928, min: 180844, total: 305556}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 1, …}
[... 151 lines cut ...]
index.js?v=2.4.4c:4337 {max: 180212, free: 197536, min: 170076, total: 305284}
index.js?v=2.4.4c:1903 {connected: false, speed: 0, fullduplex: false}
index.js?v=2.4.4c:590 {msg: 'close-died', reason: '', evt: CloseEvent, sock: WebSocket}
index.js?v=2.4.4c:592 Reconnecting socket in 3 seconds
index.js?v=2.4.4c:461 Connecting to socket...
index.js?v=2.4.4c:547 {msg: 'open', evt: Event}
index.js?v=2.4.4c:2810 {shadeId: 1, type: 1, remoteAddress: 808997, name: 'Blind Seline', direction: 0, …}
index.js?v=2.4.4c:2810 {shadeId: 2, type: 1, remoteAddress: 808998, name: 'Blind living room', direction: 0, …}
index.js?v=2.4.4c:2810 {shadeId: 3, type: 3, remoteAddress: 808999, name: 'Pergola', direction: 0, …}
index.js?v=2.4.4c:2810 {shadeId: 4, type: 1, remoteAddress: 809000, name: 'Bedroom blind', direction: 0, …}
index.js?v=2.4.4c:4349 {available: false, status: 0, error: 0, cancelled: false, checkForUpdate: true, …}
index.js?v=2.4.4c:4337 {max: 188404, free: 200700, min: 192944, total: 305700}
index.js?v=2.4.4c:288 {get: '/modulesettings', obj: {…}}
index.js?v=2.4.4c:1415 {fwVersion: 'v2.4.4', ssdpBroadcast: true, hostname: 'ESPSomfyRTS', connType: 1, chipModel: 's3', …}
index.js?v=2.4.4c:288 {get: '/networksettings', obj: {…}}
index.js?v=2.4.4c:1670 {ssdpBroadcast: true, hostname: 'ESPSomfyRTS', connType: 1, chipModel: 's3', checkForUpdate: true, …}
index.js?v=2.4.4c:288 {get: '/controller', obj: {…}}
index.js?v=2.4.4c:1956 {maxRooms: 16, maxShades: 32, maxGroups: 16, maxGroupedShades: 32, maxLinkedRemotes: 7, …}
index.js?v=2.4.4c:2234 (4) [{…}, {…}, {…}, {…}]
index.js?v=2.4.4c:4349 {available: false, status: 0, error: 0, cancelled: false, checkForUpdate: true, …}
index.js?v=2.4.4c:288 {get: '/mqttsettings', obj: {…}}
index.js?v=2.4.4c:4187 {enabled: false, pubDisco: false, protocol: 'MQTT', hostname: 'ESPSomfyRTS', port: 1883, …}
index.js?v=2.4.4c:4337 {max: 180212, free: 197736, min: 179976, total: 305284}
index.js?v=2.4.4c:4337 {max: 180212, free: 197512, min: 179976, total: 305268}
```

rstrouse, 2024-06-10:
That socket event is simply when the socket is re-established on the browser.

However the behavior you are describing sounds like the motor is in euro mode.  You need to switch it out of this mode.  Some motors switch this mode by holding down the up and down buttons on the remote until it jogs.

menaibluey, 2024-06-11:
Thanks for the quick response, much appreciated. I didn't even know there's such a thing as euro mode. I would have assumed that all my various motors would have been configured the same. But that's another story. Here's what I did.

1. I used the up and down button until I got the jug. So presumably the motor was now out of euro mode.

Result:

1. The physical remote now behaved like the ESPSomfyRTS Home screen buttons. Up and Down buttons just did a short jug or tilt. My button behaved as expected
2. It did not make any change of the behaviour of the ESPSomfyRTS Home screen buttons. At least the remote and ESPSomfyRTS are now in sync but obviously that's not working for me
3. I've reverted the change to the physical remote.

I'm not sure how we can resolve that out of sync situation. I can't reset the motor and repair as I can't access it easily. Is there way for ESPSomfyRTS to learn from the remote directly?

By the way thank you for your amazing work, simply stunning. The effort is enormous and the result something you can be very proud of. The community is in your debt forever. Let us know how we can support your continuous efforts.

rstrouse, 2024-06-11:
I need to see the frames that are being sent by the remote.  When you press the up button or the down button for that channel.  Under the configuration menu click on the Radio section.  On the remote channel press the down button and let the motor run to the blind to the bottom.   Then press the up button and let the motor run to the top.  After you do this then click the copy button on the logs tab and paste the results in a message here.

menaibluey, 2024-06-12:
Here's the Down and Up sequence logs initiated by the physical remote.

[{"encKey": 160,
  "address": 13476608,
  "rcode": 10593,
  "command": "Down",
  "rssi": -74,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [9977,10523,2461,2574,2455,2572,2472,2558,2470,2561,2450,2565,2452,2563,2479,2562,2463,2563,2462,2562,2461,2556,2461,2571,2456,2562,4789,1297,1240,1294,1236,684,615,694,611,661,615,693,610,1284,614,666,603,657,1242,1279,1271,1286,1251,1269,636,636,1267,671,605,680,615,697,602,1280,615,648,620,649,1238,1284,1266,677,615,683,593,1294,1239,702,595,1301,614,651,1244,1283,615,655,618,669,608,661,615,666,620,643,1256,690,595,1293,635,637,1245,670,626,1280,615,666,1222,694,614,1290,620,643,1259,660,621,1244,1226,682,589,705,596,698,565,1263,1216,684,569,669,583,698,565,1265,1214,1241,640,595,1227,678,588,1235,1214,689,583,673,589,1262,1208,1234,1223],
  "time": "2024-06-12T18:00:35.176+1000"
},
{"encKey": 160,
  "address": 13476608,
  "rcode": 10593,
  "command": "Down",
  "rssi": -73,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 12,
  "pulses": [4656,2476,2565,2467,2557,2462,2567,2466,2563,2454,2569,2461,2562,4760,1294,1242,1292,1251,675,622,677,619,676,610,697,610,1272,611,661,605,680,1246,1282,1254,1272,1257,1285,615,643,1257,686,614,670,611,687,616,1277,623,653,614,639,1267,1296,1236,683,616,694,611,1286,1242,671,610,1286,617,669,1235,1291,616,668,609,662,603,658,620,643,603,656,1241,679,621,1297,615,640,1242,694,610,1286,601,644,1246,682,619,1277,620,648,1264,684,593,1269,1184,708,564,686,568,699,607,1255,1213,666,590,683,568,694,560,1250,1213,1262,632,590,1217,678,594,1243,1208,681,590,687,565,1257,1216,1246,1227],
  "time": "2024-06-12T18:00:35.311+1000"
},
{"encKey": 168,
  "address": 13476608,
  "rcode": 10597,
  "command": "Up",
  "rssi": -65,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [10003,10822,2460,2592,2437,2591,2444,2574,2457,2571,2447,2578,2445,2558,2474,2561,2481,2543,2476,2567,2454,2566,2454,2542,2458,2589,4787,1296,1234,1286,1229,1315,1222,694,609,691,596,1293,1232,690,621,693,584,1290,1250,694,608,690,599,1299,1229,1291,1247,699,608,666,620,693,610,1285,593,668,601,657,1262,675,596,700,608,1291,1221,701,609,682,596,694,617,681,597,693,608,1287,1253,692,609,1292,599,669,1224,1287,1251,1271,1258,1291,1245,1279,1234,1302,1223,1291,1276,1302,1229,1273,1195,700,558,688,599,676,571,1244,1228,686,578,666,596,670,603,1251,1185,718,544,689,573,694,584,684,580,712,570,665,584,706,551,691,568,1265,620,605,625,610,1209],
  "time": "2024-06-12T18:01:46.342+1000"
},
{"encKey": 168,
  "address": 13476608,
  "rcode": 10597,
  "command": "Up",
  "rssi": -96,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 12,
  "pulses": [4668,2452,2572,2479,2574,2445,2584,2447,2563,2451,2592,2426,2573,4776,1290,1239,1290,1242,1282,1236,715,581,686,602,1309,1218,686,602,695,601,1286,1269,672,629,666,601,1297,1227,1292,1246,695,609,688,609,696,604,1278,598,673,600,676,1230,697,605,677,611,1289,1225,696,612,685,598,690,611,689,614,687,614,1267,1227,715,588,1284,648,624,1240,1315,1221,1297,1234,1290,1250,1299,1210,1318,1221,1296,1235,1293,1259,1270,1206,717,538,695,564,701,547,1270,1210,685,574,683,568,714,562,1256,1184,686,596,659,599,668,595,664,592,664,606,658,576,698,582,665,573,1276,612,609,616,614,1229],
  "time": "2024-06-12T18:01:46.883+1000"
}]

menaibluey, 2024-06-13:
> This is only that one blind that behaves this way.   Is that correct?  To be certain, when you changed the mode was the blind in the middle of its movement or at either end stop.  If it is at the end stops, the same command is used to set the end stops.

Hi, the log is from when the blind was at either respective end.  And yes it’s the only blind that behaves that way. I can provide you with a log of the same blind in a different room if that helps. Also what I’ve noticed is that not every command I press in the remote is logged in the Radio Log tab. It seems rather random. And sometimes after pressing a button it goes into a waiting display just spinning its wheel. Is that a hardware resource issue?Sent from my iPhone

rstrouse, 2024-06-13:
> Hi, the log is from when the blind was at either respective end.

I am referring to the instance above when you changed out of euro mode.  If the motor is at either of its end stops it is not changing modes when it jogs.  It is setting up to adjust the endpoints.  You need to move the motor to the middle position then hold down the UP and DOWN buttons at the same time until it jogs.

> And sometimes after pressing a button it goes into a waiting display just spinning its wheel.

That spinning wheel comes from attempting to reconnect the socket to the ESP32 from the browser.  Sometimes there is a delay when the router does not respond quickly with the mDNS address if you are using the server name.  If this does not occur when using the IP address this is likely tardy mDNS resolving the local name.

menaibluey, 2024-06-15:
> I am referring to the instance above when you changed out of euro mode. If the motor is at either of its end stops it is not changing modes when it jogs. It is setting up to adjust the endpoints. [...]
Got you, when I try to changed the mode the blind was about half closed. That response was in regards to the up and down logs.
> That spinning wheel comes from attempting to reconnect the socket to the ESP32 from the browser. Sometimes there is a delay when the router does not respond quickly with the mDNS address if you are [...]

No worries, will try IP addresses to see whether that makes it a bit more responsive.

rstrouse, 2024-06-20:
@menaibluey have you gotten your issue resolved?  Update to v2.4.5, I made some changes to the output frames for 80-bit up, down, and my.  If you are still having problems switch it to 56-bit for that particular motor and try that.

menaibluey, 2024-06-20:
Thank you. Will give that a go. At the moment version 2.4.5 is not yet visible in HA. Do you want me to download it from GitHub and replace the custom component? Happy to wait until it does and update in HA.

rstrouse, 2024-06-20:
If you go to the hacs store it will be available.

menaibluey, 2024-06-25:
I’ve updated to v2.4.5 and this resolved the issue. It’s working as expected. I got myself in trouble a bit unrelated to ESPSomfyRTS as two blinds are reacting when I operate one blind via ESPSomfyRTS. I will try and resolve it when I have some time. Thanks for getting this resolved. Much appreciated. Sent from my iPhone

rstrouse, 2024-06-26:
Awesome! I am going to close this for now.

closed 2024-06-26 by rstrouse

menaibluey, 2024-06-26:
Yes please and thanks again for all your awesome work. Sent from my iPhone