rstrouse/ESPSomfy-RTS #396: Not reliably controlling roller lind

voipster, 2024-06-11:
First, great work @rstrouse ! Possibly one of the most well thought out and detailed integrations I've ever used!

I've got mine up and running (ESP32-WROOM-32D & E07), but I have to keep hitting up or down until it controls the blind.

The ESP/E07 is about 3 metres away, line of site. I've tried increasing the power & fiddling with deviation, but it doesn't seem to improve the intermittency. 

Any suggestions on what I can look for / try?

rstrouse, 2024-06-12:
If you are using an E07 please change the tx power back to 10.  Above that it becomes noise.   Increase the number of repeats.   For blinds that do not respond right away.   Some motors are just slow to wake up their radio.   In most case you are probably sending 3 or more frames with the remote.

That being said I have had a couple of weak transceivers that simply did not transmit or receive well.

rstrouse, 2024-07-03:
@voipster did you get your issues resolved?

voipster, 2024-07-03:
> @voipster did you get your issues resolved?

Yes! Followed what you said, and it's working like a dream now. Meant to come back here and update but far less interesting stuff got in the way. Thanks again for your help :)

closed 2024-07-03 by voipster