rstrouse/ESPSomfy-RTS #398: Impossible to set pause

TapsHTS, 2024-06-14:
### Hardware

ESP32

### Firmware version

2.4.3

### Application version

2.4.3

### What happened? What did you expect to happen?

When it's been a while since I've used the virtual remote controls, I decide to close a shutter and then I decide to pause the descent but it continues. To pause it, I need to bring it back up and after i can pause it.

### How to reproduce it (step by step)

```bash
1. Wait few time (I don't have the specific time)
2. Close your shutters
3. Try to pause it
4. Go up
5. And now you can pause it
```

rstrouse, 2024-06-30:
@TapsHTS any more information regarding this?

rstrouse, 2024-07-06:
I am going to close this for now as I assume you have figured out the issue.  If not feel free to reopen it.

closed 2024-07-06 by rstrouse