rstrouse/ESPSomfy-RTS #399: Possibility to link more than one remote with different frequencies

Jpm100P2, 2024-06-16:
### Describe you new feature you'd like

In my use case I've 3 remote Somfy Telis 1 RTS Pure controllers that drives 3 different awnings. 
I successfully paired all three awning and everything works as expected.

Next step, linking the remotes...

I successfully linked 2 of 3 because it seems that they're on the same frequency but I can't link the third one unless I change its frequency, but in this case i loose the first two.

I'd like to have the possibility to manage multiple remotes.

rstrouse, 2024-06-16:
Increase the rx bandwidth. It sounds like the 3rd remote is probably not well tuned.

Jezza34000, 2024-06-18:
You have to set the frequency in the middle of all your remote.  
+ as @rstrouse  says increase bandwidth it will solve your problem

rstrouse, 2024-06-30:
@Jpm100P2 did making the changes described above fix your issues?  Also, if you have not changed the batteries in the remotes in a while this would be a good opportunity to do so.

Jpm100P2, 2024-06-30:
Hi, yesterday I did the suggestion of @Jezza34000  including your suggestion and it worked. 
Many thanks.

rstrouse, 2024-06-30:
Cool I am closing this.  If you have any other issues please submit a new issue.

closed 2024-06-30 by rstrouse