rstrouse/ESPSomfy-RTS #400: MQTT subscribed topics not working

Nonixnarrez, 2024-06-16:
### Hardware

ESP32

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

Hi, 

first of all a big "THANK YOU" for the work you have put in that project.
I´ve built my own box now and it works like a charm from the WebGUI.
The **published** topics also arrive in my MQTT broker instance (iobroker) without problems.

I´ve manually created the following two additional states (read/write) from the **subscribed** topics in my MQTT broker:
- [root topic]/shades/[shadeId]/direction/set
- [root topic]/shades/[shadeId]/target/set
Reference: [https://github.com/rstrouse/ESPSomfy-RTS/wiki/Integrations#root-topicshadesshadeiddirectionset]
I have replaced [root topic] with "somfy" and [shadeId] with the respective object/number of course.

Here comes an example of the state definition (JSON format):
`{
  "common": {
    "name": "somfy/shades/7/target/set",
    "desc": "Manuell erzeugt",
    "role": "state",
    "type": "number",
    "read": true,
    "write": true,
    "def": 0
  },
  "native": {
    "topic": "somfy/shades/7/target/set"
  },
  "type": "state",
  "_id": "mqtt.0.somfy.shades.7.targetSet",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  },
  "from": "system.adapter.admin.0",
  "user": "system.user.admin",
  "ts": 1718545777490
}`

However setting the applicable values on those states (-1/0/+1 or 0-100) does not trigger movement of the shades.
It might not be a bug but just a "newbie" problem in MQTT configuration, so please bear with me.
Any advice how to fix this / what I´m doing wrong would be very much appreciated.

Regards /// Marc

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-06-16:
I don't use ioBroker but I suspect that you should not need to set up a state variable for this.  The `set` portion of the MQTT topic is publish only for the MQTT client.  You should not need to subscribe to the `set` topics and only subscribe to the state topics.

Reading the ioBroker docs I assume that the command should look something like the following to set the the shade at id 7 to 100%.  

`adapter.sendTo('mqtt.0', 'sendMessage2Client', { topic: 'somfy/shades/7/target/set', message: '100', retain: false });`

I have verified with MQTT explorer that the functionality is working as advertised.

rstrouse, 2024-06-19:
@Nonixnarrez did you get your issues resolved?

Nonixnarrez, 2024-06-20:
Yes, I sorted it out with "sendTo" commands instead of additional states.
Thanks a lot for your advice! Issue#400 can be closed.