rstrouse/ESPSomfy-RTS #401: update to 2.4.4 fails

HansVanEylen, 2024-06-17:
### Hardware

LilyGO T-Internet POE

### Firmware version

v2.4.3

### Application version

v2.4.4

### What happened? What did you expect to happen?

Home Assistant indicated there was an update to the firmware. When I clicked to install, it failed immediately with the message : "failed to call service update/install. tuple index out of range".
Log output pasted below.

### How to reproduce it (step by step)

```bash
1. Go to Settings
2. Click on the Firmware Update message
3. Click on install
```

### Logs

```bash
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:241
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: June 16, 2024 at 18:05:30 (5 occurrences)
Last logged: 07:32:34

[140582541226720] Unexpected exception
[140581990951248] Unexpected exception
[140581989185088] Unexpected exception
[140581962967376] Unexpected exception
[140582181375856] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2741, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2784, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 977, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1049, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 461, in async_install_with_progress
    await self.async_install(version, backup)
  File "/config/custom_components/espsomfy_rts/update.py", line 118, in async_install
    success = await self._controller.create_backup()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 285, in create_backup
    return await self.api.create_backup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 601, in create_backup
    with open(file=fpath, mode="wb+") as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 123, in protected_loop_func
    raise_for_blocking_call(
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 42, in raise_for_blocking_call
    if check_allowed is not None and check_allowed(mapped_args):
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/block_async_io.py", line 28, in _check_file_allowed
    path = args[0] if type(args[0]) is str else str(args[0])  # noqa: E721
                           ~~~~^^^
IndexError: tuple index out of range
```

nsleigh, 2024-06-17:
I am getting the same error.

evandenboom, 2024-06-17:
I am also getting the same error.

Jezza34000, 2024-06-17:
This is a problem specific to HomeAssistant, this bug must be opened there [ESPSomfy-RTS-HA](https://github.com/rstrouse/ESPSomfy-RTS-HA/issues) and not here

Jezza34000, 2024-06-17:
To override this go to your ESPSomfyRTS's gateway and directly update it from interface

Jezza34000, 2024-06-17:
I've juste fixed this issues i will push a PR for fix in [ESPSomfy-RTS-HA](https://github.com/rstrouse/ESPSomfy-RTS-HA) soon

rstrouse, 2024-06-17:
It's odd that this would suddenly start failing.  Which version of the HA core are you running?  

This does need to be async on the executing task as the backup needs to complete successfully before starting to download the firmware.

nsleigh, 2024-06-17:
I'm on 2024.6.3

Jezza34000, 2024-06-17:
Yes, this error is due to a modification in the operation of Home Assistant. Using a synchronous function in an asynchronous function crashes.

HansVanEylen, 2024-06-18:
I'm on 2024.6.3 as well. Thanks Jezza3400. Meanwhile I've updated through the interface itself.

rstrouse, 2024-06-19:
I am going to close this as this issue has been resolved in the HA integration component

closed 2024-06-19 by rstrouse