rstrouse/ESPSomfy-RTS #402: Repeater can't be deleted

Jezza34000, 2024-06-17:
### Hardware

ESP32

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

Impossible to delete Repeater

### How to reproduce it (step by step)

```bash
Go to repeater try to delete a repeater with the bin icon.
```

### Logs

```bash
Request URL: http://192.168.0.1/unlinkRepeater
Request Method: PUT
Status Code: 200 OK
Answer : [140]
```

rstrouse, 2024-06-17:
I missed a cast on the output for the repeater list after reducing the json serializer.  I will post an update to the 2.4.5 pre-release this afternoon.

rstrouse, 2024-06-17:
The ci process finished on the server so if you install v2.4.5 pre it will allow you to delete the repeaters.

Jezza34000, 2024-06-17:
Fix is OK
Thank you

closed 2024-06-17 by Jezza34000