rstrouse/ESPSomfy-RTS #403: Need recommendation where to buy a 3D printed case

hajar97, 2024-06-18:
### Describe you new feature you'd like

I know you have recommended a few cases. Those assume you have your own 3D printer. Are you aware of anyone actually selling the case for an ESP32 Devkit that could also be added to the "Simple ESPSomfy RTS device" manual for those folks like me who do not have a 3D printer?

rstrouse, 2024-06-18:
Here you go

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#3d-printed-cases

hajar97, 2024-06-18:
None of these are for sale unfortunately. These seem to be free designs for DIY 3D printing.

rstrouse, 2024-06-18:
Sorry, I didn't understand at first.

I too do not have a 3d printer so I have purchased project boxes from amazon modified them to fit the antenna and usb.  

The specs for the box I used in my S3 mini build and how I modified it for use can be found here.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build

There are also ton of 3d print services that will accept the files in the wiki and print you a single part.  You simply upload the file to their site, they give you a price quote and pay online, then they print and send you the item.  You can even use free software to modify the files to your liking then upload those to these sites.

hajar97, 2024-06-18:
Ah, that’s a great ideal. Let me give it a try. Thank you!

rstrouse, 2024-06-30:
I am going to close this for now.  I hope you figured out a case that works for you.

hajar97, 2024-07-09:
Hi, I have an update for this thread. I printed this case recommended on your website:
[ESPSomfy-RTS Case - ESP32 Devkit](https://www.thingiverse.com/thing:5989117)

Unfortunately, it is by 1cm not deep enough for dupont connectors. This was my first problem.

I solved it by making my dupont connectors half the size and managed to fit it in and close the box. For whatever reason as soon as I pack everything inside the 3D case and only antenna is sticking out its sensitivity to RTS remote control drops. It only picks up signal from my remote control if I stand right next to it. If I go away more than 3 metres it stops registering. 

I have a 6 dbi antenna. I have now tried to purchase a 2 dbi antenna to see if it will do a better job of picking up signal from my remote and will let you know. If you have any other suggestions of what could be causing drop of signal as soon as I insert it inside the box I would appreciate it.

rstrouse, 2024-07-10:
That’s very odd.  Are the connectors making good contact when you put it in the box?  Also try to position the esp32 so that the transceiver is not on top of the Wi-Fi antenna.

hajar97, 2024-07-10:
There is no way to change position of transceiver vs wifi antena because of predisposed position of both boards in the case given its structure. I tried to ensure as much as I could that all connectors are making a good contact, though I cannot be 100% sure nothing was getting worse in the totally cramped box when I was desperately trying to close it. In any case, everything was working as expected 100%, but it just wouldn’t pick up the signals from remote further than 2m. Given all the challenges with 3 different transceivers (RTS, Wifi and Bluetooth) potentially interfering with each other I will try to move to an ethernet based version and see of it makes things better. I just purchased WT32-ETH01. I hope in cases of ethernet connection the firmware does make sure both wifi and bluetooth are off in order to minimise any potential interference?

rstrouse, 2024-07-11:
It is very odd that the box makes a difference.  The only thing that can be making 433mHz receiver interference is potential electrical (not radio) interference from the WiFi chip.  When you had the case made, did you have them make it from metallic materials?

Bear in mind the WT32-ETH01 works beautifully and the network is very responsive.  However, it does require different power options and a USB serial converter to get the firmware flashed originally.  After that you can do it over the network.  If you are having trouble getting it flashed I can get you through it since I have been there done that.

rstrouse, 2024-07-11:
Btw my new favorite device is the Xiao - ESP32S3 but I think you are trying to avoid the soldering iron.

hajar97, 2024-07-11:
Indeed, I have no soldering tools, so ideally trying to avoid it. 

Box is 3D printed of plastic, so it couldn’t be that.

I have a couple of those Xiao - ESP32S3, but under a different name and fully inside a compact enclosure - M5Atom. I use them as presence detection devices in all my rooms. Work surprisingly well. 

Anyway, back to my WT32-ETH01, which should arrive tomorrow. What is USB serial converter. Can you point out a sample device on Amazon that I could purchase?

rstrouse, 2024-07-11:
Search Amazon for `cp2102 usb adapter` one of these are what you are looking for.

hajar97, 2024-07-11:
Thanks a lot!

I found this one: https://www.amazon.es/DAOKAI-Convertidor-USB-UART-Adapter-Multifuncional/dp/B0BCKBHQ8Y/ref=sr_1_1?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2HH6YZKYLILCJ&dd=cu2w0yfbtmGXDHsakz1Evw%2C%2C&dib=eyJ2IjoiMSJ9.lZXgXxX6JKFsRDhXdfEbgw0PA-XwqTIOE6RAV9Wgd5SIZv-fV90YgNNOj_Ut2Z5Zq4n_K4WAYGCcZCZCIRTZvSYYT3BaKH2Vc7l4bJftUX1luKDQ1FtbLBaxdIp8dQY9dZ9LkB1WV91f-iWgfk-5MzPtMA23EsD_6oHIg7bTBuIQqQgLPQuU96xauAt2M9h6E9Gb5Au7D6OiasTVpgSZEOBzqykrapGyMLCHXAwDHd4ZNi_o7ziERvbDoGdX5uMFPzfT02oHpxSyWaq2oN3yCP5FbFg6iUbuq-V0j7QHrQQ.mqQ0mnuLbtoj_am12OYSt8ez98P9gTznfFSWWjTO4SA&dib_tag=se&keywords=cp2102&qid=1720731122&refinements=p_90%3A6820340031&rnid=6820335031&sprefix=cp2102%2Caps%2C198&sr=8-1

It has 2 DIP switches for different modes. Do you know which mode I would need to use in order to set switches correctly?

Where do I find instructions how to flash it the first time?

rstrouse, 2024-07-11:
> It has 2 DIP switches for different modes. Do you know which mode I would need to use in order to set switches correctly?

RS232 mode (TTL)

> Where do I find instructions how to flash it the first time?

Hook it up like the following then flash the `SomfyController.onboard.esp32.bin` file from the release.
https://github.com/thryvos/Tasmota-WT32-ETH01

Then after you have successfully flashed it remove the jumper from the IO0 and GND pins then reboot.   After that, connect to the softAP and set up the ethernet adapter.

hajar97, 2024-07-12:
Thanks a lot, will let you know how it goes.

https://github.com/thryvos/Tasmota-WT32-ETH01

hajar97, 2024-07-12:
Hm, I got the WT32-ETH01 v1.4 piece and the only port it has is RJ45. So I assumed it is PoE powered, but it is not switching on when I connect it to PoE switch. So I am lost. How do I turn it on?

https://github.com/thryvos/Tasmota-WT32-ETH01

rstrouse, 2024-07-12:
It is not PoE powered it is powered with the adapter mentioned above.

hajar97, 2024-07-13:
But then that adaptor is not a use once and throw away. I’d need to use it always?

rstrouse, 2024-07-13:
You can use any 5v power source connected to the 5v and GND pins. It is not really the most user friendly.

hajar97, 2024-07-13:
Bought myself a usb-c power only socket. Hopefully should connect fine. For now could use USB-serial converter for power, but it is way too bulky to keep using it just for power.

Got everything flashed and working perfectly. Hopefully no more wifi connect/disconnect issues.

Thanks a lot. All your guides have been tremendously helpful.

Couple of notes that may be helpful for someone in future:
* USB serial converter had to be set as USB-232. I think that’s what you meant, but I misunderstood it as TTL-232, which I tried first and it didn’t work.
* Once I had everything up and running, I was struggling to link my new WT32-ETH01 with Somfy RTS motor. First I thought it was an issue with wiring and so it wasn’t sending the signal properly. But then I realised that my problem could be that my motor has reached max number of remotes and that I first have to unlink my wifi ESP32 device and then try to connect the new one. This turned out to be exactly the reason for my problem. As soon as I removed previous one I could add the new one. I am so glad I didn’t reflash my wifi ESP32 yet, because if I had done that then the process of unlinking it would need to be different, but I have no idea how it should work in that case.

Now everything is working. Fingers crossed no new/different surprises with this new device. Hopefully I will get a 3D printed case for it in a week too and then it can finally be neatly packaged.

Once again, huge thanks for all the guidance.

rstrouse, 2024-07-26:
Awesome I am going to close this as I believe we have all your issues handled.

closed 2024-07-26 by rstrouse

hajar97, 2024-07-26:
Indeed, thanks a lot. Everything now works like a charm. With 3D print, none of the designs listed actually worked for me. I had to print once using 3rd party service, then realise the discrepancies, ask same service make some modifications and print again. Final result was near perfect. Rest of imperfections I had to resolve by locally cutting and melting the case. End result is a neatly looking box as on one of your pictures. Just the antenna isn’t sticking out at a perfect angle, but thats a minor.