rstrouse/ESPSomfy-RTS #404: Use esp32 devkit also for BLE presence sensing

hajar97, 2024-06-19:
### Describe you new feature you'd like

Usually these devkits can be used through ESPHome for BLE based presence sensing. Given that they are now flashed with custom firmware, is there a possibility to enable this additional usage for the device?

rstrouse, 2024-06-19:
Unfortunately, the BLE libraries are too big for the flash size.  It isn't just a matter of turning on BLE, the library sizes of WiFi and BLE are very large and each consumes ~1Mb per library.  If you do both you don't have room for anything else like OTA updates, a UI, or any custom firmware.

The other wart with BLE on the ESP32 is that it shares a radio with WiFi so when you have both enabled one or the other suffers.  I guess this is a long way around to say that until ESPSomfy RTS will probably not be adding BLE capabilities anytime soon.

hajar97, 2024-06-19:
I see. All clear, thank you for the explanation.

rstrouse, 2024-06-19:
I am going to close this out.  I don't think BLE can be added to the current platform.

closed 2024-06-19 by rstrouse