rstrouse/ESPSomfy-RTS #406: Cannot update firmware to version 2.4.5

hajar97, 2024-06-20:
### Hardware

ESP32

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

HA notified me of new firmware being available. But when I try to install it I get the following error:
Failed to call service update/install. tuple index out of range.
[image]

### How to reproduce it (step by step)

```bash
1. Go to device page
2. Click on firmware update
3. Click on install
```

hajar97, 2024-06-20:
Just noticed that this error only happened when trying to update by connecting to HA via Nabu Casa URL. When using a local IP address, then it has updated without a problem.

larserhardsen, 2024-06-21:
I have the same issue, and I am not using the Nabu Casa URL. The same error is thrown when clicking the backup button on the device's page

TranceSetter, 2024-06-21:
Here the same issue, and as @larserhardsen said it's for both Nabu Case and local URLs.

Jezza34000, 2024-06-21:
This issue is fixed with the last update : https://github.com/rstrouse/ESPSomfy-RTS-HA/releases/tag/v2.4.5

hajar97, 2024-06-21:
Unfortunately I cannot test it as I already updated to the latest version using another approach before. But thanks a lot for looking into it.

rstrouse, 2024-06-30:
I am going to close this.  The cause of this is well understood and addressed in later versions.

closed 2024-06-30 by rstrouse