rstrouse/ESPSomfy-RTS #407: Auto-reconnect to wifi in case it is down for some time.

hajar97, 2024-06-20:
### Describe you new feature you'd like

Was restarting my router today and noticed that after restarting it ESPSomfyRTS stayed offline even after router fully rebooted and started to accept wifi connections again. 

Question: Is there some periodic wifi re-connect retry logic that ensures that it will eventually come online again without having to manually restart it even if internet is down for some prolonged time, but eventually comes back online again? This is particularly important when admin is away from physical access to the device and cannot come to it to reboot after internet connection has been restored by someone else.

rstrouse, 2024-06-20:
If you do not connect to the AP it will close it as soon as it can connect to the said you defined.

hajar97, 2024-06-21:
Seems if I turn off router and some time later turn on it remains disconnected.

rstrouse, 2024-06-25:
I just did tried this again and did not experience your issue.  The one thing that can cause this to happen is to have the auto-connect settings turned on for the ESPSomfyRTS SSID.  In this case, as soon as the ESPSomfyRTS SSID becomes available then whichever device is set that way will connect to it.  While a client is connected to the AP, ESPSomfy RTS will not drop out of AP mode.  If it did then you would lose connection while you are trying to configure the device.

Are you sure you don't have any devices that you need to forget the ESPSomfyRTS network on?

hajar97, 2024-06-25:
Just tested it again:

1. Made sure I “forgot” ESPSomfy RTS SSID on all my devices
2. Turned off Wifi Access Point (part of my wifi Mesh) that is closest to ESPSomfy device
3. Checked HA to ensure it became unavailable. And that ESPSomfy RTS SSID indeed appears in the list of SSIDs but I do not connect to it. Also checked and confirmed my other ESPHome devices also went off radar.
4. Waited to see if it would pick up signal from another access point. It should be possible, though signal would be much weaker. Anyway, it didn’t in 5 mins. 
5. Turned on the access point again. Waited until its up and other ESPHome devices would come back online. They did successfully as expected.
6. Waited 15 mins pinging ESPSomfy RTS device. It didn’t re-connect to wifi in 15 mins. And ESPSomfy RTS SSID continued to be visible and accessible all that time…

After 15 mins I turned ESPSomfy RTS off and on again and it immediately connected to wifi and appeared in HA.

Enable roaming checkbox is ON in the settings of ESPSomfy RTS device.

Let me know what other info may help you find out why in my case it doesn’t reconnect until I reboot it?

rstrouse, 2024-06-26:
The pre-release of v2.4.6 has some changes that performs a scan whenever the AP mode is active and not connected.  I think my test case was previously finding the SSID on my mesh but the wireless on the main router was disabled.  In essence it was still finding the AP on startup even though it could not connect.

These changes should make the device pick up the SSID as soon as it is detected.

hajar97, 2024-06-26:
Tested. Sadly same result. It doesn’t reconnect. Waited for 30 mins. No luck.

alka79, 2024-06-26:
same for me: no change. 
The new code added in network::loop after `if(this->softAPOpened) {` to scan again is never executed.

rstrouse, 2024-06-30:
Install the current pre-release v2.4.6 as I have made changes that should rectify this.  Please test this since my network comes back online very quickly.  I stopped relying on timing to perform a new scan when the SSID cannot be found.

hajar97, 2024-06-30:
Just tested again. Switched off access point that was closest to ESPSomfy RTS device and voila it has picked up signal from another access point that is further away. Previously it didn’t do that.

Then I reconnected local access point, waited 15 mins and it reconnected to it again. I could judge it by vastly changing wifi dbms. 

All seems to be working as expected now. Thanks a lot for looking into it and fixing it promptly. Every day of using ESPSomfy RTS is a joy!

hajar97, 2024-07-01:
Hm, now a different problem started to occur which I didn’t have before. ESPSomfy started to loose wifi connection and become unavailable every so often. Sometimes every 7 minutes, sometimes hours go by without loosing connection and then it happens.There was definitely nothing affecting my wifi during this time.

hajar97, 2024-07-01:
[image]
Here is a screenshot to give you an idea. Just keeps connecting and disconnecting now. The issue with it is that often automations fire when it happens to be disconnected and then automation fails to perform correctly.

hajar97, 2024-07-01:
The issue may be on my side actually. I am investigating. I have connected 2 ESP32 devices to that single 5V USB power source and it could be interfering. I have now disconnected the second device and will observe how ESPSomfy RTS is behaving. I’ll report back tomorrow once I had it running like this for 24 hours.

hajar97, 2024-07-01:
Nope, that wasn’t the reason. It keeps loosing connection and then reconnecting every so often. Time between each connection loss is different. Sometimes there can be hours without loosing connection, sometimes it keeps loosing it and reconnecting every fee mins. I don’t seem to see any pattern to it or understand the reason. The only other test I can think of trying is to revert back to previous firmware and see if problem goes away. If it does then it must be something in the latest pre-released firmware version. Can I roll back to previous firmware version without loosing my configuration?

rstrouse, 2024-07-01:
> Can I roll back to previous firmware version without loosing my configuration

You can but don't do that.  This message may be simply that the active state is changing on the entity.  What does the WiFi entity look like?

hajar97, 2024-07-01:
Which wifi entity? Wifi SSID? It changes to “unknown” and then to my SSID then to “unknown” then to my SSID. This change to unknown and back to my wifi network happens every time when every other entity becomes unknown.

What do you recommend I try?

rstrouse, 2024-07-01:
The strength entity.

schirm22, 2024-07-02:
Hi there, it appears I have the same or a very similar issue. After upgrading to version 2.4.6 (due to Wifi disconnection issue), the device connects to the Wifi (WPA3) and quickly (10-30s) disconnects again. It doesn't open its own AP anymore. Even when it is online (i.e. connected to the Wifi), I am not able to access the web interface of the device. The device is currently placed right next to the Wifi router. MQTT is enabled. After 2-3 connection / disconnection cycles, the device does not connect anymore, nor does it open its own AP.

hajar97, 2024-07-02:
[image]

When checking Wifi Signal entity I do seem to have reading around -20dbm all the time (it is right next to the router). But those readings are taken every 5 mins. When checking Connection entity (see attached screenshot), it is clear that between disconnects and reconnects sometimes a min long gap. So if automation fires in that minute then it will fail and that's exactly what was happening to me. 

I restarted the router and the erratic behaviour seems to have stopped, at least last in the last 2 hours. To have more certainty I really need to wait 24 hours for conclusive results.

In any case, it cannot be just my router, as I have other ESP32 dev platforms and they all reconnect without a problem and keep connection without those strange dropouts every so often. 

Sorry I wish I had more scientific and reproducible results. For now it is somewhat all over the place.

rstrouse, 2024-07-02:
Actually if it had disconnected, that wifi signal would have also dropped.  This tells me there is something with the switch entity that I need to take care of.  All of the entities on the device will go unavailable if device is disconnected.

schirm22, 2024-07-02:
Could it be related to WiFi router channel switching?

rstrouse, 2024-07-02:
Yes if your router is switching channels the ESP32 will need to disconnect from the old channel to the new one.  But it shouldn't be switching channels all the time.  ESP32 has only one WiFi radio and does not support seamless channel changes.  Is the channel entity changing in HA?

hajar97, 2024-07-02:
Checked for the last 3 days and it changed channel only once. So that cannot be a reason for such frequent reconnects. Started doing it again… so router reboot didn’t help either.

hajar97, 2024-07-03:
[image]
Here is the Wifi Channel entity view over the last half a day. If you zoom in on image you will see small and large gaps in channel when device disconnects. Sometimes it is just a couple of seconds, sometimes it is minutes. It reconnects again, but there are clear gaps in connection. And also note how channel always remains the same, so its not related with channel changes.

hajar97, 2024-07-03:
I have been running ping of the ESPSomfy whole night for roughly the same period. And connection losses coincide with the dropouts of the device, though there are more of them probably because very short dropouts are more frequent than seconds long ones. Attached is a file with ping dump.
[espsomfy-ping.txt](https://github.com/user-attachments/files/16080671/espsomfy-ping.txt)

hajar97, 2024-07-03:
I may have found something relevant. I have another ESP32 device right next to it. Seems when I disconnected it the erratic behaviour of ESPSomfyRTS stopped and it held stable connection the last 12 hours. Strangely, none of the other ESP32 devices have this issue of interfering with each other. Could there be something in terms of wifi settings in the firmware that makes it so sensitive to interference from nearby devices? Mind you, connection strength is -9dbm all the time, so overall it is a very strong wifi signal.

hajar97, 2024-07-04:
Confirm. No disconnects whenever the other ESP32 device is disconnected. As soon as I reconnect it, the erratic behaviour on ESPSomfy RTS starts. The other ESP32 device runs latest ESPresence firmware.

rstrouse, 2024-07-04:
Bluetooth and ESP32 Wifi share the 2.4Ghz band.  I don't use any Bluetooth devices but the interweb is riddled with BLE and Wifi interference stories.  Some of the recommendations to cure this are to:
1. Change to the 5Ghz band but ESP32 only uses 2.4Ghz so this is not an option.
2. Find a WiFi channel that does not interfere.  Switch out of auto and choose 1,6, or 11.
3. Change the WiFi channel bandwidth to use the narrower 20Mhz instead of 40Mhz.

hajar97, 2024-07-04:
Yes, I suspected it has something to do with BLE as well. Thank you for recommending these. Let me give it a try. I’ll report back as soon as I had a chance to try it.

rstrouse, 2024-07-11:
For those folks that were having problems with ESPSomfy RTS not reconnecting to WiFi, please install the latest v2.4.6-pre version.  I believe I have fixed the issue.  As it turns out the autoReconnect feature changed in core.  While it was never really capable of connecting to the strongest AP, recent changes to it made the workarounds in the code not work properly in all instances.  It should now work as expected.

hajar97, 2024-07-13:
Hey, I flashed new version on July 11th and have been monitoring its performance since then. It all seems to be working perfectly, no issues to report. Those strange disconnects and no reconnects have stopped. Thank you for all the help once again.

rstrouse, 2024-07-14:
@hajar97 yeah I am working toward getting the Arduino core to v3.x in that process the network management changed a lot and things that did not work in earlier versions started working.  Notably, the autoReconnect function that is turned on by default.   However, since it did not work at all in earlier versions of v2.x of the Arduino core the reconnect strategy was manual.  But it started attempting to reconnect (occasionally to the wrong AP I might add) in later versions and I had not explicitly disabled it.   This had our ESP32 trying to connect one way while still trying to reconnect from the original AP.  This was not ideal when more than one AP was in play and simply failed when there was only one.

All that being said I am on the downhill toward releasing v2.4.6-pre so if you could please install the latest version posted out there today.  It contains some minor fixes to the services attached to the network connection.

hajar97, 2024-07-14:
I can install it, but now only to my ethernet version. I no longer have the wifi version unfortunately.

rstrouse, 2024-07-14:
@hajar97 there is a bonus to this in the Ethernet version as well although the wifi radio reconnects will not affect you the ethernet connection is a bit cleaner with the 2.0.17 core since it doesn't have to dance around whether the WiFi is currently trying to connect.  In the past there were conditions where the WiFi and Ethernet were available at the same time since the disconnect of the WiFi attempted to connect a few seconds later.  Even though it may not be available it still was attempting to do it.

hajar97, 2024-07-14:
Got it. Updated to 2.4.6-pre. So far all seems working well.

rstrouse, 2024-07-26:
I am closing this for now as v2.4.6 has been released.

closed 2024-07-26 by rstrouse