rstrouse/ESPSomfy-RTS #408: Shutter disappear

Jezza34000, 2024-06-21:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

Since i've updated version v2.4.5 one of my shutter has disappeared
I add it again and strange things the Rolling code was already set but not form the disappearing one.
And once finished to add the missing one, an other shutter has disappeared ...
I've try a backup restoration but same thing reappears after a while

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

Jezza34000, 2024-06-23:
I did a revert, but in fact this also happens on 2.4.4, each new cover added deletes an old one already present

rstrouse, 2024-06-25:
Can you upload a backup?  I can't seem to replicate this all the way to 32 shades.  Rolling codes are matched to the address dictionary.  Each address has an NVM storage location for the code where the address is the key.

Jezza34000, 2024-06-25:
I remove everything and re-enter and it's ok 
I close this

closed 2024-06-25 by Jezza34000

rstrouse, 2024-06-26:
Actually, I will bet that there was a quote in one of your shade names.  Unfortunately, processing strings from json is a bit of a chore in ESP32.  Sure there is a Json module but the latest version is too large to coexist without repartitioning.  But it was a bit of a memory hog anyway when trying to serialize a json document.  I will do some manual escaping when serializing the document in a future version.