rstrouse/ESPSomfy-RTS #409: Add "Up" button for pairing

atoulmin, 2024-06-23:
### Describe you new feature you'd like

HI,

I have a brand of shutters from "Motolux" when pairing a remote it requires that you press the up button, not the prog button.
Unless im doing something wrong but if we add the button i can then pair my shutters.

To ADD a New Remote or Channel
 1. Using a remote or channel ALREADY PROGRAMMED to the motor, wake the motor up by pressing and	holding the STOP (middle) button for more than three seconds. The motor will jog.
 2.	Press briefly the UP button on the NEW remote/channel within 10 seconds. The motor will jog.

IS there a way of adding this "up" button to the system so i can use it with my blinds?

Also i think the system uses a rolling code.  Maybe that's the issue.  I have included a screenshot of the remote codes captured.
[image]

Benedikt09, 2024-06-23:
Hi @atoulmin, is your shutter working on 433,42 MHz or 433,92 MHz? I have tried to log the commands of my shutters made by Kaiser Nienhaus Mercato series using 433,92 MHz which didn't work. My awning of Somfy on the other hand was able to get logged in the ESP. The awning works on 433,42 MHz.
@rstrouse How does ist work? If I configure the base frequency to 433,92 MHz in your tool, will the ESP log the traffic captured in 433,92 MHz? Or do the logs only display traffic captured on 433,42 MHz (Somfy RTS frequency).

Now back to the topic itself:
I've got a similar question as @atoulmin . In order to pair Kaiser Nienhaus shutters you have to send a "Stop"-command. Maybe that could be implemented too?

rstrouse, 2024-06-25:
@atoulmin there are a couple of manufacturers out there that do not use a prog button on the remote.   However, your motor on 433.92 you simply need to use the virtual remote tab to finish your pairing.  First, ensure you have set the protocol to RTV and the bit length to 56-bit.  Then using the existing remote place the motor in pairing mode and press the up button on the screen.  This will result in the motor being paired.  After you have successfully paired the motor, press the pair shade button and then click the shade paired button.  You will be all paired up.

@Benedikt09 there is only a single radio and it is tuned to the base frequency.  It will pick up any compatible frames that it finds in that frequency.  So if it is tuned to 433.92 it will only pick up frames in that frequency.   Conversely, if 433.42 is selected it will only pick up frames in that frequency.  

Unfortunately, since there is only one frequency, it will only work with one frequency at a time.  However, you can have two ESPSomfy RTS devices and they will coexist just fine.

I will think about some user interface options to allow for alternate pairing buttons in the future.

Benedikt09, 2024-06-25:
Thanks for your fast response.
I do understand that it is not possible to use both base frequencies at the same time. 433.42 was just a cross check to see if the ESP works in general.
My main goal was a remote for those Kaiser Nienhaus shutters using 433.92, but sadly I am not able to get any kind of logs when pressing remote buttons in your log viewer. I have tried nearly everything according to your Wiki in the transceiver configuration but without luck. I also tried to pair shutters using the virtual remote with the Stop command and the correct base frequency, but the shutters did not react to the Button press. Maybe they use another protocol and do not use RTV which I hoped for. There ist no information in the manuals either.

Benedikt09, 2024-06-25:
@rstrouse Maybe we could add a list to the wiki with 433.92 MHz manufacturers that other people got working with your tool?

rstrouse, 2024-06-25:
If you do not see any frames in the log from the remote, it is quite possible that it is not one of the RTx protocols.  Heck it may be a radio ASK protocol variant.  Decoding that is tough especially, when you don't have the equipment and the manufacturer gives no clues.

> @rstrouse Maybe we could add a list to the wiki with 433.92 MHz manufacturers that other people got working with your tool?

This is probably not a bad idea.  When I first put this together, I had no idea what equipment it would support beyond my own.  There are other hubs out there that list manufacturers maybe I can take some clues from their lists.  I am actually doing some protocol detection for the 433.92 side so if it finds the preamble it will try to decode it.

Benedikt09, 2024-06-25:
There is a German gateway called Mediola Gateway V6 which is recommended by the German manufacturer Kaiser Nienhaus but is extremely overpriced. They provide a compatibility list where one can see all the devices working with the gateway, which can be found here:
https://www.mediola.com/checkliste

Kaiser Nienhaus also provides its own USB gateway working with a smartphone app. Of course its not compatible with Home Assistant or Google Home/Alexa.
https://www.kaiser-nienhaus.de/produkt/smart-stick.html

atoulmin, 2024-06-27:
@rstrouse well further investigation the blinds do pair using the “prog” button. I had to use the virtual remote “prog” button when pairing as I had to hold it down for two seconds. Using my iPad when I held down the “prog” button on the pairing screen it wouldn’t work properly as after a second it would highlight the text. But holding down the “prog” button on the virtual remote worked. 

Might be worth amending?

rstrouse, 2024-06-27:
Thanks I will disable the selection for the buttons on the IPad.  This is likely interfering with the long press.