rstrouse/ESPSomfy-RTS #411: Somfy Movelite 35 won't pair

Roflcopterer, 2024-06-24:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

The radio receives my remote and the scanning works fine, but when i put my drapes in pair mode it won't pair with the espsomfy.

i tried changing the tx to higher and lower values
i replaced the E07-M1101D-SMA module
i rebuild the espsomfy using a differtent esp and a new radio module

### How to reproduce it (step by step)

```bash
I created the drapes, put the somfy motor in pair mode and click the pair button.
```

rstrouse, 2024-06-25:
This should work just fine for that motor.   Lets start out with the radio settings.  First, that transceiver cannot transmit beyond 10db so set it at 10.

Did you by chance change the address in ESPSomfy RTS to any address that was displayed in the radio log?  If you did, then this is likely the problem.  If not then give me a rundown of the pinouts you you used.

Roflcopterer, 2024-06-26:
Wow sorry for wasting your time, when writing down the pinouts I found the mistake. Both ESPs had inverted RX and TX pins... Thanks for your reply, in a way it fixed the issue!

closed 2024-06-26 by Roflcopterer