rstrouse/ESPSomfy-RTS #413: When ordering 2 blinds to go up or down simultaneaously, most of the time only 1 responds

MuadDib007, 2024-06-25:
### Hardware

ESP32

### Firmware version

v2.4.4

### Application version

v2.4.4

### What happened? What did you expect to happen?

2 automations in HA trigger at once, both sending an open signal to 2 different blinds.
Only 1 blind closes, the other does not respond.
Issue is intermittent.
Already set repeat to 1 time on the failing blind.

When manually controlling the blinds from HA or remote it responds (Unless state in HA is wrong, I cannot send UP if it think it is already UP)

### How to reproduce it (step by step)

```bash
1. Make 2 automations which send a blind open command to different blinds with same trigger.
2. Pull Trigger to run both automations.
3. You will only see 1 blind respond
```

rstrouse, 2024-06-25:
Do not run these as parallel in HA.  The processing is synchronous anyway so it should at least send the signals.  However, motors that are not grouped will sometimes need a cooling off period even if a command is not meant for them.  It still decodes the frame so that it can determine whether it needs to do anything.  ESPSomfy is likely sending the frame but the motor is not ready to listen to it since it is to slow to reset between commands.  In this instance add 2 repeats to each shade or actually group them together then call the automation on the group.

rstrouse, 2024-06-30:
@MuadDib007 did you get your issue resolved?

MuadDib007, 2024-06-30:
Hello @rstrouse I have decided to insert a delay of 2 seconds in the script which affects one of the blinds.
So even if the scripts run at the same time the command for the 2nd blind will be delayed by 2 seconds.
I expect this to fix the issue.

closed 2024-06-30 by MuadDib007