rstrouse/ESPSomfy-RTS #415: Ability to Override Current Position

BillyFKidney, 2024-06-26:
### Describe you new feature you'd like

The signal from our wind sensor is very weak and often not received by ESPSomfy despite relocating the antenna (10 dBi).
In order to recognize the awning is no longer extended I have a ZigBee contact (door) sensor attached to the arms.

When ESPSomfy reports the awning to be extended, but the contact sensor has been closed for 30s, Home Assistant will issue the close command correcting ESPSomfy's position. 

If I could force set the position it would be far less cumbersome.

rstrouse, 2024-06-26:
There are services to set the target position as well as the current position in the ESPSomfy services for HA.  This is the Change Position service.

BillyFKidney, 2024-06-26:
[image]

The services send a command to the motor, but I want to update ESPSomfy because the position is wrong. 

[image]

ESPSomfy believes the awning is open but in reality it is closed. 

[image]

[image]

rstrouse, 2024-06-26:
Do not use the Set Position service use Change Position

[image]

rstrouse, 2024-06-30:
@BillyFKidney did you figure out how to set the position with the service above?

BillyFKidney, 2024-07-04:
> Do not use the Set Position service use Change Position [...]

Thanks, I was not using Home Assistant's Developer Tools so I missed the additional information.

[image]

closed 2024-07-04 by BillyFKidney