rstrouse/ESPSomfy-RTS #416: Board is properly configured but there's no radio signal received

qntris, 2024-06-28:
### Hardware

ESP32

### Firmware version

v. 2.4.3

### Application version

v. 2.4.3

### What happened? What did you expect to happen?

No radio signals received
[image]
I have Somfy Situo RTS1 remote controls, but regardless of how close to the module I am pressing the buttons, there's no radio signal intercepted on the module. The hardware connection is proper (all pins are properly mapped).

### How to reproduce it (step by step)

```bash
Turn on the module
Log in and go to the Radio tab
Try to intercept radio signal from somfy remote
```

Jezza34000, 2024-06-28:
Hi, Which board do you have ? Can you give the connection mapping ?

qntris, 2024-06-28:
Hi, I have ESP-WROOM-DEV
The mappings are exactly as per the instructions. I thought if there’s a pin mapping issue, the application will show an error with the radio?
[image]
[image]
[image]
[image]

Jezza34000, 2024-06-28:
That impossible to see how it is wired...
Can you describe it ?

qntris, 2024-06-28:
The board is precisely configured according to the wiki - I have again triple-checked it (few minutes ago) before replying:
[image]

Jezza34000, 2024-06-28:
No your wiring is wrong. Check it again
Look at the picture your sent. 
Pin 7/8 from C1101 is Black & White and it goes to GPIO 12 GPIO13 that is not good !

qntris, 2024-06-28:
I have two black jumper wires. The mapping is correct - pin 7 goes to GPIO19 and pin 8 goes to GPIO12. Checked it multiple times. I have tried with two different ES32s and 2 transceiver modules.

Jezza34000, 2024-06-28:
Ok GPIO12/13 are not UART on ESP32 with 30pin
Try to change this 2 pins and plugg it to RX/TX on board (GPIO1/3) update your configuration and let's see if it's work ?

qntris, 2024-06-28:
Changed the pins as suggested, updated the config, saved the radio settings, powercycled the equipment. Still don't get any signal:
[image]

Jezza34000, 2024-06-28:
Can you take a close-up photo of the side of the circuit board where the chips are ?

Jezza34000, 2024-06-28:
And please increase RX bandwidth at 500Khz at least (maybe your remote is not centred on 433.420Mhz)

qntris, 2024-06-28:
That's the current wiring. 
[image]

Tried with the proposed RX and Freq deviation settings but I am still not getting any signals. Should I be seeing any values in the RSSI and dbm fields (on the right side of the frequency sliders)? Again, I was left the impression that if the ESP32 cannot send and receive data to the transceiver, it would pop up with an error that the Radio is not initialised. Since I am not seeing this message, doesn't that mean that the module is properly configured and connected?

qntris, 2024-06-28:
If by "invert" you mean just switching the jumper wires - tried it with no change of outcome.

Jezza34000, 2024-06-28:
Yes invert wire but not the configuration.

qntris, 2024-06-28:
No luck. Don’t know why it wouldn’t pick up any remote control radio signal… Thought it would be an easier config…

Jezza34000, 2024-06-28:
I'm running out of idea... 
If you try other hardware I don't know what it could be

Jezza34000, 2024-06-28:
One of your wire is broken ?

rstrouse, 2024-06-28:
Looks like you guys have been through quite a bit of troubleshooting.  This should just work for you.  Just to be sure, you have a remote that is an RTS remote and not an IO remote?  They look identical but on the back of the IO remote it will say IO instead of RTS.

There are a couple of things that need cleared up.  In general you should avoid any pins that are tied to a UART.  This is because board manufacturers sometimes add things to these pins.  For instance, in this devkit there is a CP2102 that is tied to the USB port on that board.  While that is great for the serial IO interface it can interfere with the operation of the pin pulses to and from the transceiver.  

I recommend you set this back to the recommended configuration.  I have color coded the wires based on the pictures above.  It is a bit hard from your pictures to ensure the wiring terminates where I think it does but here is my best shot at it.

| Pin | Color | ESP32 | Description |
| --- | ----- | ----- | -------- |
| 1 | Brown | GND | Ground |
| 2 | Red | 3v3 | VCC |
| 3 | Black | GPIO13 | GDO0 - TX Pin |
| 4 | Blue | GPIO5 | CSN |
| 5 | Purple | GPIO18 | SCK |
| 6 | Gray | GPIO23 | MOSI |
| 7 | Black | GPIO19 | MISO |
| 8 | White | GPIO12 | GDO2 - RX Pin |

The other thing that is potentially dubious here is that some of your connectors do not seem to fit tightly.  Unfortunately, after inserting and removing a Dupont female connector they become loose.  Do you have some better Dupont cables?  Finally, there is another possibility in that the E07-CC1101 that you are using is faulty.  I have had a few but I also have a ton of them.

[image]

qntris, 2024-06-28:
The remote is IO:
[image]
[
[image]
](url)
What does that mean?

rstrouse, 2024-06-28:
This means that you have io-homecontrol motors not RTS and ESPSomfy RTS will not be able to hear that remote or speak to your motors.  I am sorry.

rstrouse, 2024-06-30:
I am so sorry that ESPSomfy RTS cannot be used with your IO motors.

closed 2024-06-30 by rstrouse