rstrouse/ESPSomfy-RTS #417: Can receive signals but cannot send signals

dallingham, 2024-06-29:
### Hardware

ESP32

### Firmware version

v2.4.5

### Application version

v2.4.5

### What happened? What did you expect to happen?

I am trying to control an awning. I have a Somfy Telis 1 remote. I am using an ESP32-WROOM. I duplicated the configuration in the hardware page for this device.

Protocol : RTS
Bit Length: 56
Type: Awning

GDO0 - GPIO 13
CSN	- GPIO 5
SCK	- GPIO 18
MOSI - GPIO 23
MISO - GPIO 19
GDO2 - GPIO 12

I am able to monitor the key press from the remote. The shade on the display follows the button presses, and it identifies properly when the remote extends and retracts the awning. The home assistant integration also follows the awning properly.

The problem is that controller does not seem to be transmitting properly. The software cannot get the awning to either extend or retract. 

This is the wiring.
[image]

My configuration is:

[image]

The configuration of the transceiver is:

[image]

As a note, I also tried putting both the RX and TX on GPIO-13. Nothing different happened, so I moved it back to GPIO-12

Capturing of the existing remote:

[image]

### How to reproduce it (step by step)

```bash
1. Go to screen for the shade
2. Click on any of the buttons (my, up, or down)

Nothing happens with the awning.
```

rstrouse, 2024-06-29:
Actually if that is a WROOM 30 pin ESP32 then you GDO0 on GPIO12.  Don't change your wiring just swap the RX and TX pin settings in the UI or set both of them to GPIO12.

dallingham, 2024-06-29:
I've tried TX on GPIO12 and RX on GPIO13 and I've also tried TX and RX both on GPIO12.  

Neither works.

rstrouse, 2024-06-29:
If your ESP32 looks like the one pictured in the wiki you have GDO0 (TX) connected to GPIO12 and GDO2 (RX) connected to GPIO13.  So if you select the TX dropdown to GPIO12 and the RX dropdown to GPIO13 then we should be set up correctly.

There will not be any indication in the logs as this only logs the receiver on the CC1101.  There is no loopback on the transceiver to know whether it actually sent the frame.  Still that is a pretty good assumption that it sent a frame.

For the procedure below, I am going to recommend that you only send a couple of commands after doing this then use your remote a couple of times after that to move the motor.  Change the address in the configuration to 327466 then click on the Set Rolling Code button to ensure the rolling code is the last witnessed address from the remote + 1.  Now try to move the motor.  If it moves then ESPSomfy RTS is sending frames to the motor.  Now change the address back to 536265 and press save.  Use your remote a couple of times until the motor responds with the remote.

**NOTE:** if we do not get any response from the motor then the transceiver is not sending frames and we have another problem.  We need to review all the wiring to make sure the pin selections match the configuration.

If the motor responds to the remote address unpair the motor from ESPSomfy RTS then attempt to pair it again with the 536265 address.  Please take note of the jogs.  If it does not jog it did not unpair.  If you cannot get it to unpair then simply click the Shade Unpaired button and proceed to the next step.

Change the address to 546265 and try pairing again.  Please note that you should only get one jog after long pressing the prog button the back of the remote.  If it does not jog or it jogs more than once then it is not in pairing mode.  Then open the Pair Shade screen and press and hold the pair shade button until the motor jogs exactly once.  If it does not jog or it jogs more than once then it has not paired.

dallingham, 2024-06-29:
Changing the address and re-pairing did the trick. Thanks!

closed 2024-06-29 by dallingham