rstrouse/ESPSomfy-RTS #418: Lost room groupes, shade…

SpiderIncognito, 2024-06-29:
### Hardware

ESP32

### Firmware version

V2.4.5

### Application version

2.4.5

### What happened? What did you expect to happen?

When reseting or shutdown my esp32 wroom, i lost all configuration of Somfy’ pages
Monitoring USB port result : see logs

### How to reproduce it (step by step)

```bash
1 rest esp32
2. SHitdown esp32
...
```

### Logs

```bash
Startup/Boot…
Mounting File System…
E (404) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}
E (406) esp_littlefs: mount failed, (-84)
E (410) esp_littlefs: Failed to initialize LittleFS
Error mounting file system
Chip Model ESP32-Startup/Boot
```

rstrouse, 2024-06-29:
Which ESP32 board are you using and what are you using to power it with?  There are a bunch of them out there with WROOM on them.

Just to be sure there is not an issue with the transceiver disconnect it for now and check the boot.  I am assuming you chose the suggested pins for the transceiver from the wiki.  The most likely causes for little FS to not be able to initialize the root is, a failed chip on the board, low power when writing, or a bad memory sector.

I am assuming that the ESPSomfy UI still shows up when you access it.  Is that correct?

SpiderIncognito, 2024-06-29:
It’s a esp32 vroom
It’s ok to create room or to pair with Somfy and command the shutter , the problem it’s if I reset the esp or stop the poWer I loose all the shade rooms groups as it is t save the configuration on the littlefs of the esp. I look about your program on arduino ide and see that it’s provide from the initialization of th littlefs.
Tomorrow I m come back to home and try to power it from power adapter and not from USB connection.
Regards

SpiderIncognito, 2024-06-30:
I power the esp32 on power supply and the problem is the same, lost configuration of somfy at reboot.

rstrouse, 2024-06-30:
This sounds like the ESP32 has a faulty chip.  The initialization of the littlefs partition simply writes to the first byte and reads it back when mounting.  If you have a backup run the following command then restore the backup to see if the ESP32 marks the location as a bad sector.

Run the following command by opening a browser and enter the url below.  Give it some time to download the files from github then reboot the device.  It should take no more than 2 minutes but you will see what it is doing in the logs.
`http://<ip address>/recoverFilesystem`

SpiderIncognito, 2024-07-01:
after send the commande http://<ip address>/recoverFilesystem tes response is :
{
  "status": "OK",
  "desc": "Recovering filesystem from github please wait!!!"
}

Nothing else happen.
Sorry for my english, i m french...

SpiderIncognito, 2024-07-01:
on monitoring port usb
after send htt command yopu provide :
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (538) esp_core_dump_flash:⸮⸮⸮o core dump partition found!
E (538) esp_core_dump_flash: No core dump partition found!
E (361) psram: PSRAM ID read error: 0xffffffff

Startup/Boot…
Mounting File System…
E (403) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}

E (405) esp_littlefs: mount failed, (-84)
E (409) esp_littlefs: Failed to initialize LittleFS
Error mounting file system
Chip Model ESP32-

SpiderIncognito, 2024-07-01:
i have installed ESP32 LittleFS data upload on arduino ide and test :
esptool.py v4.6
Serial port COM7
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d8:bc:38:f9:54:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00290000 to 0x003effff...
Compressed 1441792 bytes to 98405...
Writing at 0x00290000... (14 %)
Writing at 0x00358767... (28 %)
Writing at 0x00360599... (42 %)
Writing at 0x0038063c... (57 %)
Writing at 0x00394f7f... (71 %)
Writing at 0x003af099... (85 %)
Writing at 0x003d7d17... (100 %)
Wrote 1441792 bytes (98405 compressed) at 0x00290000 in 9.8 seconds (effective 1178.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

SpiderIncognito, 2024-07-01:
i upload littleFS_test on my esp32 and no problem with writting on it
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
SPIFFS-like write file to new path and delete it w/folders
Create missing folders of: /new1/new2/new3/hello3.txt
Writing file to: /new1/new2/new3/hello3.txt
- file written
Listing directory: /
  DIR : new1
Listing directory: /new1
  DIR : new2
Listing directory: /new1/new2
  DIR : new3
Listing directory: /new1/new2/new3
  FILE: hello3.txt	SIZE: 6
Deleting file and empty folders on path: /new1/new2/new3/hello3.txt
- file deleted
Removing all empty folders on path: /new1/new2/new3/hello3.txt
Listing directory: /
Creating Dir: /mydir
Dir created
Writing file: /mydir/hello2.txt
- file written
Listing directory: /
  DIR : mydir
Listing directory: /mydir
  FILE: hello2.txt	SIZE: 6
Deleting file: /mydir/hello2.txt
- file deleted
Removing Dir: /mydir
Dir removed
Listing directory: /
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing................................................................
 - 1048576 bytes written in 16319 ms
- reading................................................................
- 1048576 bytes read in 395 ms
Deleting file: /test.txt
- file deleted
Test complete

SpiderIncognito, 2024-07-01:
I try to compile but have many's error with Network.h that doesn't recognize conn_types_t
i look that the definition is in ConfigSettings.h and that it's include before Network.h in SomfyController then i don(t understand why this errors.
Trying to include in network.h, same errors at complilation.

rstrouse, 2024-07-01:
This is likely because you are using v3.x of the Arduino Core.  That is too big to fit on an ESP32 with ESPSomfy RTS as well.

rstrouse, 2024-07-01:
Also, if you have attempted to update using a different partitioning scheme then this could also be a problem.  This would be because the formatting of the LittleFS partition would start at a different location.  Perhaps try to onboard this from scratch so that it first erases the partitions then recreates them.  

It is likely that during operation there would be no errors because LittleFS does not read back any writes to the partition and neither does ESPSomfy RTS.  LittleFS is not as robust as a filesystem on an OS as the ESP32 would have to spend all its time on IO.   Reading it back only occurs during boot and if it was written to a location in another partition boundary LittleFS would be corrupted.

SpiderIncognito, 2024-07-02:
good news, it's work now.
process 
Fist try to upload firmware with ESP32 Flash : Upload OK, working fine : nOk
then try esptools process : erase_flash, write_flash : ok, working i: nOk
the  using arduino ide 
change esp32 card 3.0 with esp32 2.07
compile and inject -> ok, working fine : nOke
Write data with ESP LittleFS Data upload, working fine -> ok
never loose the data i saved from somfy configuration

SpiderIncognito, 2024-07-02:
thanks a lot for your usefull help.
Very good job for your project.
Best regards.
Jean-Luc

closed 2024-07-02 by SpiderIncognito